cyberman54/ESP32-Paxcounter #781: Op code 88 (set time) we crash on TGGO T3 V1.6

hansju-11, 2021-04-01:
first try

[I][reset.cpp:125] enter_deepsleep(): Waiting until LMIC is idle...
[I][lorawan.cpp:449] myRxCallback(): Received 5 byte(s) of payload on port 2
[I][rcommand.cpp:379] set_time(): Remote command: set time to 1617306575
[I][timekeeper.cpp:100] setMyTime(): [71.542] UTC time: 1617306575.000 sec
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x4009166f  PS      : 0x00060b30  A0      : 0x800814dc  A1      : 0x3ffe6770
A2      : 0x00000000  A3      : 0x0000003f  A4      : 0x00000000  A5      : 0x00000000
A6      : 0x3ffc7068  A7      : 0x4051e2b0  A8      : 0x800eae55  A9      : 0x3ffe6710
A10     : 0x3ffbb350  A11     : 0x40088edc  A12     : 0x3ffba6b8  A13     : 0x00000000
A14     : 0x00000000  A15     : 0xffffffb3  SAR     : 0x00000004  EXCCAUSE: 0x0000001c
EXCVADDR: 0x00000000  LBEG    : 0x400029ac  LEND    : 0x400029cb  LCOUNT  : 0x00000000

ELF file SHA256: 0000000000000000

Backtrace: 0x4009166f:0x3ffe6770 0x400814d9:0x3ffe6790 0x400d6825:0x3ffe67c0 0x400d6b15:0x3ffe67e0 0x400d6bd6:0x3ffe6820 0x40092d22:0x3ffe6850

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)

second try 

[I][lorawan.cpp:449] myRxCallback(): Received 5 byte(s) of payload on port 2
[I][rcommand.cpp:379] set_time(): Remote command: set time to 1617306650
[I][timekeeper.cpp:100] setMyTime(): [61.234] UTC time: 1617306650.000 sec
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x4009166f  PS      : 0x00060930  A0      : 0x800814dc  A1      : 0x3ffe6770
A2      : 0x00000000  A3      : 0x0000003f  A4      : 0x00000000  A5      : 0x00000000
A6      : 0x3ffc7068  A7      : 0x404e9df3  A8      : 0x800eae55  A9      : 0x3ffe6710
A10     : 0x3ffbb38c  A11     : 0x40088edc  A12     : 0x3ffba71c  A13     : 0x00000000
A14     : 0x00000000  A15     : 0xffffffb3  SAR     : 0x00000004  EXCCAUSE: 0x0000001c
EXCVADDR: 0x00000000  LBEG    : 0x400029ac  LEND    : 0x400029cb  LCOUNT  : 0x00000000

ELF file SHA256: 0000000000000000

Backtrace: 0x4009166f:0x3ffe6770 0x400814d9:0x3ffe6790 0x400d6825:0x3ffe67c0 0x400d6b15:0x3ffe67e0 0x400d6bd6:0x3ffe6820 0x40092d22:0x3ffe6850

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 188777542, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:10124
load:0x40080400,len:5828
entry 0x400806a8
[I][configmanager.cpp:102] loadConfig(): Loading device configuration from NVRAM...
[I][configmanager.cpp:127] loadConfig(): Runtime configuration v2.4.0 loaded
[I][main.cpp:127] setup(): Starting paxcounter_f56af210 v2.4.0 (runmode=4 / restarts=3)

cyberman54, 2021-04-02:
please increase version number in platformio.ini (this triggers clearing NVRAM values during first start). Currently i cannot reproduce this issue.

cyberman54, 2021-04-06:
This may happen if the device is going to sleep, while it receives the rcommand. Need to be checked. I can't reproduce this issue with a device that does not go to sleep.

cyberman54, 2021-04-06:
This should be fixed now in code in development branch: https://github.com/cyberman54/ESP32-Paxcounter/commit/d098d6ada37931ac3625ebc58c634345a1b9d96e
Can you please test and close this issue, if this works for you? Thanks.

cyberman54, 2021-04-06:
PS: time-of-day, once set by rcommand 0x88 or by other timesource (LORA, GPS, RTC) is now preserved during deep sleep.

hansju-11, 2021-04-07:
I change the version to 3.4.0 
but still the same problem.

[I][lorawan.cpp:449] myRxCallback(): Received 5 byte(s) of payload on port 2
[D][lorawan.cpp:412] myEventCallback(): TXCOMPLETE
[I][rcommand.cpp:379] set_time(): Remote command: set time to 1617813000
[I][timekeeper.cpp:101] setMyTime(): [131.553] UTC time: 1617813000.000 sec
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x4009171f  PS      : 0x00060f30  A0      : 0x8008150d  A1      : 0x3ffef7b0
A2      : 0x00000000  A3      : 0x0000003f  A4      : 0x00000000  A5      : 0x00000000
A6      : 0x3ffc70d8  A7      : 0x406071b2  A8      : 0x800eb429  A9      : 0x3ffef750
A10     : 0x3ffcd7c8  A11     : 0x40088f8c  A12     : 0x3ffba6b8  A13     : 0x00000000
A14     : 0x00000000  A15     : 0xffffffb2  SAR     : 0x00000004  EXCCAUSE: 0x0000001c
EXCVADDR: 0x00000000  LBEG    : 0x400029ac  LEND    : 0x400029cb  LCOUNT  : 0x00000000

ELF file SHA256: 0000000000000000

Backtrace: 0x4009171f:0x3ffef7b0 0x4008150a:0x3ffef7d0 0x400d6a89:0x3ffef800 0x400d6d79:0x3ffef820 0x400d6e3a:0x3ffef860 0x40092dd2:0x3ffef890

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 188777542, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:10124
load:0x40080400,len:5828
entry 0x400806a8
[I][configmanager.cpp:102] loadConfig(): Loading device configuration from NVRAM...
[I][configmanager.cpp:127] loadConfig(): Runtime configuration v3.4.0 loaded
[I][main.cpp:127] setup(): Starting paxcounter_f56af210 v3.4.0 (runmode=1 / restarts=1)

I will do some deeper debug tomorow

cyberman54, 2021-04-07:
Did you use the current code of development branch?

Please enable esp32 backtrace in platformio logging, this will show the lines of code which cause the crash.

hansju-11, 2021-04-07:
Hello yes I just uploaded it 1 hour ago.

for me the error comes from
timekeeper.cpp line 112  (timerWrite ...)

#if (!defined GPS_INT && !defined RTC_INT)
    timerWrite(ppsIRQ, 0); // reset pps timer
    CLOCKIRQ();            // fire clock pps, this advances time 1 sec
#endif

if I disable the function in esp32-hal-timer.c  line 106 

void IRAM_ATTR timerWrite(hw_timer_t *timer, uint64_t val){

    ESP_LOGD(TAG, "hjb error 1 esp32-hal-timer.c 108");
   // timer->dev->load_high = (uint32_t) (val >> 32);
   // timer->dev->load_low = (uint32_t) (val);
   // timer->dev->reload = 1;
    ESP_LOGD(TAG, "hjb error 1 esp32-hal-timer.c 112");
}

I did not get an error.

I will activate the esp backtrace to see if I get more informations for you.

hansju-11, 2021-04-07:
here are my  paxcounter.conf and the hal file I use  ( I put .txt at the end of the file to be able to upload it)

[ttgov21new.h.txt](https://github.com/cyberman54/ESP32-Paxcounter/files/6273378/ttgov21new.h.txt)

[paxcounter.conf.txt](https://github.com/cyberman54/ESP32-Paxcounter/files/6273382/paxcounter.conf.txt)

hansju-11, 2021-04-07:
I did a check on NULL. in timekeeper.cpp 

#if (!defined GPS_INT && !defined RTC_INT)

ESP_LOGD(TAG, "hjb error 1 %d", ppsIRQ);
if (ppsIRQ == NULL) {
  ESP_LOGD(TAG, "ppsIRQ is NULL");
} 
else {
    timerWrite(ppsIRQ, 0); // reset pps timer

    CLOCKIRQ();            // fire clock pps, this advances time 1 sec
}
#endif

output ---
[I][timekeeper.cpp:101] setMyTime(): [71.560] UTC time: 1617817930.000 sec
[D][timekeeper.cpp:113] setMyTime(): hjb error 1 0
[D][timekeeper.cpp:115] setMyTime(): ppsIRQ is NULL
[D][microTime.cpp:315] setTime(): hjb error 1, timekeeper.cpp 315
-----

cyberman54, 2021-04-07:
Yes, this is the bug. Thanks for spotting it!
Now it should be finally solved:
https://github.com/cyberman54/ESP32-Paxcounter/commit/c00612ad76906c3fc67151c1f6dc5d83926f01b8

closed 2021-04-07 by cyberman54