cyberman54/ESP32-Paxcounter #783: Redirect serial output to SD card

drtorchwood, 2021-04-05:
Hi there, 

I'm looking for a simple way to redirect the serial output to the SD card. Is there any option available? I haven't found anything regarding this topic in the manual. I don't mean the "normal" paxcounter data but the debug log. 

Thanks!

cyberman54, 2021-04-05:
Yes, that would be an interesting option. But i don't have an example for this. Be careful with writes to SD card, this may trigger the watchdog if it's too slow, and you have check that access to the SD interface is mutexe'd, depending if you use data lines that are in multi purpose use.

cyberman54, 2022-04-12:
SD write is freshly prepared in code (sdcard.cpp), but for logging onto SD weed to wait until [this bug](https://github.com/espressif/arduino-esp32/issues/4990) is fixed in arduino-espressif32 framework.

cyberman54, 2022-11-27:
@drtorchwood the SD serial log redirection feature was implemented today.

closed 2022-11-27 by cyberman54