cyberman54/ESP32-Paxcounter #801: Cant build project v2.4.0

v8-rt, 2021-04-27:
Hi guys,
Can you please help me?
I strictly followed this video: https://www.youtube.com/watch?v=WuRb43B0jmo (with the exception that i used lorav21 board)
yet VS Code wont let me build the Project.

This is what i get:

*****
> Executing task: C:\Users\Andi\.platformio\penv\Scripts\platformio.exe run <

Can not remove temporary directory `C:\Users\Andi\Documents\PlatformIO\Projects\ESP32-Paxcounter-2.4.0X\.pio\build`. Please remove it manually to avoid build issues
Processing usb (platform: espressif32; board: heltec_wifi_lora_32_V2; framework: arduino)
----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
Parsing hardware configuration from C:\Users\Andi\Documents\PlatformIO\Projects\ESP32-Paxcounter-2.4.0X\src\hal\heltecv21.h
Parsing LMIC configuration from C:\Users\Andi\Documents\PlatformIO\Projects\ESP32-Paxcounter-2.4.0X\src\lmic_config.h
Parsing LORAWAN keys from C:\Users\Andi\Documents\PlatformIO\Projects\ESP32-Paxcounter-2.4.0X\src\loraconf.h
Parsing OTA keys from C:\Users\Andi\Documents\PlatformIO\Projects\ESP32-Paxcounter-2.4.0X\src\ota.conf
Target board: heltec_wifi_lora_32_V2 @ 921600bps
Partition table: min_spiffs.csv
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/heltec_wifi_lora_32_V2.html
PLATFORM: Espressif 32 (3.2.0) > Heltec WiFi LoRa 32 (V2)
HARDWARE: ESP32 240MHz, 320KB RAM, 8MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-arduinoespressif32 3.10006.210326 (1.0.6)
 - tool-esptoolpy 1.30000.201119 (3.0.0)
 - tool-openocd-esp32 2.1000.20201202 (10.0)
 - toolchain-xtensa32 2.50200.97 (5.2.0)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ deep, Compatibility ~ soft
Library Manager: Installing git+https://github.com/SukkoPera/Arduino-Rokkit-Hash.git
FileNotFoundError: [WinError 2] Das System kann die angegebene Datei nicht finden:
  File "C:\Users\Andi\.platformio\penv\Lib\site-packages\platformio\builder\main.py", line 177:
    env.SConscript("$BUILD_SCRIPT")
  File "C:\Users\Andi\.platformio\packages\tool-scons\scons-local-4.1.0\SCons\Script\SConscript.py", line 591:
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\Users\Andi\.platformio\packages\tool-scons\scons-local-4.1.0\SCons\Script\SConscript.py", line 280:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "C:\Users\Andi\.platformio\platforms\espressif32\builder\main.py", line 225:
    target_elf = env.BuildProgram()
  File "C:\Users\Andi\.platformio\packages\tool-scons\scons-local-4.1.0\SCons\Util.py", line 658:
    return self.method(*nargs, **kwargs)
  File "C:\Users\Andi\.platformio\penv\Lib\site-packages\platformio\builder\tools\platformio.py", line 62:    
    env.ProcessProjectDeps()
  File "C:\Users\Andi\.platformio\packages\tool-scons\scons-local-4.1.0\SCons\Util.py", line 658:
    return self.method(*nargs, **kwargs)
  File "C:\Users\Andi\.platformio\penv\Lib\site-packages\platformio\builder\tools\platformio.py", line 140:   
    project_lib_builder = env.ConfigureProjectLibBuilder()
  File "C:\Users\Andi\.platformio\packages\tool-scons\scons-local-4.1.0\SCons\Util.py", line 658:
    return self.method(*nargs, **kwargs)
  File "C:\Users\Andi\.platformio\penv\Lib\site-packages\platformio\builder\tools\piolib.py", line 1062:      
    project.install_dependencies()
  File "C:\Users\Andi\.platformio\penv\Lib\site-packages\platformio\builder\tools\piolib.py", line 898:       
    lm.install(spec)
  File "c:\users\andi\.platformio\penv\lib\site-packages\platformio\package\manager\_install.py", line 48:
    pkg = self._install(
  File "c:\users\andi\.platformio\penv\lib\site-packages\platformio\package\manager\library.py", line 81:
    return super(LibraryPackageManager, self)._install(
  File "c:\users\andi\.platformio\penv\lib\site-packages\platformio\package\manager\_install.py", line 97:
    pkg = self.install_from_url(spec.url, spec, silent=silent)
  File "c:\users\andi\.platformio\penv\lib\site-packages\platformio\package\manager\_install.py", line 141:
    vcs = VCSClientFactory.new(tmp_dir, url)
  File "c:\users\andi\.platformio\penv\lib\site-packages\platformio\package\vcsclient.py", line 54:
    obj = getattr(sys.modules[__name__], "%sClient" % type_.title())(
  File "c:\users\andi\.platformio\penv\lib\site-packages\platformio\package\vcsclient.py", line 137:
    self.configure()
  File "c:\users\andi\.platformio\penv\lib\site-packages\platformio\package\vcsclient.py", line 146:
    result = proc.exec_command([cls.command, "--exec-path"])
  File "c:\users\andi\.platformio\penv\lib\site-packages\platformio\proc.py", line 117:
    p = subprocess.Popen(*args, **kwargs)
  File "C:\Program Files\Python39\lib\subprocess.py", line 951:
    self._execute_child(args, executable, preexec_fn, close_fds,
  File "C:\Program Files\Python39\lib\subprocess.py", line 1420:
    hp, ht, pid, tid = _winapi.CreateProcess(executable, args,
================================================================================================================================================ [FAILED] Took 0.96 seconds ================================================================================================================================================
Environment    Status    Duration
-------------  --------  ------------
usb            FAILED    00:00:00.956
========================================================================================================================================== 1 failed, 0 succeeded in 00:00:00.956 ========================================================================================================================================== 
The terminal process "C:\Users\Andi\.platformio\penv\Scripts\platformio.exe 'run'" terminated with exit code: 1.

Terminal will be reused by tasks, press any key to close it.

*****
It doesnt work with selecting other boards either....
Any help is appreciated. Thanks!

cyberman54, 2021-04-28:
This looks like a problem with your platformio toolchain. Please try tunning platformio with administrator rights. What platform are you using, windows or linux? VSC or atom?

v8-rt, 2021-04-28:
Hi, Thank you for your quick reply! Unfortunately it's the same when running VSC as admin. I will try to build with another computer later. Do you think I need to install more extensions?

cyberman54, 2021-04-28:
You don't need any other extension that platformio for VSC.
And a working python installation under VSC.

v8-rt, 2021-04-28:
Hi cyberman54, it works with VSC (+platformio extension) for macOS. I've tried it on another windows and still had no success. Maybe I am doing something wrong there or their current version has a bug? Idk. But since your paxcounter compiled and uploaded successfully from Mac, I am already happy. Thanks for your quick response!
(if I'll try on windows again some day and it works, I'll try to figure out what went wrong)

closed 2021-04-28 by v8-rt

hungryhunter68, 2021-06-03:
Hi,
I have exactly the same issue under windows & linux - unfortunately I have no MAC available to test. Any idea what causes this issue?
Thank you very much.