cyberman54/ESP32-Paxcounter #802: Which of the listed ESP32 boards is suitable for a beginner? Or what are the pros/cons between them?

victorhooi, 2021-05-21:
I'm a relative newcomer to the world of ESP32/Arduino

This project looks really awesome =)!

I'm hoping to build this, and use it on both BLE and WiFi. (Readme mentions I will need two boards, one dedicated to BLE, and one to WiFi).

The README lists quite a few ESP32 devices (Heltec, TTGO, Pycom, RadioShuttle etc.), and I'm not familiar with any of them, or have any idea how to pick.

Is there any guidance for which board to use? Or what is well supported in the community?

- I'm based out of Australia (if that affects availability/options)
- Since I'm only deploying 3-4 units, cost is not as big of an issue at this scale
- Having an included case would be great, so it looks neat/tidy on the wall
- I could potentially run this off mains power and a power brick - but it seems like all the options are based around using a LiPo cell, right? What sort of battery life are people seeing using BLE, vs WiFi? Or does anybody run this off mains?

On the networking side - I'm just trying to understand. Is it saying we can send our payload either over LoRaWan, **or** over a normal TCP/IP network?

If we go the LoRaWan route, do I need my own LoRaWan gateway locally, that this connects to, which is then connected to my own internet connection? (e.g. [this](https://www.chirpstack.io/gateway-os/))

If we go the traditional TCP/IP route, with MQTT - what are people using for internet connectivity? I assume the Wifi chip on the ESP32 is already used for monitoring, so do you need to find a ESP32 with an Ethernet port?

Actually, are there any compatible boards with POE Ethernet ports?

If there's any recommended build guides, or tutorials?

Or I suppose the lazy route - are there any pre-built boards out there, that help support this project?

FlorianLudwig, 2021-05-22:
Hi @victorhooi 

> On the networking side - I'm just trying to understand. Is it saying we can send our payload either over LoRaWan, or over a normal TCP/IP network?

correct.

> assume the Wifi chip on the ESP32 is already used for monitoring, so do you need to find a ESP32 with an Ethernet port?

Exactly, you cannot use the WiFi for data transfer with the current implementation. So you would need an ethernet connection.

the olimex esp32-poe might be interesting to you:

https://www.olimex.com/Products/IoT/ESP32/ESP32-POE/open-source-hardware

It does work well with this project.

cyberman54, 2021-05-22:
@victorhooi If you want a display, you can go for a TTGO T-Beam, as many here do. If you don't need a display, but LORA, the pycom Lopy4 with extension board v3 and case is a suitable option. If you need ethernet, use the olimex.

PS: Please transfer this to discussion forum, i'm closing this here since it is not an issue.

closed 2021-05-22 by cyberman54