cyberman54/ESP32-Paxcounter #810: Compilation error: include/libpax_helpers.h:5:24: fatal error: libpax_api.h: No such file or directory

carusoe11, 2021-06-06:
no compilation success with VS/platformio (Core 5.1.1·Home 3.3.4) under Windows10
Version: 1.56.2 (user setup)
Commit: 054a9295330880ed74ceaedda236253b4f39a335
Date: 2021-05-12T17:13:13.157Z
Electron: 12.0.4
Chrome: 89.0.4389.114
Node.js: 14.16.0
V8: 8.9.255.24-electron.0
OS: Windows_NT x64 10.0.19042

Errormessage:
"include/libpax_helpers.h:5:24: fatal error: libpax_api.h: No such file or directory"

configuration in platformio.ini:
lib_deps_basic =
https://github.com/cyberman54/libpax.git
or
https://github.com/dbSuS/libpax.git

no libraries found with
* CLI  > platformio lib search "header:libpax_api.h"
* Web  > https://platformio.org/lib/search?query=header:libpax_api.h

cyberman54, 2021-06-06:
@FlorianLudwig can you help here?

FlorianLudwig, 2021-06-06:
Hi @carusoe11 

which version of paxcounter did you try to compile? Which branch / commit / release?

Did you run `clean` before you build?

I did a clean build from the current master branch just now and it worked as expected:

```
Library Manager: Installing git+https://github.com/cyberman54/libpax.git
Library Manager: libpax @ 0.1.3+sha.43d327a has been installed!
```

Regarding the platformio library registry - we should register libpax so it can be found via the search. The compile should work anyways as the `platformio.ini` directly refers a git repository and not by name.

carusoe11, 2021-06-06:
Hi @FlorianLudwig, thank you for your ultrafast response!

I took paxcounter-release-version 3.0.0, tried to clean before build, no luck.

Meanwhile I got >>the same code<< successfully & suprisingly running, 
compiled with Atom/PlatformIO on a Linux-machine in spite of all mainstream VS-driftings.

FlorianLudwig, 2021-06-06:
I suspect that is not related to vs vs atom nor to linux vs windows differences but some config / cache issue of platformio.

The platformio version seems not to be the issue, as I also used 5.1.1.

Could you delete all your platform.io caches?

The `clean` does not delete 100% sadly.

So that would be inside the the project folder `.pio/` and in your user folder `.platformio`.

Please note: this deletes all platformio dowloaded files and you must re-download quite a lot. Not just for this project but all platformio projects. But this makes sure that it is not a platformio setup issue.

carusoe11, 2021-06-06:
After the installation of the pio-CLI trying:

C:\Users\Mein PC\.platformio\penv\Scripts>pio system prune
Prune cached data:
 - cached API requests
 - cached package downloads
 - temporary data
Do you want to continue? [y/N]: y
Space on disk: 203.44MB

Prune unnecessary core packages:
Calculating...
Do you want to continue? [y/N]: y
Space on disk: 0B

Prune unnecessary development platform packages:
Calculating...
Do you want to continue? [y/N]: y
Space on disk: 0B

Total reclaimed space: 203.44MB
#######################################

still no luck with PlatformIO:
compilation terminated.
In file included from include/senddata.h:4:0,
                 from include/cyclic.h:5,
                 from src/cyclic.cpp:5:
include/libpax_helpers.h:5:24: fatal error: libpax_api.h: No such file or directory

cyberman54, 2021-06-07:
Could you please try adding this line to your platformio.ini: 

lib_ldf_mode = deep

cyberman54, 2022-02-22:
Toolchain problem. Could not reproduce. Closed.

closed 2022-02-22 by cyberman54