cyberman54/ESP32-Paxcounter #824: TTGO T-Beam no GPS Fix

doebi, 2021-08-05:
I am having a T-Beam 0.7, successfully configured and flashed it. The Integration with TTNV3 is also working, as I am receiving PAX values. But the GPS simply won't get a fix and send its values.

Any direction on how to approach this error? There are no significantly interesting logs nor any errors.

Thanks you very much.

cyberman54, 2021-08-05:
What's printed on the GPS chip module, NEO6  or NEO8 ?

doebi, 2021-08-05:
Its a NEO-6M.

doebi, 2021-08-05:
the hal-file I am using:

```c
// clang-format off
// upload_speed 921600
// board ttgo-t-beam

#ifndef _TTGOBEAM_H
#define _TTGOBEAM_H

#include <stdint.h>

// Hardware related definitions for TTGO T-Beam board
// (only) for older T-Beam version T22_V05 eternal wiring LORA_IO1 to GPIO33 is needed!
//
// pinouts taken from http://tinymicros.com/wiki/TTGO_T-Beam

#define HAS_LED GPIO_NUM_14 // on board green LED, only new version TTGO-BEAM V07
//#define HAS_LED GPIO_NUM_21 // on board green LED, only old verison TTGO-BEAM V05

#define HAS_LORA 1       // comment out if device shall not send data via LoRa
#define CFG_sx1276_radio 1 // HPD13A LoRa SoC
#define HAS_BUTTON GPIO_NUM_39 // on board button (next to reset)
#define BAT_MEASURE_ADC ADC1_GPIO35_CHANNEL // battery probe GPIO pin -> ADC1_CHANNEL_7
#define BAT_VOLTAGE_DIVIDER 2 // voltage divider 100k/100k on board

// GPS settings
#define HAS_GPS 1 // use on board GPS
#define GPS_SERIAL 9600, SERIAL_8N1, GPIO_NUM_12, GPIO_NUM_15 // UBlox NEO 6M
//#define GPS_INT GPIO_NUM_34 // 30ns accurary timepulse, to be external wired on pcb: NEO 6M Pin#3 -> GPIO34

// enable only if device has these sensors, otherwise comment these lines
// BME680 sensor on I2C bus
//#define HAS_BME 1 // Enable BME sensors in general
//#define HAS_BME680 SDA, SCL
//#define BME680_ADDR BME680_I2C_ADDR_PRIMARY // !! connect SDIO of BME680 to GND !!

// display (if connected)
#define HAS_DISPLAY 1
#define MY_DISPLAY_SDA SDA
#define MY_DISPLAY_SCL SCL
#define MY_DISPLAY_RST NOT_A_PIN
//#define MY_DISPLAY_FLIP  1 // use if display is rotated

//#define DISABLE_BROWNOUT 1 // comment out if you want to keep brownout feature

#endif
```

doebi, 2021-08-05:
and the console output:

```
00:19:03.722 > [I][configmanager.cpp:101] loadConfig(): Loading device configuration from NVRAM...
00:19:03.722 > [I][configmanager.cpp:126] loadConfig(): Runtime configuration v3.0.0 loaded
00:19:03.722 > [I][main.cpp:132] setup(): Starting paxcounter_f56af210 v3.0.0 (runmode=0 / restarts=0)
00:19:03.722 > [I][main.cpp:148] setup(): This is ESP32 chip with 2 CPU cores, WiFi/BT/BLE, silicon revision 1, 4MB external Flash
00:19:03.722 > [I][main.cpp:150] setup(): Internal Total heap 234772, internal Free Heap 208232
00:19:03.722 > [I][main.cpp:153] setup(): SPIRam Total heap 4194252, SPIRam Free Heap 4194252
00:19:03.722 > [I][main.cpp:156] setup(): ChipRevision 1, Cpu Freq 240, SDK Version v3.3.5-1-g85c43024c
00:19:03.722 > [I][main.cpp:158] setup(): Flash Size 4194304, Flash Speed 40000000
00:19:03.722 > [I][main.cpp:160] setup(): Wifi/BT software coexist version 1.2.0
00:19:03.722 > [I][main.cpp:164] setup(): IBM LMIC version 1.6.1468577746
00:19:03.722 > [I][main.cpp:169] setup(): Arduino LMIC version 4.0.0.0
00:19:03.722 > [I][lorawan.cpp:176] printKey(): DevEUI: XXX
00:19:03.722 > [I][lorawan.cpp:176] printKey(): AppEUI: XXX
00:19:03.722 > [I][lorawan.cpp:176] printKey(): AppKey: XXX
00:19:03.722 > [I][main.cpp:174] setup(): TinyGPS+ version 1.0.2
00:19:03.722 > [I][i2c.cpp:46] i2c_scan(): Starting I2C bus scan...
00:19:03.722 > [I][i2c.cpp:77] i2c_scan(): Device found at 0x3C, type = SSD1306
00:19:03.722 > [I][i2c.cpp:80] i2c_scan(): 1 I2C device(s) found
00:19:13.610 > [I][main.cpp:212] setup(): PSRAM found and initialized
00:19:13.610 > [I][main.cpp:245] setup(): Starting LED Controller...
00:19:13.610 > [I][power.cpp:213] calibrate_voltage(): ADC characterization based on reference voltage stored in eFuse
00:19:13.610 > [I][main.cpp:291] setup(): Starting libpax...
00:19:13.624 > [I][main.cpp:300] setup(): WIFISCAN: off
00:19:13.624 > [I][main.cpp:305] setup(): BLESCAN: on
00:19:13.624 > [I][blescan.cpp:281] start_BLE_scan(): Initializing bluetooth scanner ...
00:19:14.169 > [I][blescan.cpp:239] register_ble_callback(): Register GAP callback
00:19:14.169 > [I][blescan.cpp:270] register_ble_callback(): Set GAP scan parameters
00:19:14.169 > [I][blescan.cpp:299] start_BLE_scan(): Bluetooth scanner started
00:19:14.169 > [I][main.cpp:323] setup(): Starting rcommand interpreter...
00:19:14.191 > [I][rcommand.cpp:528] rcmd_init(): Rcommand send queue created, size 55 Bytes
00:19:14.191 > [I][gpsread.cpp:39] gps_init(): Opening serial GPS
00:19:14.191 > [I][main.cpp:330] setup(): Starting GPS Feed...
00:19:14.191 > [I][lorawan.cpp:257] lmic_init(): LORA send queue created, size 530 Bytes
00:19:14.207 > [I][lorawan.cpp:307] lmic_init(): Starting LMIC...
00:19:14.207 > [I][main.cpp:432] setup(): Starting Interrupt Handler...
00:19:14.207 > [I][main.cpp:460] setup(): Starting Timers...
00:19:14.250 > [I][main.cpp:511] setup(): Starting Timekeeper...
00:19:14.250 > [I][timekeeper.cpp:167] timepulse_init(): Timepulse: internal (ESP32 hardware timer)
00:19:14.250 > [I][main.cpp:519] setup(): Features: OLED PSRAM LED BATT BLE GPS LORA PACKED BTN_PD TIME
00:19:27.207 > [I][lorawan.cpp:79] lora_setupForNetwork(): DEVaddr: 0xXXX | Network ID: 0xXXX | Network Type: 0
00:19:27.207 > [I][lorawan.cpp:80] lora_setupForNetwork(): RSSI: 12 | SNR: 7
00:19:27.207 > [I][lorawan.cpp:84] lora_setupForNetwork(): Radio parameters: SF7 | BW125 | CR 4/5
00:20:13.640 > [I][senddata.cpp:101] sendData(): Sending libpax wifi count: 0
00:20:13.640 > [I][senddata.cpp:104] sendData(): Sending libpax ble count: 2
00:20:13.640 > [I][lorawan.cpp:227] lora_send(): 4 byte(s) sent to LORA
```

cyberman54, 2021-08-06:
All looking good. Did you go outside to get a gps fix? After coldstart this can take up to 30 minutes. Look at the "Sats" Display, it shows if and how many Sats are seen.

doebi, 2021-08-07:
Ok, I did some more tests outside and now it really works. It seems that the device needed a cold restart between these consecutive firmware edits to properly load the changes.

Thanks a lot.

closed 2021-08-07 by doebi