cyberman54/ESP32-Paxcounter #825: Remove known devices from the statistics

hbjorgo, 2021-08-14:
It would be nice to be able to trigger a function that will scan for a predefined amount of time and exclude those devices from the statistics. I'm counting devices passing by the sensor, but static devices nearby add to these statistics.

There could be two functions - one for clearing the list of "known" devices, and another to scan and add new devices.

That way one can trigger this scan and add function at an interval and add new, static devices. This could be a scheduled job in the program itself at a specific time of day (with little or no traffic), or sent over the air. Clearing the list can also be done at a (slower) interval to reset this list.

cyberman54, 2021-08-15:
I won't implement such a function, since i don't see how we could get it in compliance with privacy rules, i.e. GDPR.

If you see certain amount of static devices, it should be easy to subtract this on server / application side as a baseline from current counts.

hbjorgo, 2021-08-17:
I understand. One issue with subtracting the baseline on the application side is that not all static devices are present at all times. That will cause the application to subtract more devices than it should at some points.

closed 2021-08-17 by hbjorgo