cyberman54/ESP32-Paxcounter #832: sds011_loop(): SDS011 error: 1

colvert, 2021-12-11:
Hi,

I am trying to integrate the SDS sensor to my TTGO LoRA .
The BME integration is fine but I have still some issues with the SDS011 sensor.
I got the same issue than the one described in https://github.com/cyberman54/ESP32-Paxcounter/issues/576
I saw at the end that @x-RoNiN-x found a solution but unfortunately I did not and I am not sure that the solution was indicated in the thread.
I tried to uncomment the line as suggested by @AugustQu without success.

I checked the sensor, I got good values when I use the python lib and connect the sensor with the USB cable, so I assume the sensor is fine, the problem is the integration with the TTGO LoRa.
I checked the wiring, I declared
#define SDS_TX 12     // connect to RX on the SDS01
#define SDS_RX 35     // connect to TX on the SDS011

init looks fine
18:47:12.237 > [I][main.cpp:400] setup(): init fine-dust-sensor
but I got only
19:06:05.863 > [I][sds011read.cpp:35] sds011_loop(): SDS011 error: 1

I am a bit puzzled on how to troubleshoot
I am using  TTGO LoRa T3_V1.6
the sensor is a Nova FM sensor 20 07 08 09

any help/hint will be welcome...

x-RoNiN-x, 2021-12-11:
Hi Colvert, my solution is explained in a separate project.
But the solution is only available in German at the moment. I link you the repo here. [](https://github.com/dasdigidings/Paxcounter-SDS011-Lora-Umweltsensor) Either try to translate it or write a small email to info@dasdigidings.de. English language is welcome. :-)

Kind regards 
xRoNiNx

colvert, 2021-12-11:
thanks for your quick reply, I will have a look at the German answer...
Ich habe Deutsch gelernt als ich in der Schule war...Mein Vocable is leider sehr begrenzt aber Ich kann probieren...Und meine Frau ist Eine Deutsche Lehrerin...

Tomi05-k, 2022-02-13:
Hi Colvert,
have you found a solution to this problem? 
I have the same error.

Many greetings from Germany 
~Tom

AugustQu, 2022-02-16:
Hi,

I didn't touch the code and the device for quite a while so I don't know in detail what is going on here. And I do not have a device here that produces the error (mine works fine).

Please look at the project here: https://github.com/AugustQu/SDS011

In the cpp-file you will find the modifications I added, especially in the function wakeup(). I replaced the code with the official call as documented. So I do this: sending the documented command-string to the device.

hope that helps.
AugustQ

PS: can you please look at the examples over there. And add the 2 files (cpp and h) to your project.

cyberman54, 2022-02-16:
New approach: In development branch there is a reworked version for SDS011 device management (changed to new lib) and improved SD card handling. Please try this version, it will go in the master branch soon.

cyberman54, 2022-02-16:
Closed, because won't fix with current code. Reworked code (currently in development branch) fixes this issue.

closed 2022-02-16 by cyberman54