cyberman54/ESP32-Paxcounter #844: wrong Payload in TTNv3

ChrSchultz, 2022-01-21:
1. set remotly Payloadmask to A1(161)  `14 86` on fPort2 in ttnv3 console.
2. then request config Data: `80`on fPort2 ttnv3
3. the bytes are: `05 0e 00 00 1e 00 32 08 1e a8 **a1** 33 2e 30 2e 33 00 00 00 00 00`
4. the decoded json in ttnv3-console details are: `"received_at": "2022-01-21T12:15:14.775518998Z",
    "uplink_message": {
      "session_key_id": "AX57ZIntgdYrWb2lM8VZDw==",
      "f_port": 3,
      "f_cnt": 730,
      "frm_payload": "BQ4AAB4AMggeqKEzLjAuMwAAAAAA",
      "decoded_payload": {
        "blescantime": 50,
        "bytes": [
          5,
          14,
          0,
          0,
          30,
          0,
          50,
          8,
          30,
          168,
          161,
          51,
          46,
          48,
          46,
          51,
          0,
          0,
          0,
          0,
          0
        ],
        "flags": {
          "adr": 0,
          "antenna": 1,
          "blescan": 1,
          "countermode": 1,
          "reserved": 0,
          "screen": 0,
          "screensaver": 0
        },
        "loradr": 5,
        "payloadmask": { //wrong payloadmask =b'10101001' = A9 (169)
          "battery": 1,
          "bme": 0,
          "counter": 0,
          "gps": 1,
          "reserved": 0,
          "sensor1": 0,
          "sensor2": 1,
          "sensor3": 0
        },
        "port": 3,
        "rgblum": 8,
        "rssilimit": 0,
        "sendcycle": 30,
        "txpower": 14,
        "version": "¡3.0.3", //wrong versionstring
        "wifichancycle": 0
      },`

cyberman54, 2022-01-22:
I opened PR at TheThingsNetwork for this:
https://github.com/TheThingsNetwork/lorawan-devices/pull/371

cyberman54, 2022-01-22:
I closed the PR for the decoder again. It does not solve the issue. Needs to be analyzed again.
 
The bytes found in the console
`05 0e 00 00 1e 00 32 08 1e a8 a1 33 2e 30 2e 33 00 00 00 00 00`
are correct.
`a8` is flag mask.
`a1` is payload mask as was set by remote command.

cyberman54, 2022-01-22:
@ChrSchultz i can't reproduce this. Could you please provide more data how to reproduce this isse?

"decoded_payload": {
        "blescantime": 2,
        "bytes": [
          5,
          14,
          0,
          0,
          30,
          50,
          2,
          30,
          168,
          134,
          51,
          46,
          49,
          46,
          48,
          0,
          0,
          0,
          0,
          0
        ],
        "flags": {
          "adr": 1,
          "antenna": 0,
          "blescan": 1,
          "countermode": 0,
          "reserved": 0,
          "screen": 1,
          "screensaver": 0
        },
        "loradr": 5,
        "payloadmask": {
          "battery": 1,
          "bme": 1,
          "counter": 0,
          "gps": 0,
          "reserved": 1,
          "sensor1": 0,
          "sensor2": 0,
          "sensor3": 0
        },
        "port": 3,
        "rgblum": 30,
        "rssilimit": 0,
        "sendcycle": 30,
        "txpower": 14,
        "version": "3.1.0",
        "wifichancycle": 50
      },

ChrSchultz, 2022-01-22:
I fixed the issue, i compiled with `-D LMIC_LORAWAN_SPEC_VERSION=LMIC_LORAWAN_SPEC_VERSION_1_0_3` cause i selected the setting in ttnv3 console of my device... I tried out with out the the v1.0.3 spec, and all works fine... I think the bug belongs to the lmic repository.

closed 2022-01-22 by ChrSchultz

cyberman54, 2022-01-23:
Please don't open issue, if you modified the code. Use discussions tabs.