cyberman54/ESP32-Paxcounter #857: M5core LMIC failure

davcx, 2022-02-21:
Hi,
trying to install on M5core with Lora RA-01 module I get this error on start:

12:27:46.736 > [I][lorawan.cpp:303] lmic_init(): Starting LMIC...
...
12:27:46.840 > [I][main.cpp:501] setup(): Features: PMU OLED BATT OTA BLE LORA SD PACKED BTN_PD TIME
12:27:48.812 > FAILURE 
12:27:48.813 > .pio\libdeps\usb\MCCI LoRaWAN LMIC library\src\lmic\radio.c:801
12:28:16.856 > [I][timesync.cpp:50] timesync_request(): [41.005] Timeserver sync request started, seqNo#13
12:28:16.856 > [I][power.cpp:400] printIP5306Stats(): IP5306: Power Source: USB, Battery State: CHARGED, Battery Level: 100%
12:28:46.564 > [I][sdcard.cpp:74] sdcardWriteData(): SD: writing data
...

No one packet outcoming to lora network, but the app is still running.

cyberman54, 2022-02-21:
Did you read and execute the note in the `m5core.h` file?

`#define LORA_IO1  GPIO_NUM_34 // **must be wired by you on PCB!`**

cyberman54, 2022-02-22:
Closed, until wiring was checked as noted in `m5core.h`.
Please re-open, if issue persists after wiring was checked.

closed 2022-02-22 by cyberman54