cyberman54/ESP32-Paxcounter #858: Does not work on ESP32 Paxcounter LoRa32 T3 V1.6.1

SmurfettesBrother, 2022-02-27:
The board came with firmware preinstalled. I compiled and flashed my customized version. Now the board hangs in a loop. I think this is a display problem. I'm using the latest version from github. Can someone help?

Thank you

`rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)<\r><\n>
configsip: 188777542, SPIWP:0xee<\r><\n>
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00<\r><\n>
mode:DIO, clock div:2<\r><\n>
load:0x3fff0018,len:4<\r><\n>
load:0x3fff001c,len:1044<\r><\n>
load:0x40078000,len:10124<\r><\n>
load:0x40080400,len:5828<\r><\n>
entry 0x400806a8<\r><\n>
[I][configmanager.cpp:105] loadConfig(): Loading device configuration from NVRAM...<\r><\n>
[I][configmanager.cpp:131] loadConfig(): Runtime configuration v3.0.3 loaded<\r><\n>
[I][main.cpp:134] setup(): Starting paxcounter_dab7695c v3.0.3 (runmode=0 / restarts=6)<\r><\n>
[I][main.cpp:150] setup(): This is ESP32 chip with 2 CPU cores, WiFi/BT/BLE, silicon revision 1, 4MB embedded Flash<\r><\n>
[I][main.cpp:152] setup(): Internal Total heap 258760, internal Free Heap 232876<\r><\n>
[I][main.cpp:158] setup(): ChipRevision 1, Cpu Freq 240, SDK Version v3.3.5-1-g85c43024c<\r><\n>
[I][main.cpp:160] setup(): Flash Size 4194304, Flash Speed 40000000<\r><\n>
[I][main.cpp:162] setup(): Wifi/BT software coexist version 1.2.0<\r><\n>
[I][main.cpp:166] setup(): IBM LMIC version 1.6.1468577746<\r><\n>
[I][main.cpp:171] setup(): Arduino LMIC version 4.1.1.0<\r><\n>
[I][lorawan.cpp:173] printKey(): DevEUI: 4C7525C36274FEFF<\r><\n>
[I][lorawan.cpp:173] printKey(): AppEUI: 70B3D50000000000<\r><\n>
[I][lorawan.cpp:173] printKey(): AppKey: 00000000000000000000000000000000<\r><\n>
[I][i2c.cpp:46] i2c_scan(): Starting I2C bus scan...<\r><\n>
[I][i2c.cpp:77] i2c_scan(): Device found at 0x3C, type = Unknown<\r><\n>
[I][i2c.cpp:80] i2c_scan(): 1 I2C device(s) found<\r><\n>
[E][display.cpp:69] dp_setup(): FAILURE in src/display.cpp:69<\r><\n>
/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/tasks.c:4826 (xTaskNotify)- assert failed!<\r><\n>
abort() was called at PC 0x400954d4 on core 1`

cyberman54, 2022-02-27:
which board did you select in paxcounter.ini?

cyberman54, 2022-02-27:
Note: your log shows that v3.0.3 is installed on the board. This is not the current version (3.2.0), but should work anyway.

SmurfettesBrother, 2022-03-01:
Strange, I got the repository with git clone...
Now, I got the zip file, but it still doesn't work. Now, I can't compile anymore. The error "src/i2c.cpp:15:30: error: 'class TwoWire' has no member named 'end'" comes up. No matter which board I choose.

The first time I tried different boards:
heltec.h
heltecv2.h
heltecv21.h
ttgov1.h
ttgov2.h
ttgov21old.h
ttgov21new.h

On my Heltec board this works without any problems

cyberman54, 2022-03-01:
Please check your toolchain, this issue looks like you're using outdated libraries.
[Wire.cpp](https://github.com/espressif/arduino-esp32/blob/master/libraries/Wire/src/Wire.cpp#L223)

zoernert, 2022-04-11:
Was was the trick that made issue 1 solved? (so not the i2c.cpp) but the display.cpp error message. I get the same and I think I've got the latest code (same device here - Lora32 T3): 
```
03:18:22.268 > [    42][I][configmanager.cpp:98] loadConfig(): Loading device configuration from NVRAM...
03:18:22.279 > [    44][I][configmanager.cpp:124] loadConfig(): Runtime configuration v3.2.1 loaded
03:18:22.290 > [    48][I][main.cpp:115] setup(): Starting paxcounter_cfc65d02 v3.2.1 (runmode=0 / restarts=46)
03:18:22.301 > [    55][I][main.cpp:116] setup(): code build date: 1649643373
03:18:22.301 > [    61][I][main.cpp:132] setup(): This is ESP32 chip with 2 CPU cores, WiFi/BT/BLE, silicon revision 3, 4MB external Flash
03:18:22.312 > [    72][I][main.cpp:134] setup(): Internal Total heap 225124, internal Free Heap 197132
03:18:22.323 > [    79][I][main.cpp:140] setup(): ChipRevision 3, Cpu Freq 240, SDK Version v4.4-367-gc29343eb94
03:18:22.334 > [    88][I][main.cpp:142] setup(): Flash Size 4194304, Flash Speed 40000000
03:18:22.334 > [    95][I][main.cpp:144] setup(): Wifi/BT software coexist version 1.2.0
03:18:22.345 > [   101][I][main.cpp:148] setup(): IBM LMIC version 1.6.1468577746
03:18:22.345 > [   107][I][main.cpp:153] setup(): Arduino LMIC version 4.1.1.0
03:18:22.374 > [   133][I][i2c.cpp:49] i2c_scan(): Starting I2C bus scan...
03:18:22.557 > [   321][I][i2c.cpp:77] i2c_scan(): Device found at 0x34, type = AXP192
03:18:22.558 > [   321][I][i2c.cpp:80] i2c_scan(): 1 I2C device(s) found
03:18:22.564 > [   321][E][Wire.cpp:159] begin(): Bus already started in Master Mode.
03:18:23.565 > [  1329][E][display.cpp:69] dp_setup(): FAILURE in src/display.cpp:69
```

tried with different board configs too (as @SmurfettesBrother did).

cyberman54, 2022-04-11:
Your log shows board has AXP192 chip, so probably is a T-Beam, not T3. Sure you're using the right .hal file?

zoernert, 2022-04-11:
> Sure you're using the right .hal file?

Ups... sorry! - I was googling - checking, trying, reading ... and I think I missed one big thing: `clean` before `build & monitor`- However the display.cpp error misleaded my focus.

bordeux, 2022-04-25:
Also having this same issue:
```
20:31:12.088 > ets Jun  8 2016 00:22:57
20:31:12.088 > 
20:31:12.088 > rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
20:31:12.088 > configsip: 188777542, SPIWP:0xee
20:31:12.088 > clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
20:31:12.088 > mode:DIO, clock div:2
20:31:12.088 > load:0x3fff0018,len:4
20:31:12.088 > load:0x3fff001c,len:1044
20:31:12.088 > load:0x40078000,len:10124
20:31:12.088 > load:0x40080400,len:5828
20:31:12.088 > entry 0x400806a8
20:31:12.450 > [I][configmanager.cpp:98] loadConfig(): Loading device configuration from NVRAM...
20:31:12.461 > [I][configmanager.cpp:124] loadConfig(): Runtime configuration v3.2.0 loaded
20:31:12.466 > [I][main.cpp:120] setup(): Starting paxcounter_1c5fc883 v3.2.0 (runmode=0 / restarts=8)
20:31:12.472 > [I][main.cpp:121] setup(): code build date: 1650915012
20:31:12.477 > [I][main.cpp:137] setup(): This is ESP32 chip with 2 CPU cores, WiFi/BT/BLE, silicon revision 1, 4MB embedded Flash
20:31:12.488 > [I][main.cpp:139] setup(): Internal Total heap 253368, internal Free Heap 222920
20:31:12.494 > [I][main.cpp:145] setup(): ChipRevision 1, Cpu Freq 240, SDK Version v3.3.5-1-g85c43024c
20:31:12.505 > [I][main.cpp:147] setup(): Flash Size 4194304, Flash Speed 40000000
20:31:12.510 > [I][main.cpp:149] setup(): Wifi/BT software coexist version 1.2.0
20:31:12.516 > [I][main.cpp:153] setup(): IBM LMIC version 1.6.1468577746
20:31:12.522 > [I][main.cpp:158] setup(): Arduino LMIC version 4.1.1.0
20:31:12.527 > [I][lorawan.cpp:173] printKey(): AppEUI: **removed**
20:31:12.533 > [I][lorawan.cpp:173] printKey(): DevEUI: **removed**
20:31:12.538 > [I][lorawan.cpp:173] printKey(): AppKey: **removed**
20:31:12.544 > [I][i2c.cpp:46] i2c_scan(): Starting I2C bus scan...
20:31:12.655 > [I][i2c.cpp:77] i2c_scan(): Device found at 0x3C, type = Unknown
20:31:12.666 > [I][i2c.cpp:80] i2c_scan(): 1 I2C device(s) found
20:31:12.666 > [E][display.cpp:69] dp_setup(): FAILURE in src/display.cpp:69
20:31:12.672 > /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/tasks.c:4826 (xTaskNotify)- assert failed!
20:31:12.688 > abort() was called at PC 0x400955c0 on core 1
20:31:12.688 > 
20:31:12.688 > ELF file SHA256: 0000000000000000
20:31:12.694 > 
20:31:12.694 > Backtrace: 0x40091cd0:0x3ffd02a0 0x40091f4d:0x3ffd02c0 0x400955c0:0x3ffd02e0 0x400d3e97:0x3ffd0300 0x400d3407:0x3ffd0320 0x400d35fa:0x3ffd0350 0x400d51f3:0x3ffd03a0 0x400ed446:0x3ffd0480 0x40092f5e:0x3ffd04a0
20:31:12.880 >   #0  0x40091cd0:0x3ffd02a0 in invoke_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c:715
20:31:12.880 >   #1  0x40091f4d:0x3ffd02c0 in abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c:715
20:31:12.880 >   #2  0x400955c0:0x3ffd02e0 in xTaskNotify at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/tasks.c:4826 (discriminator 1)
20:31:12.880 >   #3  0x400d3e97:0x3ffd0300 in mask_user_IRQ() at src/irqhandler.cpp:114
20:31:12.880 >   #4  0x400d3407:0x3ffd0320 in dp_setup(int) at src/display.cpp:707
20:31:12.880 >   #5  0x400d35fa:0x3ffd0350 in dp_init(bool) at src/display.cpp:707
20:31:12.880 >   #6  0x400d51f3:0x3ffd03a0 in setup() at src/main.cpp:196
20:31:12.880 >   #7  0x400ed446:0x3ffd0480 in loopTask(void*) at /Users/noname/.platformio/packages/framework-arduinoespressif32/cores/esp32/main.cpp:18
20:31:12.880 >   #8  0x40092f5e:0x3ffd04a0 in vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c:355 (discriminator 1)
20:31:12.880 > 
```

My board is `TTGO LoRa32 V2.1` 
[image]

And platformio.ini setup is

```; PlatformIO Project Configuration File
;
; Please visit documentation for the other options and examples
; http://docs.platformio.org/page/projectconf.html

; ---> SELECT THE TARGET PLATFORM HERE! <---
[board]
;halfile = generic.h
;halfile = ebox.h
;halfile = eboxtube.h
;halfile = ecopower.h
;halfile = heltec.h
;halfile = heltecv2.h
;halfile = heltecv21.h
;halfile = ttgov1.h
;halfile = ttgov2.h
[... 98 lines cut ...]
board_build.mcu = esp32
board_build.partitions = min_spiffs.csv
upload_speed = ${common.upload_speed}
upload_port = /dev/cu.wchusbserial54250257671
platform = ${common.platform_espressif32}
lib_deps = ${common.lib_deps_all}
build_flags = ${common.build_flags_all}
upload_protocol = ${common.upload_protocol}
extra_scripts = ${common.extra_scripts}
monitor_speed = ${common.monitor_speed}
monitor_filters = time, esp32_exception_decoder, default

[env:ota]
upload_protocol = custom

[env:usb]
upload_protocol = esptool

[env:dev]
upload_protocol = esptool
build_type = debug
platform = https://github.com/platformio/platform-espressif32.git#feature/arduino-upstream
```

The issue is on 3.2 release downloaded as zip folder & on master branch too.  What are the best way to debug it and find the problem?

cyberman54, 2022-04-25:
I can confirm this. I have reproduced this on a TTGO T3 v1.6.1 board.
The issue is caused by the used display library, and needs to be further tracked down there.
I opened issued there:
https://github.com/bitbank2/OneBitDisplay/issues/29

cyberman54, 2022-04-25:
Meanwhile you can try to upgrade to Arduino 2.x and use the latest version of onebitdisplay library by editing following two lines in platformio.ini:

bordeux, 2022-04-25:
I think it can be i2c bus issue or issue with Wire library (but less likely)
I tested master branch of OneBitDisplay, like you purposed, but problem is this same. 
So then i decided to downgrade this package to version 1.10 and 1.9.1 - nothing changed - problem still persist.
Before each change of version i clicked "clean all" on VS platform.io plugin , to be sure the library is this same like i defined in config.

Hardware issue i think i can exclude, because stock firmware works well - version 3.0.3

> Meanwhile you can try* to upgrade to Arduino 2.x 
With this im not sure if did it correctly, because new version for mac os is separated application, not overwriting  older version 1.8. So in this case i removed completely 1.8 version from the system.

Let me know how else i can help, so i will try to give support.

cyberman54, 2022-04-25:
To wrap up, it seems two different phenomens can be seen:

1) display is not detected by onebitdisplay lib, causing assert failure in display.cpp of paxcounter

2) display is detected and works, but showing only two lines and with wrong text placement

I currently have no idea what's going wrong here.

cyberman54, 2022-04-26:
This should now be fixed by this [commit](https://github.com/cyberman54/ESP32-Paxcounter/pull/867/commits/70a963ae6d8a8db806106096c755acd5f4cb4633)

Solution was to force onebitdisplay lib to use Wire for i2c bus, not hardware bitbanging.

@bordeux @SmurfettesBrother could you please retest and report here? Thanks.

bordeux, 2022-04-26:
@cyberman54 checked latest master branch with your changes, but problem still appears. 

I will try to debug it by myself - like you said the problem can be with onebit display library or wrong parameters. I will let know about my findings.

cyberman54, 2022-04-26:
@bordeux i double checked the patch with three boards (TTGO T3 1.6.1, Olimex PoE, T-Fox) - all with working display. Strange.

bordeux, 2022-04-26:
@cyberman54 yeah, that is strange. I still debugging it, i just created simple code bellow, to test OneBitDisplay library.
There are 2 functions:
- ssd1306lib - works fine like expected
- onebitlib - not working, unable to find display (so this same issue like with pax counter)
Tested with different parameters, with and without i2c address, different speeds and any good results from my side.

How possible is to share to me a binary from your TTGO T3 1.6.1, without of course lorawan and other credentials? Because it for you it works there are 2 possibilities for me:
- the problem is between my chair and my monitor (highly possible) 
- problem is in TTGO T3 1.6.1 itself - maybe different, cheaper parts.

```c
#include <Arduino.h>
#include <OneBitDisplay.h>
#include <ArduinoLog.h>
#include <Wire.h>
#include <Adafruit_I2CDevice.h>
#include <Adafruit_GFX.h>
#include <Adafruit_SSD1306.h>

#define SDA_PIN 21
#define SCL_PIN 22

#define OLED_ADDR 0x3c

#define FLIP180 0
// don't invert the display
#define INVERT 0
// Bit-Bang the I2C bus
#define USE_HW_I2C 0

#define RESET_PIN -1

#define SCREEN_WIDTH 128 // OLED display width, in pixels
#define SCREEN_HEIGHT 64 // OLED display height, in pixels

void onebitlib()
{
  OBDISP obd;
  Log.verboseln("Start testing using OneBitDisplay library");

  int rc = obdI2CInit(&obd, OLED_128x64, OLED_ADDR, FLIP180, INVERT, 0, SDA_PIN, SCL_PIN, RESET_PIN, 400000L); // use standard I2C bus at 400Khz
  if (rc != OLED_NOT_FOUND)
  {
    Log.verboseln("Oled found!");
    obdFill(&obd, 0, 1);
    obdWriteString(&obd, 0, 0, 0, "OneBitScreen!", FONT_8x8, 0, 1);
  }
  else
  {
    Log.verboseln("Oled NOT found!");
  }
}

void ssd1306lib()
{
  Adafruit_SSD1306 display(SCREEN_WIDTH, SCREEN_HEIGHT, &Wire, RESET_PIN);
  Log.verboseln("Start testing using SSD1306 library");
  if (!display.begin(SSD1306_SWITCHCAPVCC, OLED_ADDR))
  {
    Log.verboseln(F("SSD1306 allocation failed"));
  }
  else
  {
    Log.verboseln("Oled using SSD1306 lib works!");
    display.display();
    display.clearDisplay();
    display.setTextSize(1);            
    display.setTextColor(SSD1306_WHITE); 
    display.setCursor(0, 0);     
    display.cp437(true);
    display.println(F("Lorem ipsum"));
    display.display();
    Log.verboseln("Finished rendering!");
  }
}

void setup()
{
  Serial.begin(9600);
  Log.begin(LOG_LEVEL_VERBOSE, &Serial);
  Log.verboseln("Starting application");
  delay(1000);
  //ssd1306lib();
  onebitlib();
}

void loop() {}
```

and platform io
```ini
[platformio]
default_envs = ttgo-lora32-v21

[env]
lib_deps = 
	andeepmistry/arduino-LoRa @ ^0.8.0
	bitbank2/OneBitDisplay @ ^1.11.0
	thijse/ArduinoLog@^1.1.1
        adafruit/Adafruit SSD1306@^2.5.3
build_flags = 
	-D VERSION=1.2.3
	-D DEBUG=1
lib_deps_builtin = 
	SPI
	Wire
upload_port = /dev/cu.wchusbserial54250257671

[env:ttgo-lora32-v21]
platform = espressif32
board = ttgo-lora32-v21
framework = arduino
```

cyberman54, 2022-04-26:
@bordeux before paxcounter initializes the display, it does a i2c scan with bitbang lib by same author. I made the modification here. Please look in i2c.cpp and add that piece of code to your test code. It made the issue disappearing here, but i yet cannot explain why.

bordeux, 2022-04-26:
Just added
```c
void onebitlib()
{
  BBI2C bbi2c;

  memset(&bbi2c, 0, sizeof(bbi2c));
  bbi2c.bWire = 1;
  bbi2c.iSDA = SDA_PIN;
  bbi2c.iSCL = SCL_PIN;
  I2CInit(&bbi2c, 100000L); // Scan at 100KHz low speed
  delay(100);

  OBDISP obd;
  Log.verboseln("Start testing using OneBitDisplay library");

  int rc = obdI2CInit(&obd, OLED_128x64, OLED_ADDR, FLIP180, INVERT, 0, SDA_PIN, SCL_PIN, RESET_PIN, 400000L); // use standard I2C bus at 400Khz
  if (rc != OLED_NOT_FOUND)
  {
    Log.verboseln("Oled found!");
    obdFill(&obd, 0, 1);
    obdWriteString(&obd, 0, 0, 0, "OneBitScreen!", FONT_8x8, 0, 1);
  }
  else
  {
    Log.verboseln("Oled NOT found!");
  }
}
```

But OneBitDisplay is using own instance of BBI2C, so modifications on i2c scanner part from my opinion should not affect onebitdisplay. 
I2C scanner on paxcounter works fine, even before modifications, correctly finding 0x3c device

mietse, 2022-04-26:
Change this file and now my display is working. 

file: display.cpp

#if (HAS_DISPLAY) == 1 // I2C OLED

  int rc = obdI2CInit(&ssoled, OLED_TYPE, OLED_ADDR, MY_DISPLAY_FLIP,
                      MY_DISPLAY_INVERT, USE_HW_I2C, MY_DISPLAY_SDA,
                      MY_DISPLAY_SCL, MY_DISPLAY_RST,
                      OLED_FREQUENCY); // use standard I2C bus at 400Khz
  //_ASSERT(rc != OLED_NOT_FOUND);  //<< blockout this line >>

  // set display buffer
  obdSetBackBuffer(&ssoled, displaybuf);
  obdSetTextWrap(&ssoled, true);
  dp_font = MY_FONT_NORMAL;

bordeux, 2022-04-26:
@mietse lol :D Yeah, this works! Funny, OneBit display returning not found status, but it found :)

cyberman54, 2022-04-27:
of course we can dismiss the assert. But this does not solve the problem. The question is, why is the display not found?

If is was found before, it is evident that some change, either in display lib or wire driver of framework is the root cause, since there were no changes in paxcounter.

commit 55bc8563 (2022-04-27): fix #858: temporary disabled oled assert
closed 2022-04-27 by cyberman54