cyberman54/ESP32-Paxcounter #859: Cant turn off sleep cycle (without factory reset)

JMyrng, 2022-02-27:
Hi there, first of all thank you for this great open source project!

Board: TTGO T-Beam T22_V1.1 with NEO-6M (ttgobeam10.h with disabled display)
Paxcounter V3.1.0

I'm using paxcounter as a TTN mapper, so I've configured
```
#define BLECOUNTER                      0
#define WIFICOUNTER                     0
```

At first I've started with a sleep cycle, but then I can't get a fix before going to sleep. 

So I've changed to 
`#define SLEEPCYCLE                      0`
but the T-Beam is going to sleep anyway. Is this intentional?

Then I've tried setting SLEEPCYCLE with a TTN downlink. 
Sending `19 00 1E` works as it should and gives me the following log:
```
16:09:20.190 > [I][lorawan.cpp:417] myRxCallback(): Received 3 byte(s) of payload on port 2
16:09:20.201 > [I][rcommand.cpp:93] set_sleepcycle(): Remote command: set sleep cycle to 300 seconds
```
But sending `19 00 00` gives:
```
16:09:42.645 > [I][lorawan.cpp:417] myRxCallback(): Received 3 byte(s) of payload on port 2
```
I think this is intentional based on the following lines: 
https://github.com/cyberman54/ESP32-Paxcounter/blob/7f6cea2e0222c4bc950b38c4d64bb2706cb84cfc/src/rcommand.cpp#L88-L89

In the end I've I got it working with a reset to factory settings with the downlink `09 02`, but I wanted to clarify whether this behavior is intentional. 

Best regards
Jonas

cyberman54, 2022-02-27:
The #defines are for factor default setting. If you change them and upload a new build, you must clear NVRAM, otherwise the device loads the previous runtime settings from NVRAM.

To change runtime settings use rcommand. To make new runtime setting persistent, add rcommand save.

closed 2022-02-27 by cyberman54

JMyrng, 2022-02-27:
Thanks for the quick answer, so it is intentional that I cant turn off sleep via downlink (`19 00 00`)?

cyberman54, 2022-02-27:
you're right, this is a bug.  Fixed.