cyberman54/ESP32-Paxcounter #866: SDS011 sensor + TTGOV1.6 LoRa issue

FabTangi, 2022-04-07:
Hello, 
Using this code with an SDS011 sensor and a TTGOV1.6, I have an issue. 
Lora handshake and SDS011 data are OK. 
1 frame is sent, and the the sensor continue, but no more frames are sent. 
There is also an error just after sending the Lora frame : 

_22:22:47.075 > [ 43665][I][sds011read.cpp:37] sds011_loop(): SDS011: pm25: 4.10, pm10: 14.20
22:23:14.917 > [ 71507][I][lorawan.cpp:224] lora_send(): 6 byte(s) sent to LORA
22:23:19.958 > FAILURE
22:23:19.958 > .pio/libdeps/usb/MCCI LoRaWAN LMIC library/src/lmic/radio.c:1065
22:23:47.076 > [103665][I][sds011read.cpp:37] sds011_loop(): SDS011: pm25: 3.90, pm10: 13.10
22:24:47.076 > [163665][I][sds011read.cpp:37] sds011_loop(): SDS011: pm25: 4.00, pm10: 13.00_ 

Any idea ?

cyberman54, 2022-04-07:
To which pins did you connect the SDS011?

FabTangi, 2022-04-07:
Well I used pin 19 and 23 and added in hal file ttgov21new.h : 

// SDS011 dust sensor settings
#define HAS_SDS011 1 // use SDS011
// used pins on the ESP-side:
#define SDS_TX 19     // connect to RX on the SDS011
#define SDS_RX 23     // connect to TX on the SDS011

cyberman54, 2022-04-07:
Pins 19+23 are used on TTGO v1.6 for SPI interface of Lora chip. Thus, you cannot use these pins for serial interface of SDS011.

Check your wiring.

cyberman54, 2022-04-08:
Use IO12 and IO35.

closed 2022-04-09 by cyberman54