cyberman54/ESP32-Paxcounter #870: possible dependency issue in i2c.cpp

doebi, 2022-06-01:
Fresh clone with any board results in this error:

```
src/i2c.cpp: In function 'void i2c_scan()':
src/i2c.cpp:75:46: error: too few arguments to function 'int I2CDiscoverDevice(BBI2C*, uint8_t, uint32_t*)'
         iDevice = I2CDiscoverDevice(&bbi2c, i);
                                              ^
In file included from include/i2c.h:5,
                 from src/i2c.cpp:3:
.pio/libdeps/usb/BitBang_I2C/src/BitBang_I2C.h:151:5: note: declared here
 int I2CDiscoverDevice(BBI2C *pI2C, uint8_t i, uint32_t *pCapabilities);
     ^~~~~~~~~~~~~~~~~
Compiling .pio/build/usb/src/power.cpp.o
*** [.pio/build/usb/src/i2c.cpp.o] Error 1
```

Is this a dependency issue with the BitBang library?

doebi, 2022-06-01:
I was able to workaround by applying:

```diff
diff --git a/src/i2c.cpp b/src/i2c.cpp
index cc13910..e7bc51f 100644
--- a/src/i2c.cpp
+++ b/src/i2c.cpp
@@ -72,7 +72,7 @@ void i2c_scan(void) {
       if (map[i >> 3] & (1 << (i & 7))) // device found
       {
         iCount++;
-        iDevice = I2CDiscoverDevice(&bbi2c, i);
+        iDevice = I2CDiscoverDevice(&bbi2c, i, 0);
         ESP_LOGI(TAG, "Device found at 0x%X, type = %s", i,
                  szNames[iDevice]); // show the device name as a string
       }
```

But I guess, there is a better resolution.

loratrak, 2022-06-01:
I had the same issue some days ago, did a downgraded in BitBang and it works!!

cyberman54, 2022-06-03:
This should be fixed now.
https://github.com/cyberman54/ESP32-Paxcounter/commit/30b56c8bcffd654885735ecf97f392369ae09308

closed 2022-06-03 by cyberman54