cyberman54/ESP32-Paxcounter #871: what if you don't want to use paxexpress

heaversm, 2022-06-05:
Getting the error `KeyError: 'PAXEXPRESS_USER':`

```
Parsing OTA keys from /Users/mheavers/Documents/Sites/Raphael/paxcounter/ESP32-Paxcounter/src/ota.conf
Target board: esp32dev @ 921600bps
Partition table: min_spiffs.csv
KeyError: 'PAXEXPRESS_USER':
  File "/Users/mheavers/.platformio/penv/lib/python3.9/site-packages/platformio/builder/main.py", line 182:
    env.SConscript(item, exports="env")
  File "/Users/mheavers/.platformio/packages/tool-scons/scons-local-4.3.0/SCons/Script/SConscript.py", line 597:
    return _SConscript(self.fs, *files, **subst_kw)
  File "/Users/mheavers/.platformio/packages/tool-scons/scons-local-4.3.0/SCons/Script/SConscript.py", line 285:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "/Users/mheavers/Documents/Sites/Raphael/paxcounter/ESP32-Paxcounter/build.py", line 86:
    user = mykeys["PAXEXPRESS_USER"]
==================================================================== [FAILED] Took 0.62 seconds ====================================================================
```

These are commented out in my `ota.conf`:

```
#[paxexpress]
#PAXEXPRESS_USER = MyPaxexpressUser
#PAXEXPRESS_REPO = MyPaxexpressRepo
#PAXEXPRESS_API_TOKEN = 3894a7a51d70c6523c1b7479261c34845ebf7878
```

cyberman54, 2022-06-06:
There is no need to comment out these lines. You leave them untouched.

closed 2022-06-06 by cyberman54

heaversm, 2022-06-06:
@cyberman54  - if I do this, I get:

`Failed to submit package: 401 Client Error: Unauthorized for url: https://pax.express/content/MyPaxexpressUser/MyPaxexpressRepo/ttgov21new/3.2.1/firmware_ttgov21new_v3.2.1.bin`