cyberman54/ESP32-Paxcounter #875: PaxCounter's signal strengh very low

paddlix, 2022-07-01:
Hello,
I have some issues with the range of my PaxCounter and I wondered, whether this could be a software or a hardware problem:
I can only achieve ranges up to 5m and if the PaxCounter and the gateway are right next to each other, the gateway receives the PaxCounter's messages only with a signal strength less than -70dbm, even though it's txpower is set to the maximum of +14dbm. Other LoRaWAN devices reach a signal strength up to -5dbm in the same scenario.
The PaxCounter is a LILYGO TTGO ESP32 PaxCounter. It is set for use in the EU.

Could this be a hardware failure or can I fix it with some changes in the settings of the PaxCounter?

Thanks in advice!

cyberman54, 2022-07-01:
What type of gateway and lorawan network provider or server are you using? Where did you buy the board? Please post picture of board and antenna.

cyberman54, 2022-07-01:
And: did you make sure that the board is using the gateway you expect, not another one?

paddlix, 2022-07-01:
I am using a kerlink Wirnet iStation 868 connected to the European server of the TheThingsNetwork, where the PaxCounter is registered as an EndDevice. I am 100% sure that the PaxCounter is using my gateway, because there is no other gateway in my region and if I turn off the gateway, the PaxCounter doesn't join any network. The gateway EUI does fit too.

Here are two pictures of the board with it's antenna. One shows the front, the other one the back of the PaxCounter. Although the board says "433/470Mhz" I am transmitting data with 868Mhz. I also tried an antenna twice as long as this one: same issue. This antenna is 4.1cm long.

[image]
[image]

cyberman54, 2022-07-01:
you already hit the nail: wrong version for 433. MHz, not 868 MHz.

closed 2022-07-01 by cyberman54