cyberman54/ESP32-Paxcounter #877: utilizing the pax count from senddata.cpp over serial in a connected node.js web app

heaversm, 2022-07-08:
I have a node.js web app that can read from the serial port of the computer [using this library](https://github.com/serialport/node-serialport).

I want to be able to plug in my Lilygo LORA32  (TTGO v3 v1.6.1) to USB on the computer and ingest the `pax` value to use in my code to drive a web visualization.

I can see that this is being logged in the serial monitor of VSCODE via : `senddata.cpp:91` :

`sendData() Sending count results pax=7 wifi=1 ble=6`

how would I set something up such that I could listen expressly for the `pax` value coming from the serial monitor and store that to a variable in node.js? I don't need any actual node.js code, just need to know how to set up the serial output in the esp32-paxcounter code (presumably in `senddata.cpp`?

Thanks for the help!

cyberman54, 2022-07-09:
don't do it in senddata.cpp.
Make a new sendserisl.cpp using spisend.cpp as template.

heaversm, 2022-07-09:
I don't see spisend.cpp in this repo.  Where would I find that template?

cyberman54, 2022-07-09:
https://github.com/cyberman54/ESP32-Paxcounter/blob/master/src/spislave.cpp

cyberman54, 2022-07-09:
Here is a How To:

You need
1) a messagebuffer, to store the payload
2) a queue, as buffer for the serial data 
3) a protocol, as needed for your application
4) a serial port

1) + 2) see spislave.cpp (change the SPI transmit calls by serial port calls)
3) consider overhead and checksum, e.g. transfer the payload as byte array or UTF8 string, e.g. comma separated string with checksum, as used in NMEA
4) consider if you want to use the USB port, then you probably want to mute/reroute all console output; or using other GPIO

heaversm, 2022-07-10:
Really wish I had the knowledge and expertise to understand how to do this - most of this sounds pretty outside of my wheelhouse (web development / basic arduino / pi). Think I'll need to find a way to hire someone to figure this out for me. Appreciate your help!

cyberman54, 2022-07-12:
Do you have a concept for No. 3?

heaversm, 2022-07-13:
Well - all my application needs to do is ingest the total number of devices detected (ble + wifi). If I can get to the point where I can read that value over serial, I am good to go. I tried using the [packed decoder](https://github.com/cyberman54/ESP32-Paxcounter/blob/master/src/TTN/packed_decoder.js) from your library that gets put into The Things Network. When I put it into my own code and passing the serial output, it returns a wifi / bluetooth count - but I'm not convinced that it is returning accurate values. 

This is what I have:

```
const parser = serialPort.pipe(
  new ReadlineParser({
    delimiter: "\n",
  })
);

const decodedData = Decoder(data);

where `Decoder` uses:

```
function Decoder(bytes) {
  var decoded = {};

  if (bytes.length === 0) {
    return {};
  }

  // only wifi counter data, no gps
  if (bytes.length === 2) {
    return decode(bytes, [uint16], ["wifi"]);
  }
  // wifi + ble counter data, no gps
  if (bytes.length === 4) {
    return decode(bytes, [uint16, uint16], ["wifi", "ble"]);
  }
}
```

cyberman54, 2022-07-14:
@heaversm 

 A simple approach:

TTGO boards have ESP32 without USB, but U0RXD/U0TXD of ESP32 connected to USB chip on board, which is normally used to flash the board and for serial console. For your approach you could "misuse" the serial console. That's simple, try this:

1. Mute device output on the serial console (= USB port of TTGO board), as far as possible
2. Add prints of payload to serial console, as needed
3. Switch off Lora, as it's not needed in this use case

How To:

1)
in _paxcounter.conf_:
`#define VERBOSE 0 `

in _platformio.ini_:
`debug_level = 0`

2)
in _senddata.cpp_:
add this line
`ets_printf("$PAX,%d,%d,%d\r\n", count.pax, count.wifi_count, count.ble_count);`
after this line
`ESP_LOGD(TAG, "Sending count results: pax=%d / wifi=%d / ble=%d", count.pax, count.wifi_count, count.ble_count);`

3)
in halfile, e.g. _ttgov21new.h_:
#define HAS_LORA  0

Then you should get output like this on USB:

```
$PAX,22,5,17
$PAX,21,1,20
$PAX,21,1,20
$PAX,21,0,21
```

cyberman54, 2022-07-14:
reopened as enhancement idea with new titel

https://docs.espressif.com/projects/esp-idf/en/latest/esp32s2/api-reference/peripherals/usb_device.html

heaversm, 2022-07-15:
It worked! [!celebration](https://media.giphy.com/media/4TtoGMqgSMikCW4mEh/giphy.gif)
[image]

cyberman54, 2022-07-16:
ESP_LOGD is based on ets_printf, but considers log level. Which we set to NONE. To mute all console output for your use case.

cyberman54, 2022-07-20:
@heaversm so, could you proceed with you project?