cyberman54/ESP32-Paxcounter #878: Add support for reporting over the Helium network

DeflateAwning, 2022-07-16:


cyberman54, 2022-07-17:
Paxcounter is based in LMIC stack which is LoRaWAN 1.0.3 compliant. Thus, it can be used with every LoRaWAN network which is supporting this standard. Check with your network provider.

closed 2022-07-17 by cyberman54

cyberman54, 2022-07-18:
PS: i tested to registrate a paxcounter in Helium network. Worked as expected. But burns your $HNT wallet pretty fast since it reports all 60 seconds...

DeflateAwning, 2022-07-18:
I'm wondering if you'd be able to share the tutorial or something? I think there might be a good opportunity to add a tutorial or explanation to this Git repo for anyone looking to get started!

cyberman54, 2022-07-18:
it's just registrating a lorawan device in a lorawan network. Nothing special here, just enter keys and go. A tutorial would be just about lorawan basics. There are some, so why reinvent this wheel here again?