cyberman54/ESP32-Paxcounter #882: screen not working on heltec v2,1

luigi20000, 2022-07-29:
hi my screen isn't working on my heltec v2,1 he does not find the screen while on another program it works
 I have filled in the pin numbers directly but this doesn't change anything
here is the terminal output
20:09:50.770 > Rebooting...
20:09:50.797 >
20:09:50.797 > rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
20:09:50.797 > configsip: 0, SPIWP:0xee
20:09:50.797 > clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
20:09:50.797 > mode:DIO, clock div:2
20:09:50.797 > load:0x3fff0030,len:1184
20:09:50.797 > load:0x40078000,len:12784
20:09:50.797 > load:0x40080400,len:3032
20:09:50.797 > entry 0x400805e4
20:09:51.377 > [    41][I][configmanager.cpp:98] loadConfig(): [src/configmanager.cpp] Loading device configuration from NVRAM...
20:09:51.377 > [    43][I][configmanager.cpp:124] loadConfig(): [src/configmanager.cpp] Runtime configuration v3.3.0 loaded
20:09:51.377 > [    51][I][main.cpp:115] setup(): [src/main.cpp] Starting paxcounter_2ec82ecd v3.3.0 (runmode=0 / restarts=7)
20:09:51.377 > [    59][I][main.cpp:116] setup(): [src/main.cpp] code build date: 1659117498
20:09:51.377 > [    66][I][main.cpp:132] setup(): [src/main.cpp] This is ESP32 chip with 2 CPU cores, WiFi/BT/BLE, silicon revision 1, 8MB external Flash
20:09:51.421 > [    78][I][main.cpp:134] setup(): [src/main.cpp] Internal Total heap 227472, internal Free Heap 199764
20:09:51.421 > [    87][I][main.cpp:140] setup(): [src/main.cpp] ChipRevision 1, Cpu Freq 240, SDK Version v4.4.1-1-gb8050b365e
20:09:51.421 > [    97][I][main.cpp:142] setup(): [src/main.cpp] Flash Size 8388608, Flash Speed 40000000
20:09:51.421 > [   105][I][main.cpp:144] setup(): [src/main.cpp] Wifi/BT software coexist version 1.2.0
20:09:51.421 > [   113][I][main.cpp:148] setup(): [src/main.cpp] IBM LMIC version 1.6.1468577746
20:09:51.466 > [   120][I][main.cpp:153] setup(): [src/main.cpp] Arduino LMIC version 4.1.1.0
20:09:51.466 > [   127][I][lorawan.cpp:173] printKey(): [src/lorawan.cpp] AppEUI: 70B3D50000000000
20:09:51.466 > [   135][I][lorawan.cpp:173] printKey(): [src/lorawan.cpp] DevEUI: C8C9A3C8C928FEFF
20:09:51.466 > [   142][I][lorawan.cpp:173] printKey(): [src/lorawan.cpp] AppKey: 00000000000000000000000000000000
20:09:51.466 > [   152][I][i2c.cpp:41] i2c_scan(): [src/i2c.cpp] Starting I2C bus scan...
20:09:51.466 > [   157][W][Wire.cpp:204] begin(): Bus already started in Master Mode.
20:09:51.574 > [   282][I][i2c.cpp:75] i2c_scan(): [src/i2c.cpp] 0 I2C device(s) found
20:09:51.721 > [   282][W][Wire.cpp:204] begin(): Bus already started in Master Mode.
20:09:51.721 >
20:09:51.721 > assert failed: void dp_setup(int) display.cpp:63 (oled.allocBuffer())
20:09:51.721 >
20:09:51.721 >
20:09:51.721 > Backtrace:0x400841cd:0x3ffcf7900x40096951:0x3ffcf7b0 0x4009c911:0x3ffcf7d0 0x400d4026:0x3ffcf900 0x400d41d4:0x3ffcf920 0x400d5b27:0x3ffcf970 0x400f1c06:0x3ffcfa50
20:09:51.721 >

cyberman54, 2022-07-29:
Please state which hal file you've used.

luigi20000, 2022-07-29:
heltecv21.h

cyberman54, 2022-07-29:
I can reproduce this. Need to check this deeper, probably an issue with onebitdisplay lib.

cyberman54, 2022-08-06:
fixed.
https://github.com/cyberman54/ESP32-Paxcounter/pull/885

closed 2022-08-06 by cyberman54