cyberman54/ESP32-Paxcounter #895: Unable to build

t-huyeng, 2022-09-28:
I am not able to build the current master version with platformio + vscode. 

`Processing usb (framework: arduino; board: esp32dev; platform: espressif32@3.4.0)`

`.pio\libdeps\usb\ezTime\src\ezTime.cpp:10:22: fatal error: EEPROM.h: No such file or directory`

I can temporarily fix this by adding `#include "EEPROM.h"` to `main.h`.

&

`include/reset.h:5:31: fatal error: soc/reset_reasons.h: No such file or directory`

I can not find the `reset_reasons.h` file in the repo. 

Is there a known fix for me?

cyberman54, 2022-09-28:
You are using a very old version of paxcounter: 
Processing usb (framework: arduino; board: esp32dev; **platform: espressif32@3.4.0**)
The current version does not use ezTime any more.
Check you toolchain and make sure you're using the current version of this repository.

If you had cloned the repo before and updated the code, make sure to copy platformio_org.ini to platformio.ini

t-huyeng, 2022-09-28:
You are right, sorry about that. Used my old `platformio.ini` ...

closed 2022-09-28 by t-huyeng