cyberman54/ESP32-Paxcounter #921: BLE Scanner

Qmick01, 2022-11-24:
Hi, I am on build 3.4.3 and after flashing a few boards for BLE sensing only have found the BLE count is consistently 0.

I believe there may be an issue here and would appreciate any thoughts or a fix?

Paxcounter conf is below:

/ default settings for transmission of sensor data (first list = data on / second line = data off)
#define PAYLOADMASK                                                                             \
            ((GPS_DATA | MEMS_DATA | COUNT_DATA | SENSOR1_DATA | SENSOR2_DATA | SENSOR3_DATA | BATT_DATA) \
           & (~RESERVED_DATA))

// MAC sniffing parameters
#define BLECOUNTER                      1       // set to 0 if you do not want to install the BLE sniffer
#define WIFICOUNTER                     0       // set to 0 if you do not want to install the WIFI sniffer
#define RSSILIMIT                       0       // 0...-128, set to 0 if you do not want to filter signals

// BLE scan parameters
#define BLESCANTIME                     0       // [seconds] scan duration, 0 means infinite [default], see note below
#define BLESCANWINDOW                   80      // [milliseconds] scan window, see below, 3 .. 10240, default 80ms
#define BLESCANINTERVAL                 80      // [illiseconds] scan interval, see below, 3 .. 10240, default 80ms = 100% duty cycle

t-huyeng, 2022-11-24:
Can you provide logs from the board?

Qmick01, 2022-11-24:
SERIAL PORT DATA

09:32:11.849 >
09:32:11.849 > rst:0x5 (DEEPSLEEP_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
09:32:11.853 > configsip: 188777542, SPIWP:0xee
09:32:11.853 > clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
09:32:11.866 > mode:DIO, clock div:2
09:32:11.866 > load:0x3fff0030,len:1184
09:32:11.866 > load:0x40078000,len:13132
09:32:11.871 > load:0x40080400,len:3036
09:32:11.871 > entry 0x400805e4
09:32:11.930 > [    20][I][configmanager.cpp:96] loadConfig(): [src/configmanager.cpp] Loading device configuration from NVRAM...
09:32:11.937 > [    22][I][configmanager.cpp:122] loadConfig(): [src/configmanager.cpp] Runtime configuration v3.4.3 loaded
09:32:11.947 > [    30][I][reset.cpp:72] do_after_reset(): [src/reset.cpp] Time spent in deep sleep: 600097 ms
09:32:11.958 > [    38][I][main.cpp:115] setup(): [src/main.cpp] Starting paxcounter_7dd43646 v3.4.3 (runmode=2 / restarts=0)
09:32:11.965 > [    47][I][main.cpp:116] setup(): [src/main.cpp] code build date: 1669277974
09:32:11.975 > [    54][I][esp32-hal-i2c.c:75] i2cInit(): Initialising I2C Master: sda=21 scl=22 freq=100000
09:32:11.980 > [    63][I][power.cpp:194] calibrate_voltage(): [src/power.cpp] ADC characterization based on reference voltage stored in eFuse
09:32:11.992 > [    76][I][main.cpp:275] setup(): [src/main.cpp] Starting libpax...
09:32:11.997 > [    79][I][main.cpp:285] setup(): [src/main.cpp] WIFISCAN: off
09:32:12.004 > [    85][I][main.cpp:291] setup(): [src/main.cpp] BLESCAN: on
09:32:12.008 > [    91][I][blescan.cpp:203] start_BLE_scan(): [bluetooth] Initializing bluetooth scanner ...
09:32:12.636 > [   731][I][blescan.cpp:113] hci_cmd_send_ble_scan_start(): [bluetooth] BLE Scanning started
09:32:12.836 > [   931][I][blescan.cpp:264] start_BLE_scan(): [bluetooth] Bluetooth scanner started
09:32:12.848 > [   931][I][main.cpp:309] setup(): [src/main.cpp] Starting rcommand interpreter...
09:32:12.852 > [   935][I][rcommand.cpp:515] rcmd_init(): [src/rcommand.cpp] Rcommand send queue created, size 55 Bytes
09:32:12.865 > [   944][I][lorawan.cpp:248] lmic_init(): [src/lorawan.cpp] LORA send queue created, size 530 Bytes
09:32:12.874 > [   969][I][lorawan.cpp:537] LoadLMICFromRTC(): [src/lorawan.cpp] LMIC state loaded
09:32:12.886 > [   969][I][lorawan.cpp:293] lmic_init(): [src/lorawan.cpp] Already joined
09:32:12.891 > [   972][I][lorawan.cpp:297] lmic_init(): [src/lorawan.cpp] Starting LMIC...
09:32:12.897 > [   979][I][sdcard.cpp:95] sdcard_init(): [src/sdcard.cpp] looking for SD-card...
09:32:12.921 > E (1012) sdmmc_common: sdmmc_init_ocr: send_op_cond (1) returned 0x107
09:32:12.933 > E (1013) vfs_fat_sdmmc: sdmmc_card_init failed (0x107).
09:32:12.938 > [  1017][E][sdcard.cpp:156] sdcard_init(): [src/sdcard.cpp] SD-card not found (263)
09:32:12.944 > E (1020) wifi_init: Failed to deinit Wi-Fi driver (0x3001)
09:32:12.950 > [  1029][I][main.cpp:412] setup(): [src/main.cpp] Starting Interrupt Handler...
09:32:12.955 > [  1036][I][main.cpp:440] setup(): [src/main.cpp] Starting Timers...
09:32:12.961 > [  1042][I][timekeeper.cpp:391] time_init(): [src/timekeeper.cpp] Starting time pulse...
09:32:12.971 > [  1051][I][timekeeper.cpp:194] timepulse_init(): [src/timekeeper.cpp] Timepulse: internal (ESP32 hardware timer)
09:32:12.978 > [  1060][I][main.cpp:485] setup(): [src/main.cpp] Features: BATT OTA BLE LORA PACKED TIME
09:33:11.998 > [ 60096][I][reset.cpp:88] enter_deepsleep(): [src/reset.cpp] Preparing to sleep...
09:33:12.009 > [ 60096][I][blescan.cpp:277] stop_BLE_scan(): [bluetooth] Shutting down bluetooth scanner ...
09:33:12.015 > [ 60103][I][lorawan.cpp:219] lora_send(): [src/lorawan.cpp] 4 byte(s) sent to LORA
09:33:12.059 > [ 60157][I][blescan.cpp:286] stop_BLE_scan(): [bluetooth] Bluetooth scanner stopped
09:33:12.070 > [ 60157][I][reset.cpp:110] enter_deepsleep(): [src/reset.cpp] Waiting until send queues are empty...
09:33:18.200 > [ 66300][I][lorawan.cpp:219] lora_send(): [src/lorawan.cpp] 2 byte(s) sent to LORA
09:33:19.063 > [ 67162][I][reset.cpp:119] enter_deepsleep(): [src/reset.cpp] Waiting until LMIC is idle...
09:33:25.063 > [ 73162][I][lorawan.cpp:532] SaveLMICToRTC(): [src/lorawan.cpp] LMIC state saved
09:33:25.074 > [ 73163][I][reset.cpp:179] enter_deepsleep(): [src/reset.cpp] Going to sleep, good bye.

Qmick01, 2022-11-24:
I have just added to the thread.

cyberman54, 2022-11-25:
I can reproduce this. Will have to look deeper in this issue. Solution is probably to restructure code by removing compiler directives WIFICOUNTER and BLECOUNTER, and switch Wifi and BLE using runtime variables instead.

Qmick01, 2022-11-25:
Thank you.
Would you know where this became an issue?
I can report that version 3.3.2 does NOT have this bug and is working fine on sensors I have deployed.

cyberman54, 2022-11-25:
I compared 3.3.2 and 3.4.0, and did not find any relevant diffs in paxcounter code. But probably this could be causing this issue, is need to check further:

[image]

cyberman54, 2022-11-26:
This should be fixed now. Bumped new code as v3.4.4
/cc @Qmick01

Qmick01, 2022-11-26:
Hi,

I have been testing for 2 hours. 
I am now on 3.4.4 and have flashed two different versions of the same board, one a new type and one an older version I have stock of and used for many deployments.  I am still unable to get a BLE number reported back. 
BLE 0 is all I get in several locations where other boards report BLE number.
Also my tests revealed this build does not respond to an "80" on port 2 downlink.

cyberman54, 2022-11-26:
Then you probably use outdated code in your toolchain. Update Arduino Espressif32 framework and libraries of the project (i.e. libpax). Make sure that your settings in paxcounter.conf for WIFICOUNTER and BLECOUNTER are appropriate. Clear the NVRAM if board were flashed with other paxcounter build before.

Qmick01, 2022-11-26:
It's a brand new install of the code and the correct espressif framework
came down.

I presume you used the same board and the new build worked?

Qmick01, 2022-11-26:
Here's the debug info:

But I keep getting 0 BLE.

23:37:48.814 > rst:0x5 (DEEPSLEEP_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
23:37:48.822 > configsip: 188777542, SPIWP:0xee
23:37:48.822 >
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
23:37:48.828 > mode:DIO, clock div:2
23:37:48.828 > load:0x3fff0030,len:1184
23:37:48.828 > load:0x40078000,len:13132
23:37:48.833 > load:0x40080400,len:3036
23:37:48.833 > entry 0x400805e4
23:37:48.881 > [     7][D][esp32-hal-cpu.c:244] setCpuFrequencyMhz(): PLL:
480 / 2 = 240 Mhz, APB: 80000000 Hz
23:37:48.889 > [    21][I][configmanager.cpp:96] loadConfig():
[src/configmanager.cpp] Loading device configuration from NVRAM...
23:37:48.901 > [    23][I][configmanager.cpp:122] loadConfig():
[src/configmanager.cpp] Runtime configuration v3.4.4 loaded
23:37:48.912 > [    30][D][reset.cpp:51] do_after_reset(): [src/reset.cpp]
Timezone set to CET-1CEST,M3.4.0/2,M10.4.0/3
23:37:48.923 > [    39][I][reset.cpp:72] do_after_reset(): [src/reset.cpp]
Time spent in deep sleep: 10105 ms
23:37:48.928 > [    48][I][main.cpp:115] setup(): [src/main.cpp] Starting
paxcounter_531f6c7e v3.4.4 (runmode=2 / restarts=0)
23:37:48.940 > [    57][I][main.cpp:116] setup(): [src/main.cpp] code build
date: 1669491922
23:37:48.945 > [    64][I][esp32-hal-i2c.c:75] i2cInit(): Initialising I2C
Master: sda=21 scl=22 freq=100000
23:37:48.956 > [    73][I][power.cpp:194] calibrate_voltage():
[src/power.cpp] ADC characterization based on reference voltage stored in
eFuse
23:37:48.967 > [    87][I][main.cpp:274] setup(): [src/main.cpp] Starting
libpax...
23:37:48.973 > [    89][I][main.cpp:284] setup(): [src/main.cpp] WIFISCAN:
off
23:37:48.978 > [    95][I][main.cpp:290] setup(): [src/main.cpp] BLESCAN: on
23:37:48.984 > [   101][I][blescan.cpp:203] start_BLE_scan(): [bluetooth]
Initializing bluetooth scanner ...
23:37:49.620 > [   751][I][blescan.cpp:113] hci_cmd_send_ble_scan_start():
[bluetooth] BLE Scanning started
23:37:49.826 > [   952][I][blescan.cpp:264] start_BLE_scan(): [bluetooth]
Bluetooth scanner started
23:37:49.866 > [   952][I][main.cpp:300] setup(): [src/main.cpp] Starting
rcommand interpreter...
23:37:49.866 > [   956][I][rcommand.cpp:498] rcmd_init():
[src/rcommand.cpp] Rcommand send queue created, size 55 Bytes
23:37:49.866 > [   965][I][lorawan.cpp:248] lmic_init(): [src/lorawan.cpp]
LORA send queue created, size 530 Bytes
23:37:49.891 > [   990][I][lorawan.cpp:537] LoadLMICFromRTC():
[src/lorawan.cpp] LMIC state loaded
23:37:49.891 > [   990][I][lorawan.cpp:293] lmic_init(): [src/lorawan.cpp]
Already joined
23:37:49.891 > [   993][I][lorawan.cpp:297] lmic_init(): [src/lorawan.cpp]
Starting LMIC...
23:37:49.891 > [  1000][I][sdcard.cpp:95] sdcard_init(): [src/sdcard.cpp]
looking for SD-card...
23:37:49.906 > E (1033) sdmmc_common: sdmmc_init_ocr: send_op_cond (1)
returned 0x107
23:37:49.917 > E (1034) vfs_fat_sdmmc: sdmmc_card_init failed (0x107).
23:37:49.923 > [  1038][E][sdcard.cpp:156] sdcard_init(): [src/sdcard.cpp]
SD-card not found (263)
23:37:49.928 > [  1045][I][main.cpp:396] setup(): [src/main.cpp] Starting
Interrupt Handler...
23:37:49.934 > [  1052][I][main.cpp:424] setup(): [src/main.cpp] Starting
Timers...
23:37:49.939 > [  1058][I][timekeeper.cpp:391] time_init():
[src/timekeeper.cpp] Starting time pulse...
23:37:49.950 > [  1067][I][timekeeper.cpp:194] timepulse_init():
[src/timekeeper.cpp] Timepulse: internal (ESP32 hardware timer)
[... 35 lines cut ...]
23:38:55.661 > [ 66787][D][lorawan.cpp:400] myEventCallback():
[src/lorawan.cpp] RXSTART        1
23:38:55.698 > [ 66829][D][lorawan.cpp:400] myEventCallback():
[src/lorawan.cpp] TXCOMPLETE     1
23:38:56.409 > [ 67541][I][lorawan.cpp:219] lora_send(): [src/lorawan.cpp]
2 byte(s) sent to LORA
23:38:57.034 > [ 68161][I][reset.cpp:119] enter_deepsleep():
[src/reset.cpp] Waiting until LMIC is idle...
23:39:54.946 > [126073][D][lorawan.cpp:400] myEventCallback():
[src/lorawan.cpp] TXSTART
23:40:00.590 > [131722][D][lorawan.cpp:400] myEventCallback():
[src/lorawan.cpp] RXSTART
23:40:01.595 > [132722][D][lorawan.cpp:400] myEventCallback():
[src/lorawan.cpp] RXSTART
23:40:01.631 > [132764][D][lorawan.cpp:400] myEventCallback():
[src/lorawan.cpp] TXCOMPLETE
23:40:02.029 > [133161][I][lorawan.cpp:532] SaveLMICToRTC():
[src/lorawan.cpp] LMIC state saved
23:40:02.041 > [133162][I][reset.cpp:179] enter_deepsleep():
[src/reset.cpp] Going to sleep, good bye.
23:40:11.983 > ets Jun  8 2016 00:22:57
23:40:11.988 >
Michael Joseph
+44 (0) 7790928628
"In Velo, Veritas"

cyberman54, 2022-11-27:
please post platformio.ini and paxconter.conf of your build.

cyberman54, 2022-11-27:
Make sure you're using current libpax lib, look in the build processing notes:

Scanning dependencies...
Dependency Graph
|-- libpax @ 1.0.2+sha.2f774b5

Qmick01, 2022-11-27:
Thanks.

I have assumed a new install will automatically pull the right versions of
everything down and install but perhaps not.

Does the downlink work for you on version 3.4.4?

I couldn't get a response.

I will have a look at the version log later.

cyberman54, 2022-11-27:
I don't see any problems with downlinks. Everything is working, as it did before.

Qmick01, 2022-11-27:
Ok. Thanks for checking. I will do some more testing.

cyberman54, 2022-11-28:
@Qmick01 for debugging, you can use new feature logging system output to SD card

cyberman54, 2022-12-02:
_Received by mail from @Qmick01:_

I have just re-cloned paxcounter on a different machine. It does compile and everything looks great - but still no BLE count.

I need to flash some more boards this week and have been using a version from February as this works reliably for most things.

I attach the compile output - perhaps there is something in there I am not seeing?

Can not remove temporary directory `C:\Users\Mike\Documents\PlatformIO\Projects\3.4.4 new\ESP32-Paxcounter\.pio\build`. Please remove it manually to avoid build issues
Processing usb (framework: arduino; board: esp32dev; platform: espressif32@5.2.0)
---------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
Parsing hardware configuration from C:\Users\Mike\Documents\PlatformIO\Projects\3.4.4 new\ESP32-Paxcounter\src\hal\ttgov21new.h
Parsing LMIC configuration from C:\Users\Mike\Documents\PlatformIO\Projects\3.4.4 new\ESP32-Paxcounter\src\lmic_config.h
Parsing LORAWAN keys from C:\Users\Mike\Documents\PlatformIO\Projects\3.4.4 new\ESP32-Paxcounter\src\loraconf.h
Parsing OTA keys from C:\Users\Mike\Documents\PlatformIO\Projects\3.4.4 new\ESP32-Paxcounter\src\ota.conf
Target board: esp32dev @ 921600bps
Partition table: min_spiffs.csv
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32dev.html
PLATFORM: Espressif 32 (5.2.0) > Espressif ESP32 Dev Module
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-arduinoespressif32 @ 3.20005.220925 (2.0.5)
 - tool-esptoolpy @ 1.40201.0 (4.2.1)
 - tool-mkfatfs @ 2.0.1
 - tool-mklittlefs @ 1.203.210628 (2.3)
 - tool-mkspiffs @ 2.230.0 (2.30)
 - tool-openocd-esp32 @ 2.1100.20220706 (11.0)
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch3
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 54 compatible libraries
Scanning dependencies...
Dependency Graph
|-- libpax @ 1.0.2+sha.1402107
|-- RokkitHash @ 20151212.0.0+sha.8500494
|-- ArduinoJson @ 6.19.4
|-- RTC @ 2.3.5
|-- OneButton @ 2.0.3
|-- XPowersLib @ 0.1.4
|   |-- Wire @ 2.0.0
|-- MQTT @ 2.5.0
|-- MCCI LoRaWAN LMIC library @ 4.1.1
|   |-- SPI @ 2.0.0
|-- OneBitDisplay @ 2.2.1
|   |-- BitBang_I2C @ 2.2.1
|   |   |-- Wire @ 2.0.0
|   |-- SPI @ 2.0.0
|-- bb_spi_lcd @ 2.4.0
|   |-- Wire @ 2.0.0
|   |-- SPI @ 2.0.0
|-- QRCode @ 0.0.1
|-- FastLED @ 3.5.0
|   |-- SPI @ 2.0.0
|-- TinyGPSPlus @ 1.0.3
|-- Adafruit Unified Sensor @ 1.1.6
|-- Adafruit BME280 Library @ 2.2.2
|   |-- Adafruit BusIO @ 1.14.1
|   |   |-- Wire @ 2.0.0
|   |   |-- SPI @ 2.0.0
|   |-- Wire @ 2.0.0
|   |-- SPI @ 2.0.0
|   |-- Adafruit Unified Sensor @ 1.1.6
|-- Adafruit BMP085 Library @ 1.2.2
|   |-- Adafruit BusIO @ 1.14.1
|   |   |-- Wire @ 2.0.0
|   |   |-- SPI @ 2.0.0
|   |-- Wire @ 2.0.0
|-- BSEC Software Library @ 1.6.1480
|   |-- SPI @ 2.0.0
[... 139 lines cut ...]
Writing at 0x000fabc8... (74 %)
Writing at 0x001003ef... (76 %)
Writing at 0x00106085... (78 %)
Writing at 0x0010bb58... (80 %)
Writing at 0x0011140d... (82 %)
Writing at 0x0011702d... (84 %)
Writing at 0x0011fd5e... (86 %)
Writing at 0x001283b6... (88 %)
Writing at 0x0012dfd9... (90 %)
Writing at 0x00133671... (92 %)
Writing at 0x00138d37... (94 %)
Writing at 0x0013eb85... (96 %)
Writing at 0x00143fc1... (98 %)
Writing at 0x001496ac... (100 %)
Wrote 1306960 bytes (834821 compressed) at 0x00010000 in 11.4 seconds (effective 917.8 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
===================================================================================== [SUCCESS] Took 73.41 seconds =====================================================================================

Environment    Status    Duration
-------------  --------  ------------
usb            SUCCESS   00:01:13.414
====================================================================================== 1 succeeded in 00:01:13.414 ====================================================================================== *  Terminal will be reused by tasks, press any key to close it.

cyberman54, 2022-12-02:
The toolchain seems current, i checked the compiler log.

I cannot reproduce the issue.
@t-huyeng @spmrider could you, please, test the current code and look if you can reproduce this issue? Thanks.

cyberman54, 2022-12-02:
@Qmick01 please look on the display, while running v3.4.4 - what is shown in line with "RLIM:" ?

t-huyeng, 2022-12-02:
I can not reproduce this issue. Looks like everything is working fine. 

I also cloned the repository completed new and used a new board (Heltec). 
See my configs [here](https://gist.github.com/t-huyeng/42e08a22ca2a7f71ea00f681dded2d9f).

My steps as descript in the documentations after cloning:

```zsh

➜  ESP32-Paxcounter  (master) ✔ cp platformio_orig.ini platformio.ini

➜  ESP32-Paxcounter  (master) ✔ cp src/paxcounter_orig.conf src/paxcounter.conf

➜  ESP32-Paxcounter  (master) ✔ cp src/loraconf_sample.h src/loraconf.h

➜  ESP32-Paxcounter  (master) ✔ cp src/ota_sample.conf src/ota.conf
```
I only updated `paxcounter.conf` to enable BLECOUNTER see [here](https://gist.github.com/t-huyeng/42e08a22ca2a7f71ea00f681dded2d9f#file-paxcounter-conf-L30) and set my board to `heltecv21.h` in the `platformio.ini` see [here ](https://gist.github.com/t-huyeng/42e08a22ca2a7f71ea00f681dded2d9f#file-platformio-ini-L13).

I also updated the lorawan keys `loraconf.h`.

Running Heltec board: 
[image]

Grafana Dashboard:
[image]

Qmick01, 2022-12-02:
So this morning the 3.4.4 board is showing a ble count of 1 but two units on an older build are running next to it and they both show a ble count of 5.

The 3.4.4 screen shots are below. It shows RSSI 37 and RLIM off.

I don't set the RSSI in my config file so that is whatever it is defaulted to. See image of the conf file.

If that's the actual RSSI then it could be a problem. What should I set it to in the conf file so it defaults to no limit?

Perhaps this is the issue.
[image]
[image]

t-huyeng, 2022-12-02:
@Qmick01 I think your config file picture is missing. 
You are also sharing some private pictures with your pictures currently. 

It is also the best to share your config file as formatted code as I described https://github.com/cyberman54/ESP32-Paxcounter/issues/921#issuecomment-1326727826 here.

Qmick01, 2022-12-02:
I have added the photo of the Paxcounter.conf section showing the RSSILIMIT value at zero.

Any thoughts?

t-huyeng, 2022-12-02:
Have you read the comment behind `#define RSSILIMIT 0` ?

It states this: ` // 0...-128, set to 0 if you do not want to filter signals` 
So if you have it set to 0 (default) no signal filtering will happen. 

It would also be much easier to help you if you could share config files as formatted code as I described https://github.com/cyberman54/ESP32-Paxcounter/issues/921#issuecomment-1326727826 here or as a gist.

I also tried my board with `#define WIFICOUNTER                     0` and it works fine. 
I would suggest you check again if your files are like this https://gist.github.com/t-huyeng/42e08a22ca2a7f71ea00f681dded2d9f

How old is the version you are running on board which are showing more BLE-Counts? Which version of libpax is running there?

As far as I know there has been changes to the libpax for better filtering (maybe this here https://github.com/dbSuS/libpax/commit/277a5d2959512f33c6aac74b9eed0f002673f274).

cyberman54, 2022-12-02:
>As far as I know there has been changes to the libpax for better filtering (maybe this here >https://github.com/dbSuS/libpax/commit/277a5d2959512f33c6aac74b9eed0f002673f274).

Exactly. Thus, count values generated with older versions are not comparable with newer ones.

Qmick01, 2022-12-02:
Yes. No RSSILIMIT value means to filtering on RSSI. That's what I would expect.

The other boards are running 3.1.2 version of the code. Is the new LIBPAX subsequent to that version?

Qmick01, 2022-12-02:
Thanks for this.

That's a very significant change as it means historical data no longer
lines up with current data so making a timeline of activity over time
becomes quite challenging as the meaning of changing pax values could be
mixed up between the software changes and the actual pax values.

Is there a clear sense of what the change is based on?

For example, was the change made to eliminate a specific issue or count
problem?

Michael Joseph
+44 (0) 7790928628
"In Velo, Veritas"

cyberman54, 2022-12-02:
This change is now already ~1 year in place. Reasons were
1) improving privacy
2) studies showed that using randomized MACs gives better results on number of pax, because headless devices like cars, scooters etc. are 
better filtered this way

closed 2022-12-02 by cyberman54

Qmick01, 2022-12-02:
Thanks for help figuring this out.

I have had cases where this build sees no BLE even when I am in the room
with my phone BLE in the same room. This morning it saw one BLE when the
others saw 5 and no doubt there is evidence as to why either number is
correct.

I will do some more testing.

Michael Joseph
+44 (0) 7790928628
"In Velo, Veritas"

cyberman54, 2022-12-02:
Remember that libpax lib is an approach to estimate crowds and/or relative changes over time. Not single persons, not absolute values.

Qmick01, 2022-12-02:
Yes, that's how I have been using it, and often having some
challenges getting numbers that map to a rough headcount so if a change in
the algorithm makes this better then that's great news.

Do you have any insights as to how well the new count numbers equate to
actual headcount?

Michael Joseph
+44 (0) 7790928628
"In Velo, Veritas"