cyberman54/ESP32-Paxcounter #924: Build error 'ONE_BIT_DISPLAY' does not name a type; did you mean 'HAS_DISPLAY'?

joaquinpolom, 2022-12-26:
Error at compile with PlatformIO
include/display.h:19:8: error: 'ONE_BIT_DISPLAY' does not name a type; did you mean 'HAS_DISPLAY'?
extern ONE_BIT_DISPLAY *dp;

src/display.cpp:62:3: error: 'dp' was not declared in this scope
dp = new ONE_BIT_DISPLAY;

In platform.ini
halfile = ttgov21new.h

cyberman54, 2022-12-27:
@joaquinpolom please post your compiler build output, i.e. library dependencies / removal notes.

joaquinpolom, 2022-12-27:
I have clean and build with the same problem.
I have delete the .pio directory. I have build the proyect with this error.

> Executing task: C:\Users\joaqu\.platformio\penv\Scripts\platformio.exe run <

Can not remove temporary directory `C:\Users\joaqu\OneDrive\Documentos_de_Joaquin\Proyects_IoT\GitHub\ESP32-Paxcounter\.pio\build`. Please remove it manually to avoid build issues
Processing usb (framework: arduino; board: esp32dev; platform: espressif32@4.4.0)
------------------------------------------------------------------------------------------------------------------------------------------------
Library Manager: Installing git+https://github.com/dbSuS/libpax.git
git version 2.35.3.windows.1
Cloning into 'C:\Users\joaqu\.platformio\.cache\tmp\pkg-installing-ygutzp9d'...
remote: Enumerating objects: 29, done.
remote: Counting objects: 100% (29/29), done.
remote: Compressing objects: 100% (27/27), done.
remote: Total 29 (delta 1), reused 9 (delta 0), pack-reused 0
Receiving objects: 100% (29/29), 32.83 KiB | 111.00 KiB/s, done.
Resolving deltas: 100% (1/1), done.
Library Manager: libpax@1.1.0+sha.fe98131 has been installed!
Library Manager: Installing git+https://github.com/SukkoPera/Arduino-Rokkit-Hash.git
git version 2.35.3.windows.1
Cloning into 'C:\Users\joaqu\.platformio\.cache\tmp\pkg-installing-1ezgcf7o'...
remote: Enumerating objects: 16, done.
remote: Counting objects: 100% (16/16), done.
remote: Compressing objects: 100% (13/13), done.
remote: Total 16 (delta 1), reused 10 (delta 0), pack-reused 0
Receiving objects: 100% (16/16), 5.74 KiB | 1.91 MiB/s, done.
Resolving deltas: 100% (1/1), done.
Library Manager: RokkitHash@20151212.0.0+sha.8500494 has been installed!
Library Manager: Installing bblanchon/ArduinoJson @ ^6
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
Library Manager: ArduinoJson@6.20.0 has been installed!
Library Manager: Installing makuna/RTC @ ^2.3.5
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
Library Manager: RTC@2.3.5 has been installed!
Library Manager: Installing spacehuhn/SimpleButton
Unpacking  [####################################]  100%
Library Manager: SimpleButton@0.0.0-alpha+sha.026bc1e41a has been installed!
Library Manager: Installing 256dpi/MQTT @ ^2.4.8
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
Library Manager: MQTT@2.5.0 has been installed!
Library Manager: Installing mathertel/OneButton @ ^2.0.3
Unpacking  [####################################]  100%
Library Manager: OneButton@2.0.3 has been installed!
Library Manager: Installing git+https://github.com/cyberman54/sds-dust-sensors-arduino-library.git
git version 2.35.3.windows.1
Cloning into 'C:\Users\joaqu\.platformio\.cache\tmp\pkg-installing-3q68umit'...
remote: Enumerating objects: 23, done.
remote: Counting objects: 100% (23/23), done.
remote: Compressing objects: 100% (18/18), done.
Receiving objects: 100% (23/23)sed 10 (delta 3), pack-reused 0
Receiving objects: 100% (23/23), 11.74 KiB | 5.87 MiB/s, done.
Resolving deltas: 100% (4/4), done.
Library Manager: Nova Fitness Sds dust sensors library@1.5.1+sha.47bc356 has been installed!
Library Manager: Installing seeed-studio/Ultrathin_LED_Matrix @ ^1.0.0
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
Library Manager: Ultrathin_LED_Matrix@1.0.0 has been installed!
Verbose mode can be enabled via `-v, --verbose` option
Parsing hardware configuration from C:\Users\joaqu\OneDrive\Documentos_de_Joaquin\Proyects_IoT\GitHub\ESP32-Paxcounter\src\hal\ttgov21new.h
Parsing LMIC configuration from C:\Users\joaqu\OneDrive\Documentos_de_Joaquin\Proyects_IoT\GitHub\ESP32-Paxcounter\src\lmic_config.h
Parsing LORAWAN keys from C:\Users\joaqu\OneDrive\Documentos_de_Joaquin\Proyects_IoT\GitHub\ESP32-Paxcounter\src\loraconf.h
Parsing OTA keys from C:\Users\joaqu\OneDrive\Documentos_de_Joaquin\Proyects_IoT\GitHub\ESP32-Paxcounter\src\ota.conf
Target board: esp32dev @ 921600bps
Partition table: min_spiffs.csv
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32dev.html
PLATFORM: Espressif 32 (4.4.0) > Espressif ESP32 Dev Module
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
[... 256 lines cut ...]
 extern ONE_BIT_DISPLAY *dp;
        ^~~~~~~~~~~~~~~
        HAS_DISPLAY
*** [.pio\build\usb\src\irqhandler.cpp.o] Error 1
In file included from include/cyclic.h:10,
                 from include/senddata.h:6,
                 from include/libpax_helpers.h:6,
                 from include/rcommand.h:6,
                 from include/lorawan.h:7,
                 from include/led.h:11,
                 from src/led.cpp:3:
include/display.h:19:8: error: 'ONE_BIT_DISPLAY' does not name a type; did you mean 'HAS_DISPLAY'?
 extern ONE_BIT_DISPLAY *dp;
        ^~~~~~~~~~~~~~~
        HAS_DISPLAY
*** [.pio\build\usb\src\led.cpp.o] Error 1
========================================================= [FAILED] Took 464.36 seconds =========================================================

Environment    Status    Duration
-------------  --------  ------------
usb            FAILED    00:07:44.356
==================================================== 1 failed, 0 succeeded in 00:07:44.356 ==================================================== 
The terminal process "C:\Users\joaqu\.platformio\penv\Scripts\platformio.exe 'run'" terminated with exit code: 1.

Terminal will be reused by tasks, press any key to close it.

cyberman54, 2022-12-27:
Your toolchain is outdated. It is not using the current code: 
"platform: espressif32@4.4.0"

Current code is using espressif@5.2.0, see platformio_orig.ini.

If you already build this code before: did you re-copy current platformio_orig.ini to platformio.ini?

joaquinpolom, 2022-12-28:
Solved.
Deleted .pio. Rename platformio.ini to platformio_old.ini. Copy platformio_orig.ini to platformio.ini.
Clean and build and compile ok

closed 2022-12-28 by joaquinpolom