cyberman54/ESP32-Paxcounter #940: lilygo esp paxcounter module. trying to build. but ....

mayday77, 2023-02-21:
1. edit all the stuff
2. press Clear 
3. Press build and ....
4. sorry , am kind of lame , but trying! =)
https://pastebin.com/YdNciPw1

cyberman54, 2023-02-21:
Please post your lmic_config.h including markup of your edits in this file.
It looks like you selected a bandplan (CFGxxx) which is not supported by LMIC Lorawan stack.

mayday77, 2023-02-22:
seem to be there is no plave for 433 mhz 
-----
#if defined(CFG_eu868)
# include "lmic_bandplan_eu868.h"
#elif defined(CFG_us915)
# include "lmic_bandplan_us915.h"
#elif defined(CFG_au915)
# include "lmic_bandplan_au915.h"
#elif defined(CFG_as923)
# include "lmic_bandplan_as923.h"
#elif defined(CFG_kr920)
# include "lmic_bandplan_kr920.h"
#elif defined(CFG_in866)
# include "lmic_bandplan_in866.h"
#else
# error "CFG_... not properly set for bandplan"
#endif----

cyberman54, 2023-02-22:
you're right, there is no EU433 setting in lmic. As stated here https://github.com/mcci-catena/arduino-lmic#selecting-the-lorawan-region-configuration

I will update lmic_config.h soon. Thanks for raising this issue.

cyberman54, 2023-02-24:
unsupported bands removed.

https://github.com/cyberman54/ESP32-Paxcounter/commit/651869d55ee524a97f806efd2a7921802413750f

closed 2023-02-24 by cyberman54