cyberman54/ESP32-Paxcounter #948: TTN Time Sync

MichaelJumble, 2023-03-06:
May be wrong issue here, but I struggle long time to solve it before I add this
here.
I try to get epoch Time of the TTN Network to synchronize the local time ESP32 node.
For now, I work with the “LMIC-NODE-DEMO” Code and Down- / Uplink work fine. 
After grabbing the whole Internet I can’t find a working example for TTN3 to ask for TimeSync.
Searching the entire Internet, and found out that the PAX Sample here with a working time sync function.
The PAX Demo here works fine and also get a TimeSync from the Network.

Be sure, the PAX Code here is fantastic and written in professional style.
But I can’t get it working in a simple LMIC code.
Maybe someone did have an example that only covers the time sync with the TTN
Network ?

cyberman54, 2023-03-06:
Please move this to tab discussions, thank you.

closed 2023-03-06 by cyberman54

cyberman54, 2023-03-06:
@MichaelJumble look at [this example](https://github.com/mcci-catena/arduino-lmic/tree/master/examples/ttn-otaa-network-time)

MichaelJumble, 2023-03-06:
Your right. Its not an issue, so disc would be the right place. But at first I try out your links. Seems to look good. THX for your great job.