cyberman54/ESP32-Paxcounter #956: #define HAS_RTC MY_DISPLAY_SDA, MY_DISPLAY_SCL // SDA, SCL does not work for #if (...

TomRottler, 2023-04-16:
We are using an external RTC, connected to the LILYGO® LoRa 1.6.
Therefor we've created our own HAL with #define HAS_RTC MY_DISPLAY_SDA, MY_DISPLAY_SCL, as used for other pre-defined devices.

Now we faced the problem that this define does not work properly in some cases e.g. in timekeeper.cpp:
#if ((HAS_GPS) || (HAS_RTC))

We did a workaround in rtctime.cpp @ rtc_init: 
  Wire.begin(MY_DISPLAY_SDA, MY_DISPLAY_SCL);

The following line was already fixed to the GPIO, in this case this should make sense.
Additionally we changed the HAS_RTC to a single value in our HAL.

Can you confirm this?

Thanks
TOM

cyberman54, 2023-04-17:
fixed. Please confirm, thanks.

TomRottler, 2023-04-19:
it works. Thanks.

closed 2023-04-19 by TomRottler