cyberman54/ESP32-Paxcounter #969: Supported LED Matrix display

cgruber-sdc, 2023-05-23:
Any technical hint for the supported _LED Matrix display_ mentioned in https://cyberman54.github.io/ESP32-Paxcounter/hardware/#supported-esp32-based-boards unfortunately the aliexpress link is broken in the doc page. Would be a 64x32 RGB like https://eckstein-shop.de/RGB-Full-Color-LED-Matrix-Panel-64x32-4mm-Pitch-1-32-Scan work also or has it to be a monochrome one?

cyberman54, 2023-05-24:
The interface of your 64x32 alternative should be compatible, but it's a RGB display. The example LEd controller in paxcounter is monochrome only.

cyberman54, 2023-05-24:
The "original" type seems still available on aliexpress: 

€ 14,97  2%OFF | P 4,75 Innen Rote Farbe LED Display Modul Panel 304mm x 76mm 64x16 Pixel Led-anzeige gitter Module
https://a.aliexpress.com/_ExTcLsh

closed 2023-06-03 by cyberman54