cyberman54/ESP32-Paxcounter #971: question: what about high numbers?

AugustQu, 2023-05-31:
I reactivated my Paxcounter and made it working again. In my first test it counted between 120 and 160 smartphones on a place.

Is this a limit? Ca it detect let's say >200 devices? Or maybe 300?

Or is it better to change the interval from 60 seconds to 120 seconds? 

In paxcounter.conf I found SENDCYCLE with a value of 30. But changing this seems not to modify the behaviour.

AugustQ

cyberman54, 2023-05-31:
Sendcycle is stored in NVRAM of board. If you change the default on a new build, you must clear the NVRAM, otherwise the old value is restored during startup, and there is no effect. Or you change it during runtime by rcommand.

cyberman54, 2023-06-03:
Limit by memory is 65535. But the real memory is a bottleneck by cpu speed and limited radio channels.

cyberman54, 2023-06-03:
I close this, since it's not an issue.
For further questions or discussion, please move to discussion tab.

closed 2023-06-03 by cyberman54