cyberman54/ESP32-Paxcounter #978: BLE Scanner not working on v3.6.0 on TTGO LoRa

littlecxm, 2023-06-19:
Hi, I am on build v3.6.0 and flash it, but in my board it shown `BLTH: off`and I'm sure configured `BLECOUNTER` to `1`
I also found some issue in #921, but it seems the situation is different
I'm not sure its a issue or disabled by some reason?

[image]

my `paxcounter.conf` file

```c
// clang-format off

// ----- Paxcounter user config file ------
//
// --> adapt to your needs and use case <--
//
// Notes:
// 1. After editing, before "build", use "clean" button in PlatformIO!
// 2. Clear NVRAM of the board to delete previous stored runtime settings (pio run -t erase)
// Device options
#define VERBOSE                         1       // set to 0 to silence the device, 1 enables additional debug output
#define BOOTMENU                        0       // 0 = no bootmenu, 1 = device brings up boot menu before starting application
#define BOOTDELAY                       30      // time [seconds] while devices waits in boot menue for input
#define BOOTTIMEOUT                     300     // time [seconds] while devices waits to finish upload a firmware file
#define SDLOGGING                       0       // set to 1 for system logging to SD card (if device has one)

// Payload send cycle and encoding
#define SENDCYCLE                       30      // payload send cycle [seconds/2], 0 .. 255
#define SLEEPCYCLE                      0       // sleep time after a send cycle [seconds/10], 0 .. 65535; 0 means no sleep [default = 0]
#define PAYLOAD_ENCODER                 2       // payload encoder: 1=Plain, 2=Packed, 3=Cayenne LPP dynamic, 4=Cayenne LPP packed
#define COUNTERMODE                     0       // 0=cyclic, 1=cumulative, 2=cyclic confirmed
#define SYNCWAKEUP                      300     // shifts sleep wakeup to top-of-hour, when +/- X seconds off [0=off]

// default settings for transmission of sensor data (first list = data on / second line = data off)
#define PAYLOADMASK                                                                             \
            ((GPS_DATA | MEMS_DATA | COUNT_DATA | SENSOR1_DATA | SENSOR2_DATA | SENSOR3_DATA) & \
            (~BATT_DATA) & (~RESERVED_DATA))

// MAC sniffing settings
#define BLECOUNTER                      1       // set to 0 if you do not want to start the BLE sniffer
#define WIFICOUNTER                     1       // set to 0 if you do not want to start the WIFI sniffer
#define RSSILIMIT                       0       // 0...-128, set to 0 if you do not want to filter signals

// BLE scan parameters
#define BLESCANTIME                     0       // [seconds] scan duration, 0 means infinite [default], see note below
#define BLESCANWINDOW                   80      // [milliseconds] scan window, see below, 3 .. 10240, default 80ms
#define BLESCANINTERVAL                 80      // [illiseconds] scan interval, see below, 3 .. 10240, default 80ms = 100% duty cycle

/* Note: guide for setting bluetooth parameters
*
* |< Scan Window >       |< Scan Window >       | ... |< Scan Window >       |
* |<    Scan Interval   >|<    Scan Interval   >| ... |<    Scan Interval   >|
* |<                   Scan duration                                        >|
*
* Scan duration sets how long scanning should be going on, before starting a new scan cycle. 0 means infinite (default).
* Scan window sets how much of the interval should be occupied by scanning. Should be >= BLESCANINTERVAL.
* Scan interval is how long scanning should be done on each channel. BLE uses 3 channels for advertising.
* -> Adjust these values with power consumption in mind if power is limited.
* -> Scan interval can be changed during runtime by remote comammand.
*/

// WiFi scan parameters
#define WIFI_MY_COUNTRY                 "01"    // select 2-letter locale for Wifi RF settings, e.g. "DE"; use "01" for world safe mode
#define	WIFI_CHANNEL_SWITCH_INTERVAL    50      // [seconds/100] -> 0,5 sec.
#define WIFI_CHANNEL_MAP                WIFI_CHANNEL_ALL  // possible values see libpax_api.h

// LoRa payload default parameters
#define MEM_LOW                         2048    // [Bytes] low memory threshold triggering a send cycle
#define RETRANSMIT_RCMD                 5       // [seconds] wait time before retransmitting rcommand results
#define PAYLOAD_BUFFER_SIZE             51      // maximum size of payload block per transmit
#define PAYLOAD_OPENSENSEBOX            0       // send payload compatible to sensebox.de (swap geo position and pax data)
[... 358 lines cut ...]
Writing at 0x000e3b84... (76 %)
Writing at 0x000e95b5... (78 %)
Writing at 0x000eed75... (80 %)
Writing at 0x000f7a17... (82 %)
Writing at 0x001001b4... (84 %)
Writing at 0x00105a8e... (86 %)
Writing at 0x0010b35b... (89 %)
Writing at 0x00111197... (91 %)
Writing at 0x001166f2... (93 %)
Writing at 0x0011c64f... (95 %)
Writing at 0x00121a49... (97 %)
Writing at 0x0012730a... (100 %)
Wrote 1161296 bytes (749261 compressed) at 0x00010000 in 11.5 seconds (effective 806.3 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
================================================== [SUCCESS] Took 19.15 seconds ==================================================

Environment    Status    Duration
-------------  --------  ------------
usb            SUCCESS   00:00:19.155
================================================== 1 succeeded in 00:00:19.155 ==================================================
 *  Terminal will be reused by tasks, press any key to close it.
```

cyberman54, 2023-06-19:
Make sure you do not have BLE disabled in NVRAM runtime parameters as a result from an earlier run by either
- clear NVRAM (pio run -t erase), or
- send rcommand to enable BLE and persitent save BLE state

littlecxm, 2023-06-19:
it works now after I clean the NVRAM
Thanks a lot!

closed 2023-06-19 by littlecxm