cyberman54/ESP32-Paxcounter #981: LoPy4 Fails to join LoRa network (TTN & North) unless for SF7

markwaters306, 2023-06-28:
I have a LoPy4 that fails to join the LoRa network unless the device is within 10m of the gateway.

I have tried two different services TTN and North and experience the same issue.

I have also tried two different LoPy4 boards and antennas and the issue remains.

I've also uncommented '#define CLOCK_ERROR_PROCENTAGE 5' to try help but still no luck.

I'm wondering if there is an issue with the lmic stack. I've left the device on for 24 hours and still nothing came through on the gateway. Here is output log with '#define LMIC_DEBUG_LEVEL 2'.

The only time I can get it to join is by taking the device extreamly close to the gateway. Over 10m and it struggles.

09:13:21.339 > -41293353: RXMODE_SINGLE, freq=869525000, SF=12, BW=125, CR=4/5, IH=0
09:13:21.579 > rxtimeout: entry: -41278983 rxtime: -41293360 entry-rxtime: 14377 now-entry: 128 rxtime-txend: 374875
09:13:21.580 > -41278829: processRx2Jacc txrxFlags 0x2 --> 00
09:13:21.580 > -41278608: engineUpdate, opmode=0x4
09:14:03.296 > W (68101538) src/lorawan.cpp: LORA sendqueue is full
09:15:03.292 > W (68161538) src/lorawan.cpp: LORA sendqueue is full
09:15:23.502 > -33658233: engineUpdate, opmode=0x4
09:15:23.502 > -33658176: TXMODE, freq=868500000, len=23, SF=9, BW=125, CR=4/5, IH=0
09:15:28.708 > -33333483: setupRx1 txrxFlags 00 --> 01
09:15:28.709 > start single rx: now-rxtime: 3
09:15:28.709 > -33332916: RXMODE_SINGLE, freq=868500000, SF=9, BW=125, CR=4/5, IH=0
09:15:28.737 > rxtimeout: entry: -33331046 rxtime: -33332923 entry-rxtime: 1877 now-entry: 128 rxtime-txend: 312375
09:15:29.683 > -33270983: setupRx2 txrxFlags 0x1 --> 02
09:15:29.707 > start single rx: now-rxtime: 3
09:15:29.707 > -33270416: RXMODE_SINGLE, freq=869525000, SF=12, BW=125, CR=4/5, IH=0
09:15:29.949 > rxtimeout: entry: -33256046 rxtime: -33270423 entry-rxtime: 14377 now-entry: 128 rxtime-txend: 374875
09:15:29.949 > -33255892: processRx2Jacc txrxFlags 0x2 --> 00
09:15:29.949 > -33255671: engineUpdate, opmode=0x4
09:16:03.294 > W (68221538) src/lorawan.cpp: LORA sendqueue is full
09:17:03.288 > W (68281538) src/lorawan.cpp: LORA sendqueue is full
09:18:03.286 > W (68341538) src/lorawan.cpp: LORA sendqueue is full
09:19:03.298 > W (68401538) src/lorawan.cpp: LORA sendqueue is full
09:19:18.616 > -18963796: engineUpdate, opmode=0x4
09:19:18.616 > -18963738: TXMODE, freq=868100000, len=23, SF=9, BW=125, CR=4/5, IH=0
09:19:23.823 > -18639046: setupRx1 txrxFlags 00 --> 01
09:19:23.823 > start single rx: now-rxtime: 3
09:19:23.823 > -18638478: RXMODE_SINGLE, freq=868100000, SF=9, BW=125, CR=4/5, IH=0
09:19:23.853 > rxtimeout: entry: -18636608 rxtime: -18638485 entry-rxtime: 1877 now-entry: 128 rxtime-txend: 312375
09:19:24.798 > -18576546: setupRx2 txrxFlags 0x1 --> 02
09:19:24.820 > start single rx: now-rxtime: 2
09:19:24.820 > -18575978: RXMODE_SINGLE, freq=869525000, SF=12, BW=125, CR=4/5, IH=0
09:19:25.064 > rxtimeout: entry: -18561608 rxtime: -18575985 entry-rxtime: 14377 now-entry: 128 rxtime-txend: 374875
09:19:25.064 > -18561454: processRx2Jacc txrxFlags 0x2 --> 00
09:19:25.064 > -18561233: engineUpdate, opmode=0x4
09:20:03.291 > W (68461538) src/lorawan.cpp: LORA sendqueue is full
09:21:03.302 > W (68521538) src/lorawan.cpp: LORA sendqueue is full
09:22:03.298 > W (68581538) src/lorawan.cpp: LORA sendqueue is full
09:22:52.328 > -5605733: engineUpdate, opmode=0x4
09:22:52.351 > -5605677: TXMODE, freq=868300000, len=23, SF=9, BW=125, CR=4/5, IH=0
09:22:57.555 > -5280983: setupRx1 txrxFlags 00 --> 01
09:22:57.556 > start single rx: now-rxtime: 3
09:22:57.556 > -5280416: RXMODE_SINGLE, freq=868300000, SF=9, BW=125, CR=4/5, IH=0
09:22:57.587 > rxtimeout: entry: -5278546 rxtime: -5280423 entry-rxtime: 1877 now-entry: 128 rxtime-txend: 312375
09:22:58.531 > -5218483: setupRx2 txrxFlags 0x1 --> 02
09:22:58.554 > start single rx: now-rxtime: 3
09:22:58.554 > -5217916: RXMODE_SINGLE, freq=869525000, SF=12, BW=125, CR=4/5, IH=0
09:22:58.791 > rxtimeout: entry: -5203546 rxtime: -5217923 entry-rxtime: 14377 now-entry: 128 rxtime-txend: 374875
09:22:58.792 > -5203392: processRx2Jacc txrxFlags 0x2 --> 00
09:22:58.792 > -5203171: engineUpdate, opmode=0x4
09:23:03.289 > W (68641538) src/lorawan.cpp: LORA sendqueue is full
09:24:03.300 > W (68701538) src/lorawan.cpp: LORA sendqueue is full
09:25:03.297 > W (68761538) src/lorawan.cpp: LORA sendqueue is full
09:26:03.293 > W (68821538) src/lorawan.cpp: LORA sendqueue is full
09:27:03.304 > W (68881538) src/lorawan.cpp: LORA sendqueue is full
09:27:04.180 > 10133392: engineUpdate, opmode=0x4
09:27:04.180 > 10133449: TXMODE, freq=868500000, len=23, SF=10, BW=125, CR=4/5, IH=0
09:27:09.526 > 10468517: setupRx1 txrxFlags 00 --> 01
09:27:09.550 > start single rx: now-rxtime: 3
[... 92 lines cut ...]
10:04:03.343 > W (71101538) src/lorawan.cpp: LORA sendqueue is full
10:05:03.337 > W (71161538) src/lorawan.cpp: LORA sendqueue is full
10:06:03.334 > W (71221538) src/lorawan.cpp: LORA sendqueue is full
10:07:03.346 > W (71281538) src/lorawan.cpp: LORA sendqueue is full
10:08:03.343 > W (71341538) src/lorawan.cpp: LORA sendqueue is full
10:09:03.338 > W (71401538) src/lorawan.cpp: LORA sendqueue is full
10:10:03.351 > W (71461538) src/lorawan.cpp: LORA sendqueue is full
10:11:03.346 > W (71521538) src/lorawan.cpp: LORA sendqueue is full
10:12:03.342 > W (71581538) src/lorawan.cpp: LORA sendqueue is full
10:13:03.339 > W (71641538) src/lorawan.cpp: LORA sendqueue is full
10:14:03.350 > W (71701538) src/lorawan.cpp: LORA sendqueue is full
10:15:03.347 > W (71761538) src/lorawan.cpp: LORA sendqueue is full
10:15:59.193 > 193568704: engineUpdate, opmode=0x4
10:15:59.193 > 193568761: TXMODE, freq=868300000, len=23, SF=11, BW=125, CR=4/5, IH=0
10:16:03.355 > W (71821538) src/lorawan.cpp: LORA sendqueue is full
10:16:04.988 > 193932079: setupRx1 txrxFlags 00 --> 01
10:16:05.014 > start single rx: now-rxtime: 2
10:16:05.014 > 193932647: RXMODE_SINGLE, freq=868300000, SF=11, BW=125, CR=4/5, IH=0
10:16:05.134 > rxtimeout: entry: 193939892 rxtime: 193932640 entry-rxtime: 7252 now-entry: 128 rxtime-txend: 312375
10:16:06.018 > 193994579: setupRx2 txrxFlags 0x1 --> 02
10:16:06.018 > start single rx: now-rxtime: 3
10:16:06.018 > 193995148: RXMODE_SINGLE, freq=869525000, SF=12, BW=125, CR=4/5, IH=0
10:16:06.258 > rxtimeout: entry: 194009517 rxtime: 193995140 entry-rxtime: 14377 now-entry: 128 rxtime-txend: 374875
10:16:06.259 > 194009670: processRx2Jacc txrxFlags 0x2 --> 00
10:16:06.259 > 194009892: engineUpdate, opmode=0x4

cyberman54, 2023-06-28:
The log just shows that LMiC is increasing the SF level during join process. This works as designed. Please post log which shows the begin of the join proces, and a successful join.

Which network server or network provider are you using for join?
Which country settings?
Which type of gateway?
Do you have access to gateway and/or network server and can you provide logs from this side?
Did you double check that your network credentials are valid?

I don't expect an lmic issue here, since the used mcci lmic lib has reached a pretty stable and well tested state.

markwaters306, 2023-06-28:
Thanks for your response.

Which network server or network provider are you using for join? 
- I've tested both TTN community and a private provider called North
Which country settings?
-EU 868MHz
Which type of gateway?
- The outdoor one is proprietary so I don't have the model, the TTN one is TBMH100
Do you have access to gateway and/or network server and can you provide logs from this side?
- I'll attach a copy below, the RSSI is very low for being so close to the gateway
- I've also tested with a adenuis signal meter in the same location and get RSSI values above -70dBm
Did you double check that your network credentials are valid?
- The device will join so the credentials must be correct

[messages.csv](https://github.com/cyberman54/ESP32-Paxcounter/files/11893762/messages.csv)

Have you had a LoPy4 working at distance before?

cyberman54, 2023-06-28:
The gateway log show that the server accepted the join. So it's not a crendential problem. I looks like you may have an RX problem, what means the downlink with the accept does not lmic. This is a common reason why OTA joining fails.

Make sure
- to use an external antenna with your Lopy4, it won't work without antenna
- check that the antenna connection / cable is not broken
- check that you're using an _868MHz_ antenna
- check that you're using the _868MHz U.FL connector_, not the 433 MHZ connector and not the Wifi/BLE connector (see pic, the Lopy4 has 3 U.FL connectors)
- check that the log from the network server shows _your local_ gateway, and not another one which may be too far away for the downlink

[image]

markwaters306, 2023-06-28:
Had the antenna connected to wrong port... stupid avoidable mistake! RTFM

Thanks for the support and sorry for the non-code issue!

closed 2023-06-28 by cyberman54