cyberman54/ESP32-Paxcounter #990: Hardware support for T-Beam V1.2 with AXP2101

XAct-6510, 2023-09-26:
Please add support for the new version of the Lilygo T-Beam V1.2 with the new power management chip AXP2101.

cyberman54, 2023-09-28:
Can you provide link to specs of v1.2

XAct-6510, 2023-09-28:
Hi, thanks for your reply! Here are the links:
https://www.amazon.de/LILYGO-Entwicklungsboard-Meshtastic-Mainboards-Soldered/dp/B0B454TB6S
https://github.com/Xinyuan-LilyGO/LilyGO-T-Beam
https://github.com/Xinyuan-LilyGO/LilyGo-LoRa-Series
AXP2101 chip: https://github.com/meshtastic/firmware/blob/7bb281d5c569870c27cdd65bcd14507148e4ca9b/src/Power.cpp#L516

cyberman54, 2023-09-29:
In all your links, i can't find a schematic or specs for v1.2, it all renders as v1.1?

XAct-6510, 2023-09-29:
Yes, they hid this fact pretty well, unfortunately. By changing the PMU from AXP192 to AXP2102 they have changed the versioning from 1.1 to 1.2. Please see the screenshot.

[image]

cyberman54, 2023-10-07:
i checked the Xpowerslib example, and found that there is some refactoring of power.cpp needed. Can you do this, and make a PR?

XAct-6510, 2023-10-09:
Unfortunately I'm not a programmer, I'm a networking guy... But I can provide you one of my 5 T-Beam V1.2 if that helps.

XAct-6510, 2023-10-10:
As mentioned I'm not a programmer, but maybe this can be used? 
https://github.com/meshtastic/firmware/blob/master/src/Power.cpp
The T-Beam V1.2 comes preinstalled with the Meshtastic firmware, which obviously works with the new hardware revision.

cyberman54, 2023-10-10:
i saw that, but would stick to the example code of XPowersLib, because Meshtastic has more complexity. But again, you or someone else would be welcome to show up with a completely refactored power.cpp

TD-er, 2024-06-10:
Just for completeness, the schematics of the 1.2 version of the T-beam:
https://github.com/Xinyuan-LilyGO/LilyGo-LoRa-Series/tree/master/schematic

Also they split off the older boards, so here are all the LilyGo LoRa series schematics:
https://github.com/Xinyuan-LilyGO/LilyGo-LoRa-Series/tree/v0.0.1/schematic

idimou, 2024-06-14:
Anybody had a change to look at this guys?

TD-er, 2024-06-14:
I am looking into this.
First I am ironing out some issues with AXP192 (my PRs have just been merged) and then I will look into the AXP2101.

I have both versions here on my desk, but since those T-beam boards are quite expensive these days, I wanted to make sure I knew how it all should work.

cyberman54, 2024-06-26:
> I am looking into this. First I am ironing out some issues with AXP192 (my PRs have just been merged) and then I will look into the AXP2101.
> 
> I have both versions here on my desk, but since those T-beam boards are quite expensive these days, I wanted to make sure I knew how it all should work.

@TD-er I could do some testing. Received a T-Beam "Supreme" with AXP2101 this week. I finalized a hal file with matching pinout already, build paxcounter code for S3 and flashed it successfully on the T-Supreme. The board seems to be alive, because i can see the USB port triggers on reset. But nothing else happenes. Probably becaue AXP2101 is missing proper initialization, thus the board components are probably not powered.

TD-er, 2024-06-27:
I have been reading into the existing libraries.
There is even an example of how to make it runtime detectable.
This does require some changes like keeping a pointer to the base class.
See: https://github.com/lewisxhe/XPowersLib/blob/master/examples/XPowersLibInterface_Example/XPowersLibInterface_Example.ino

Still I'm puzzling a bit on how to make this as fool proof as possible as wrong settings for power chips like these can for sure ruin your device or stuff connected to it.

The last 2 weeks have been extremely unproductive for me, as apparently Covid is still lurking around and even though I've had it 3 times already, it still hits you every time. 
At least I'm already feeling better now, so I will have another look at this code.

cyberman54, 2024-07-01:
Support for T-Beam v1.2 is now implemented in branch [feature-asp2101](https://github.com/cyberman54/ESP32-Paxcounter/tree/feature-axp2101)

Does anyone who reads this own this board, and could test, please? Thanks.

XAct-6510, 2024-07-02:
I've set the "halfile" to "ttgobeam12.h" and compiled with my parameters set, but get errors:
....
Compiling .pio\build\usb\src\main.cpp.o
src/button.cpp: In function 'void button_init()':
src/button.cpp:44:10: error: 'class OneButton' has no member named 'setPressMs'; did you mean 'setPressTicks'?
   button.setPressMs(1000);
          ^~~~~~~~~~
          setPressTicks
*** [.pio\build\usb\src\button.cpp.o] Error 1
src/lorawan.cpp:34:16: error: 'virtual u1_t MyHalConfig_t::queryBusyPin()' marked 'override', but does not override
   virtual u1_t queryBusyPin(void) override {
                ^~~~~~~~~~~~
src/lorawan.cpp:42:16: error: 'virtual bool MyHalConfig_t::queryUsingDcdc()' marked 'override', but does not override
   virtual bool queryUsingDcdc(void) override { return true; }
                ^~~~~~~~~~~~~~
src/lorawan.cpp:43:16: error: 'virtual bool MyHalConfig_t::queryUsingDIO2AsRfSwitch()' marked 'override', but does not override
   virtual bool queryUsingDIO2AsRfSwitch(void) override { return true; }
                ^~~~~~~~~~~~~~~~~~~~~~~~
src/lorawan.cpp:44:16: error: 'virtual bool MyHalConfig_t::queryUsingDIO3AsTCXOSwitch()' marked 'override', but does not override
   virtual bool queryUsingDIO3AsTCXOSwitch(void) override { return true; }
                ^~~~~~~~~~~~~~~~~~~~~~~~~~
*** [.pio\build\usb\src\lorawan.cpp.o] Error 1
================================================================================================================================== [FAILED] Took 12.54 seconds ==================================================================================================================================
Environment    Status    Duration
-------------  --------  ------------
usb            FAILED    00:00:12.540

XAct-6510, 2024-07-02:
Can't get it to work yet - after flashing the new build the device hangs on boot:

rst:0x1 (POWERON_RESET),boot:0x12 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13232
load:0x40080400,len:3028
entry 0x400805e4
I (17) src/configmanager.cpp: Loading device configuration from NVRAM...
I (20) src/configmanager.cpp: Runtime configuration v3.6.2 loaded
I (21) src/main.cpp: Starting paxcounter_bb0aff61 v3.6.2 (runmode=0 / restarts=0                                                                                                   )
I (26) src/main.cpp: code build date: 1719913557
I (31) src/main.cpp: This is ESP32 chip with 2 CPU cores, WiFi/BT/BLE, silicon r                                                                                                   evision 3, 4MB external Flash
I (40) src/main.cpp: Internal Total heap 225948, internal Free Heap 198120
I (47) src/main.cpp: ChipRevision 3, Cpu Freq 240, SDK Version v4.4.5
I (53) src/main.cpp: Flash Size 4194304, Flash Speed 40000000
I (59) src/main.cpp: Wifi/BT software coexist version 1.2.0
I (64) src/main.cpp: Wifi STA MAC: B0:B2:1C:5F:59:24
I (68) src/main.cpp: IBM LMIC version 1.6.1468577746
I (73) src/main.cpp: Arduino LMIC version 4.2.0.1
I (78) src/lorawan.cpp: DevEUI: 70B3D57ED00657F9
I (82) src/main.cpp: TinyGPS+ version 1.1.0
I (86) C:/Users/GIS-Team/.platformio/packages/framework-arduinoespressif32@3.200                                                                                                   11.230801/cores/esp32/esp32-hal-i2c.c: Initialising I2C Master: sda=21 scl=22 fr                                                                                                   eq=100000
I (102) .pio/libdeps/usb/XPowersLib/src/XPowersCommon.tpp: SDA:21 SCL:22
W (107) C:/Users/GIS-Team/.platformio/packages/framework-arduinoespressif32@3.20                                                                                                   011.230801/libraries/Wire/src/Wire.cpp: Bus already started in Master Mode.
E (121) C:/Users/GIS-Team/.platformio/packages/framework-arduinoespressif32@3.20                                                                                                   011.230801/libraries/Wire/src/Wire.cpp: NULL TX buffer pointer
E (133) C:/Users/GIS-Team/.platformio/packages/framework-arduinoespressif32@3.20                                                                                                   011.230801/libraries/Wire/src/Wire.cpp: NULL TX buffer pointer
I (146) .pio/libdeps/usb/XPowersLib/src/XPowersAXP2101.tpp: ~XPowersAXP2101

cyberman54, 2024-07-02:
Looks like the i2c bus did not initialize for some reason. Thus, there is no communication with the AXP2101 chip, so it does not power up all board circuits.

cyberman54, 2024-07-02:
I can reproduce this with a T-Beam with AXP192 chip and single i2c bus. 
The reason is that there is a conflict in power.cpp by the already opened i2c bus by `i2c_init()` from main.cpp. The code needs some refactoring here, e.g. using a pointer to the i2c object.

cyberman54, 2024-07-02:
i committed a hot fix, it should work (please re-test), but yet to be discussed if this is a straight forward approach for board with two i2c buses. /cc @XAct-6510

XAct-6510, 2024-07-03:
I have applied the hotfix and it is now working. I'm still running some tests, but it seems to be running fine.

cyberman54, 2025-01-26:
Code is now able to support boards with PMU on 2nd I2C bus.
See hal file for T-Supreme S3. This may be adapted for T-Beam v1.2 with AXP2101.
/cc @XAct-6510

cyberman54, 2025-01-26:
A board hal file ttgobeam12.h was added now.

PR #1015 (2024-06-27): [AXPxxx] Add support for AXP2101 and AXP202
Split the code into several sections for AXP192/AXP2101/AXP202

ToDo:
- [ ] Test AXP192 on real hardware
- [ ] Test AXP2101 on real hardware
- [ ] Test AXP202 on real hardware
- [ ] Add more defines/boards using these chips
- [ ] Add defines on which LDO/DC ports should be enabled/disabled to be used in board .h files.

The code for AXP192 should not have changed, so that should probably be safe to use.

**_!!!The other chips have absolutely NOT been tested on real hardware!!!_**

Status: "it compiles"

**Do NOT run this on real hardware unless you r [... cut]
closed 2025-01-26 by cyberman54