cyberman54/ESP32-Paxcounter #994: Endian Encoding wrong: BMExxx Temperature in Mode PAYLOAD_ENCODER == 2

commeco, 2023-12-30:
Hello,
imho in the case of PAYLOAD_ENCODER == 2 the endian encoding here is wrong.
https://github.com/cyberman54/ESP32-Paxcounter/blob/2e08deb5e60233372e3df9c3fa8f6b972a7f8e87/src/payload.cpp#L287

The lines 287 and 288 have to be switched.

Or am i wrong? :)

Regards Matthias

cyberman54, 2024-01-05:
Why do you think the endianess is wrong? It depends on your decoder. If you use the enclosed encoder, it should work. Float conversion is used for temperature values.

The code was taken from https://github.com/thesolarnomad/lora-serialization/tree/master

commeco, 2024-01-08:
Hi,
the code itself is correct, but it's the same endian here like in PAYLOAD_DECODER == 1.

e.g. PAYLOAD_ENCODER == 1

void PayloadConvert::addVoltage(uint16_t value) {
  buffer[cursor++] = highByte(value);
  buffer[cursor++] = lowByte(value);
}

Definition of the highByte macro
-> #define highByte(w) ((uint8_t) ((w) >> 8))

And in PAYLOAD_DECODER == 2 the order is the same, and that is imho wrong.
  buffer[cursor++] = (byte)((t >> 8) & 0xFF);
  buffer[cursor++] = (byte)t & 0xFF;

Regards
  Matthias

cyberman54, 2024-01-08:
What's the problem here? If it is matching with the logic of the decoder, it has no impact?

commeco, 2024-01-08:
I understand. But i don't use the TTN Decoder backend. We use the Webhook integration and wrote our own "encoder" in PHP, based on the documentation here: https://cyberman54.github.io/ESP32-Paxcounter/payloadformat/
In the documentation isn't mentioned that the temperature float values (Function writeFloat) are always represented in big endian. Anyway, if no one else have a problem, than i solve it on my side. :)

Regards
  Matthias

closed 2024-01-08 by commeco

cyberman54, 2024-01-08:
Thus, we have a documentation problem, not a code issue. I will edit the docs. Thanks for your hint!

commeco, 2024-01-08:
You're welcome. :)