cyberman54/ESP32-Paxcounter #1001: Aobut LilyGo

Neogene, 2024-02-16:
Hi, the repo is already configured to be flashed into a LilyGo 1.x? We bought the device and it has installed version 3.2.1 which is not listed inside the release area of this repo.

It's enough to rename the .orig files (and the other conf files) to the version to be used following the documentation page?

We asked to LilyGo support without any answer and we don't want to flash without knowing if there is some setup to do to avoid bricking it.

any help will be pareciated.

cyberman54, 2024-02-16:
you need to select the appropriate .hal file (look in shared dir of repo) for your board. If you bought the "original" paxcounter board from LilyGo store this will work plug&play, no risk to brick the device.

Neogene, 2024-02-16:
Thank you, yes we brought the original one, It's written on the chipset LilyGo - 433/470Mhz Modle: LORA32, T3_v1.6.1 - 202110104

cyberman54, 2024-02-17:
This one will build & work, but be aware that you have a 433 MHz version.

https://github.com/cyberman54/ESP32-Paxcounter/issues/992

closed 2024-02-17 by cyberman54