cyberman54/ESP32-Paxcounter #1003: GPS not starting and BME680 not sending data on Lilygo v2.1 New

proffalken, 2024-03-07:
Hello :)

I just received my LILYGO V2.1 from the official store on AliExpress and I'm trying to get it working with my Neo 6M GPS dongle.

I've uncommented the required lines, connect the GPS serial as instructed, but it doesn't show in the logs as being connected.

I've also got a BME680 connected to the board, and this *is* recognised on startup, but data is not sent from the board as I'd expect it to.

The data for BLE and WiFi devices *is* recorded and sent to TTN, so I know that part of things is working fine, but I don't see any data on ports 4 (GPS) or 7(BME680).

Have I missed a step in the setup?

### My paxcounter.conf

```
// clang-format off

// ----- Paxcounter user config file ------
//
// --> adapt to your needs and use case <--
//
// Notes:
// 1. After editing, before "build", use "clean" button in PlatformIO!
// 2. Clear NVRAM of the board to delete previous stored runtime settings (pio run -t erase)

// Device options
#define VERBOSE                         1       // set to 0 to silence the device, 1 enables additional debug output
#define BOOTMENU                        0       // 0 = no bootmenu, 1 = device brings up boot menu before starting application
#define BOOTDELAY                       30      // time [seconds] while devices waits in boot menue for input
#define BOOTTIMEOUT                     300     // time [seconds] while devices waits to finish upload a firmware file
#define SDLOGGING                       0       // set to 1 for system logging to SD card (if device has one)

// Payload send cycle and encoding
#define SENDCYCLE                       30      // payload send cycle [seconds/2], 0 .. 255
#define SLEEPCYCLE                      0       // sleep time after a send cycle [seconds/10], 0 .. 65535; 0 means no sleep [default = 0]
#define PAYLOAD_ENCODER                 2       // payload encoder: 1=Plain, 2=Packed, 3=Cayenne LPP dynamic, 4=Cayenne LPP packed
#define COUNTERMODE                     0       // 0=cyclic, 1=cumulative, 2=cyclic confirmed
#define SYNCWAKEUP                      300     // shifts sleep wakeup to top-of-hour, when +/- X seconds off [0=off]

// default settings for transmission of sensor data (first list = data on / second line = data off)
#define PAYLOADMASK                                                                             \
            ((GPS_DATA | MEMS_DATA | COUNT_DATA | SENSOR1_DATA | SENSOR2_DATA | SENSOR3_DATA) & \
            (~BATT_DATA) & (~RESERVED_DATA))

// MAC sniffing settings
#define BLECOUNTER                      1       // set to 0 if you do not want to start the BLE sniffer
#define WIFICOUNTER                     1       // set to 0 if you do not want to start the WIFI sniffer
#define RSSILIMIT                       0       // 0...-128, set to 0 if you do not want to filter signals

// BLE scan parameters
#define BLESCANTIME                     0       // [seconds] scan duration, 0 means infinite [default], see note below
#define BLESCANWINDOW                   80      // [milliseconds] scan window, see below, 3 .. 10240, default 80ms
#define BLESCANINTERVAL                 80      // [illiseconds] scan interval, see below, 3 .. 10240, default 80ms = 100% duty cycle

/* Note: guide for setting bluetooth parameters
*
* |< Scan Window >       |< Scan Window >       | ... |< Scan Window >       |
* |<    Scan Interval   >|<    Scan Interval   >| ... |<    Scan Interval   >|
* |<                   Scan duration                                        >|
*
* Scan duration sets how long scanning should be going on, before starting a new scan cycle. 0 means infinite (default).
* Scan window sets how much of the interval should be occupied by scanning. Should be >= BLESCANINTERVAL.
* Scan interval is how long scanning should be done on each channel. BLE uses 3 channels for advertising.
* -> Adjust these values with power consumption in mind if power is limited.
* -> Scan interval can be changed during runtime by remote comammand.
*/

// WiFi scan parameters
#define WIFI_MY_COUNTRY                 "01"    // select 2-letter locale for Wifi RF settings, e.g. "DE"; use "01" for world safe mode
#define	WIFI_CHANNEL_SWITCH_INTERVAL    50      // [seconds/100] -> 0,5 sec.
[... 377 lines cut ...]
20:01:24.165 > I (270) src/i2c.cpp: Device found at 0x76, type = BME680
20:01:24.165 > I (270) src/i2c.cpp: 2 I2C device(s) found
20:01:24.176 > W (272) /home/mmw/.platformio/packages/framework-arduinoespressif32/libraries/Wire/src/Wire.cpp: Bus already started in Master Mode.
20:01:34.321 > I (10425) src/main.cpp: Starting LED Controller...
20:01:34.322 > I (10425) src/power.cpp: ADC characterization based on reference voltage stored in eFuse
20:01:34.333 > I (10429) src/main.cpp: Starting libpax...
20:01:34.333 > I (10430) src/main.cpp: WIFISCAN: on
20:01:34.333 > I (10433) src/main.cpp: BLESCAN: on
20:01:34.341 > I (10437) .pio/libdeps/usb/libpax/lib/libpax/blescan.cpp: Initializing bluetooth scanner ...
20:01:35.046 > I (11153) .pio/libdeps/usb/libpax/lib/libpax/blescan.cpp: BLE Scanning started
20:01:35.247 > I (11354) .pio/libdeps/usb/libpax/lib/libpax/blescan.cpp: Bluetooth scanner started
20:01:35.260 > I (11363) src/main.cpp: Starting rcommand interpreter...
20:01:35.260 > I (11363) src/rcommand.cpp: Rcommand send queue created, size 55 Bytes
20:01:35.266 > I (11364) src/lorawan.cpp: LORA send queue created, size 530 Bytes
20:01:35.282 > I (11386) src/lorawan.cpp: Starting LMIC...
20:01:35.282 > I (11387) src/main.cpp: Starting Interrupt Handler...
20:01:35.282 > I (11387) src/main.cpp: Starting Timers...
20:01:36.317 > I (12420) src/timekeeper.cpp: Starting time pulse...
20:01:36.317 > I (12421) src/timekeeper.cpp: Timepulse: internal (ESP32 hardware timer)
20:01:37.313 > I (13421) src/main.cpp: Features: DISP LED BATT OTA LORA PACKED TIME
20:01:37.344 > I (13450) src/timesync.cpp: [13.456] Timeserver sync request started, seqNo#131
20:01:44.546 > I (20648) src/lorawan.cpp: DEVaddr: 0x260B3F94 | Network ID: 0x000013 | Network Type: 0
20:01:44.546 > I (20649) src/lorawan.cpp: RSSI: 5 | SNR: 6
20:01:44.553 > I (20650) src/lorawan.cpp: Radio parameters: SF7 | BW125 | CR 4/5
```

proffalken, 2024-03-07:
I've also tried copying the relevant lines into `ttgov21new.h` and that hasn't had an effect either:

```
// clang-format off
// upload_speed 921600
// board esp32dev

#ifndef _TTGOV21NEW_H
#define _TTGOV21NEW_H

#include <stdint.h>

/*  Hardware related definitions for TTGO V2.1 Board
// ATTENTION: check your board version!
// This settings are for boards labeled v1.6 or v1.6.1 on pcb, NOT for v1.5 or older
*/

#define HAS_LORA         1 // comment out if device shall not send data via LoRa
#define CFG_sx1276_radio 1 // HPD13A LoRa SoC

// enable only if you want to store a local paxcount table on the device
#define HAS_SDCARD  2      // this board has a SD MMC card-reader/writer
#define SDCARD_SLOTWIDTH 4 // 4-line interface

#define HAS_DISPLAY 1
#define HAS_LED (25) // green on board LED
//#define HAS_LED NOT_A_PIN

#define BAT_MEASURE_ADC ADC1_GPIO35_CHANNEL // battery probe GPIO pin -> ADC1_CHANNEL_7
#define BAT_VOLTAGE_DIVIDER 2 // voltage divider 100k/100k on board

// Pins for I2C interface of OLED Display
#define MY_DISPLAY_SDA (21)
#define MY_DISPLAY_SCL (22)
#define MY_DISPLAY_RST NOT_A_PIN

// Pins for LORA chip SPI interface, reset line and interrupt lines
#define LORA_SCK  (5) 
#define LORA_CS   (18)
#define LORA_MISO (19)
#define LORA_MOSI (27)
#define LORA_RST  (23)
#define LORA_IRQ  (26)
#define LORA_IO1  (33)
#define LORA_IO2  (32)

// BME680 sensor on I2C bus
#define HAS_BME 1 // Enable BME sensors in general
#define HAS_BME680 GPIO_NUM_21, GPIO_NUM_22 // SDA, SCL
//#define BME680_ADDR BME68X_I2C_ADDR_LOW // connect SDIO of BME680 to GND

// GPS settings
#define HAS_GPS 1 // use on board GPS
#define GPS_SERIAL 9600, SERIAL_8N1, GPIO_NUM_12, GPIO_NUM_15 // UBlox NEO 6M RX, TX
//#define GPS_INT GPIO_NUM_13 // 30ns accurary timepulse, to be external wired on pcb: NEO 6M Pin#3 -> GPIO13
#endif
```

Logs:

```
20:13:04.152 > ets Jun  8 2016 00:22:57
20:13:04.152 > 
20:13:04.152 > rst:0x1 (POWERON_RESET),boot:0x33 (SPI_FAST_FLASH_BOOT)
20:13:04.152 > configsip: 188777542, SPIWP:0xee
20:13:04.152 > clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
20:13:04.163 > mode:DIO, clock div:2
20:13:04.163 > load:0x3fff0030,len:1184
20:13:04.163 > load:0x40078000,len:13232
20:13:04.167 > load:0x40080400,len:3028
20:13:04.167 > entry 0x400805e4
20:13:04.723 > I (16) src/sdcard.cpp: looking for SD-card...
20:13:04.759 > E (46) sdmmc_common: sdmmc_init_ocr: send_op_cond (1) returned 0x107
20:13:04.760 > E (46) vfs_fat_sdmmc: sdmmc_card_init failed (0x107).
20:13:04.760 > I (46) src/sdcard.cpp: No SD-card found (263)
20:13:04.770 > I (50) src/configmanager.cpp: Loading device configuration from NVRAM...
20:13:04.770 > I (58) src/configmanager.cpp: Runtime configuration v3.6.1 loaded
20:13:04.781 > I (63) src/main.cpp: Starting paxcounter_11d2f6dd v3.6.1 (runmode=0 / restarts=0)
20:13:04.792 > I (69) src/main.cpp: code build date: 1709822600
20:13:04.792 > I (74) src/main.cpp: This is ESP32 chip with 2 CPU cores, WiFi/BT/BLE, silicon revision 1, 4MB embedded Flash
20:13:04.804 > I (83) src/main.cpp: Internal Total heap 223900, internal Free Heap 196140
20:13:04.804 > I (90) src/main.cpp: ChipRevision 1, Cpu Freq 240, SDK Version v4.4.5
20:13:04.815 > I (96) src/main.cpp: Flash Size 4194304, Flash Speed 40000000
20:13:04.815 > I (102) src/main.cpp: Wifi/BT software coexist version 1.2.0
20:13:04.826 > I (107) src/main.cpp: IBM LMIC version 1.6.1468577746
20:13:04.826 > I (112) src/main.cpp: Arduino LMIC version 4.1.1.0
20:13:04.837 > I (117) src/lorawan.cpp: DevEUI: E86BEA341754FEFF
20:13:04.837 > I (121) /home/mmw/.platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-i2c.c: Initialising I2C Master: sda=21 scl=22 freq=100000
20:13:04.848 > I (135) src/i2c.cpp: Starting I2C bus scan...
20:13:04.859 > W (138) /home/mmw/.platformio/packages/framework-arduinoespressif32/libraries/Wire/src/Wire.cpp: Bus already started in Master Mode.
20:13:05.982 > I (1269) src/i2c.cpp: Device found at 0x3C, type = Unknown
20:13:05.982 > I (1271) src/i2c.cpp: Device found at 0x76, type = BME680
20:13:05.982 > I (1271) src/i2c.cpp: 2 I2C device(s) found
20:13:05.993 > W (1273) /home/mmw/.platformio/packages/framework-arduinoespressif32/libraries/Wire/src/Wire.cpp: Bus already started in Master Mode.
20:13:16.139 > I (11426) src/main.cpp: Starting LED Controller...
20:13:16.139 > I (11426) src/power.cpp: ADC characterization based on reference voltage stored in eFuse
20:13:16.150 > I (11430) src/main.cpp: Starting libpax...
20:13:16.150 > I (11431) src/main.cpp: WIFISCAN: on
20:13:16.150 > I (11434) src/main.cpp: BLESCAN: on
20:13:16.159 > I (11438) .pio/libdeps/usb/libpax/lib/libpax/blescan.cpp: Initializing bluetooth scanner ...
20:13:16.868 > I (12158) .pio/libdeps/usb/libpax/lib/libpax/blescan.cpp: BLE Scanning started
20:13:17.069 > I (12359) .pio/libdeps/usb/libpax/lib/libpax/blescan.cpp: Bluetooth scanner started
20:13:17.081 > I (12368) src/main.cpp: Starting rcommand interpreter...
20:13:17.082 > I (12368) src/rcommand.cpp: Rcommand send queue created, size 55 Bytes
20:13:17.088 > I (12369) src/lorawan.cpp: LORA send queue created, size 530 Bytes
20:13:17.104 > I (12391) src/lorawan.cpp: Starting LMIC...
20:13:17.105 > I (12392) src/main.cpp: Starting Interrupt Handler...
20:13:17.105 > I (12393) src/main.cpp: Starting Timers...
20:13:18.139 > I (13425) src/timekeeper.cpp: Starting time pulse...
20:13:18.139 > I (13426) src/timekeeper.cpp: Timepulse: internal (ESP32 hardware timer)
20:13:19.134 > I (14426) src/main.cpp: Features: DISP LED BATT OTA LORA PACKED TIME
20:13:19.166 > I (14455) src/timesync.cpp: [14.461] Timeserver sync request started, seqNo#77
```

proffalken, 2024-03-07:
I spotted the bit in platform.io about running `platformio run -t clean` so did that and now I have an MD5 mismatch:

```
Wrote 1338960 bytes (856697 compressed) at 0x00010000 in 10.2 seconds (effective 1054.8 kbit/s)...
File  md5: a77fa4d1b614c30ef9dd28ac017ccdfa
Flash md5: 6422caf17eddc9261c09d6955031d580
MD5 of 0xFF is bf1d28f6275e0a4b6539ff8e3b7b7a98

A fatal error occurred: MD5 of file does not match data in flash!
*** [upload] Error 2
```

This appears to be [because the GPS is connected on pin12](https://www.esp32.com/viewtopic.php?t=4061), so I removed that and flashed it successfully, but I still don't see the GPS being started in the logs, and I'm still not getting BME680 data :(

cyberman54, 2024-03-08:
1. Check your GPS wiring: RX/TX swapped?
2. Clear NVRAM by pio run -t erase
3. Switch on GPS and BME data transmission by using rcommand 0x03 0x01 0x21 and 0x15 0x01 0x21

proffalken, 2024-03-08:
Thanks.

GPS wiring is correct (have checked and rechecked multiple times!) and after cleaning, clearing the NVRAM, and rebuilding logs now show:

```
12:48:10.232 > I (11373) src/gpsread.cpp: Opening serial GPS
12:48:10.448 > I (11594) src/main.cpp: Starting GPS Feed...
```

The GPS doesn't have a fix yet, so no data is being sent, but my understanding is that this is correct operating procedure.

The BME is also enabled, but I'm getting the following in the logs that means all the readings are zero:

```
12:48:10.500 > I (11645) src/bmesensor.cpp: BSEC v1.4.9.2
12:48:10.500 > I (11647) src/bmesensor.cpp: no BSEC state stored in NVRAM, starting sensor with defaults
12:48:10.512 > E (11647) src/bmesensor.cpp: BME680 error -2
12:48:10.512 > E (11650) src/main.cpp: BME sensor could not be initialized

```

This [appears to be a comms error](https://github.com/mongoose-os-libs/bme680/blob/master/BSEC_1.4.7.4_Generic_Release/bsec/src/bme680/bme680_defs.h#L135) so I'll see if I've got a 280 or similar that I can swap in and test the connections with because the wiring is definitely correct.

Is there a list of the commands for enabling things like the SDS011 sesnsor? [The manual](https://cyberman54.github.io/ESP32-Paxcounter/remotecontrol/) seems to cover GPS, BME, and various other things, but not the SDS011 (unless that falls under "custom sensors"?)

proffalken, 2024-03-08:
OK, this looks like a pin config thing.

The defaults in the code suggest that the SDS011 should be on pins 19 & 23, however these are already in use on the TTGO v2.1 as MSIO and RST for the LORA module.

I'm going through and trying different pins to find a combination that works for the SDS011, then I'll come back to the GPS and the BME680 to work out the pins for those.

The BME680 is the one that's throwing me at the moment though, because that's I2C so it shouldn't matter that the OLED is already on that bus, and indeed I can see in my logs that both devices are picked up

cyberman54, 2024-03-08:
BME680: you're probably using wrong chip address. Try enabling this line in ttgov21new.h file:

//#define BME680_ADDR BME68X_I2C_ADDR_LOW // connect SDIO of BME680 to GND

cyberman54, 2024-03-08:
GPS: as long as it has no fix, there will be no data sent. Go outside to get a fix. Can take up to 20 minutes after a cold start.

cyberman54, 2024-03-08:
SDS011: not sure if ttgov21 board has enough free GPIOs if you 're already using two GPIOs for serial GPS. Check the pinout. If you find free GPIOs you can drive SDS011 on this. Paxcounter firmware will pickup data of all sensors and GPS in parallel, no limitation.

cyberman54, 2024-03-08:
If you don't need wifi/ble sniffing in you application (e.g. using sensors only), you should disable it in paxcounter.conf to save power and cpu time.

proffalken, 2024-03-08:
> //#define BME680_ADDR BME68X_I2C_ADDR_LOW // connect SDIO of BME680 to GND

Thanks, I'll do that tomorrow. FWIW I'm using 0x76 which is what it displays as in the logs when it's detected.  Do I need to actually solder SDIO to GND in order for this to work?

> SDS data is always sent, if an SDS011 is present. SDS payload is never masked, so no need and no option to switch on/off.

Thanks again, good to know, I'll check this out tomorrow as well.

> If you don't need wifi/ble sniffing in you application (e.g. using sensors only), you should disable it in paxcounter.conf to save power and cpu time.

Of all the sensors I can do with out, it's probably the BME680, and then the GPS (this won't be mobile, it's just useful to have incase someone moves it).

This is for a "how many people are in the area vs. what's the air quality like" project, so the SDS011 and BLE/WiFi tracking are essential.

Just want to say thank you for both the project and your time btw - I remember seeing this idea way back in 2019 when I was running a LoRaWAN consultancy and wanted to use it back then but couldn't get the funding. It's an awesome project and I really appreciate the effort you've put into it! :)

> GPS: as long as it has no fix, there will be no data sent. Go outside to get a fix. Can take up to 20 minutes after a cold start.

I'll try again tomorrow, but I had it outside for three hours with no data being sent.  The "fix" LED on the NEO 6M module is flashing though, which suggests it is seeing something, but switching the TX/RX around doesn't seem to make a difference.  I'll try a test sketch tomorrow as well to try and rule out the module.

cyberman54, 2024-03-09:
BME680: I2c scanner detects your BME module at 0x76 = lower address, so probably your module does connect SDIO to GND. Thus, you must change the chip address on software side.

GPS: blinking LED on Neo 6M boards usually shows PPS pulse, this must not indicate it has a fix.

proffalken, 2024-03-09:
OK, I dropped the GPS and changed the code for the BME680 back to the one in the generic.h file.

I'm getting readings from the BME680 sensor now (along with a load of "error 263" in the logs that don't seem to be causing any harm!), and I can hear the SDS011 spinning up and taking readings so I'll leave it for a while and see if the zeros that it's sending back change after a few readings.

cyberman54, 2024-03-09:
What is error 263, from where? BME280 lib/driver?

proffalken, 2024-03-09:
Yes, seems to be, but the readings are getting through:

```
13:03:24.771 > I (9585549) src/cyclic.cpp: BME680 Temp: 15.97°C | IAQ: 50.00 | IAQacc: 0
13:03:27.740 > E (9588514) /home/mmw/.platformio/packages/framework-arduinoespressif32/libraries/Wire/src/Wire.cpp: i2cRead returned Error 263
13:03:30.740 > E (9591514) /home/mmw/.platformio/packages/framework-arduinoespressif32/libraries/Wire/src/Wire.cpp: i2cRead returned Error 263
13:03:33.740 > E (9594514) /home/mmw/.platformio/packages/framework-arduinoespressif32/libraries/Wire/src/Wire.cpp: i2cRead returned Error 263
13:03:49.665 > I (9610444) src/lorawan.cpp: 8 byte(s) sent to LORA
13:03:53.734 > I (9614511) src/cyclic.cpp: BME680 Temp: 15.60°C | IAQ: 50.00 | IAQacc: 0

```

Still nothing from the SDS011, but I'm wondering if that's a dodgy connection - I'm going to try some new wires in a bit

cyberman54, 2024-03-09:
Please carefully check your wirings and correct settings of you sensors, before opening issues here. Thank you.

proffalken, 2024-03-09:
OK, I'll close this for now.  The SDS011 still isn't working as expected (The fan starts/stops, but no readings are sent), and when I reboot the TTGO v2.1 with the BME680 ground pin connected then it sometimes fails to read the values, but I'll get it moved over onto a PCB that I'm designing specifically for this and see if that makes a difference.

Thanks again for your patience and help.

closed 2024-03-09 by proffalken

proffalken, 2024-03-11:
For anyone else hitting similar issues, check the output voltage pins and GND pins on your board with a multimeter.

I found that on the board I ordered, the 5VDC pin was non-functional, and the GND closest to the 3.3v pin was also non-functional.

Once I'd moved the power supplies for the BME680 and SDS011 off the board and direct to the source, with the battery connector providing the power/GND for both, then everything sprang into life and started working.

cyberman54, 2024-03-11:
Where did you purchase that board, from original LilyGo Shop, or elsewhere? Can you post a high resolution picture of that board here?

proffalken, 2024-03-11:
Sure, https://www.aliexpress.com/item/32915894264.html is the item I purchased, looks like it's from the official LilyGo store.

I've just zoomed right in and I can see that not all of the pad has solder on it where I was trying to avoid burning the OLED cable whilst attaching the pins, so it could well be that causing the issue here rather than a manufacturing fault - if anyone has advice on how to solder underneath that ribbon without damaging it I'd love to hear it! :)

[image]
[image]

cyberman54, 2024-03-11:
This is the "original", and problems with non functional GND / 5V pins were - yet - never reported here, so a good point to check your soldering.