cyberman54/ESP32-Paxcounter #1005: Crash when running on  TTGOV2 (TXCO version)

nachoplus, 2024-03-20:
I am not able to run on TTGOv2 TXCO version. It compile but crush at run time:

22:54:00.194 > Rebooting...
22:54:00.205 > ets Jun  8 2016 00:22:57
22:54:00.205 > 
22:54:00.205 > rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
22:54:00.205 > configsip: 188777542, SPIWP:0xee
22:54:00.205 > clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
22:54:00.216 > mode:DIO, clock div:2
22:54:00.216 > load:0x3fff0030,len:1184
22:54:00.216 > load:0x40078000,len:13232
22:54:00.221 > load:0x40080400,len:3028
22:54:00.221 > entry 0x400805e4
22:54:00.803 > I (25) src/sdcard.cpp: looking for SD-card...
22:54:00.840 > E (55) sdmmc_common: sdmmc_init_ocr: send_op_cond (1) returned 0x107
22:54:00.840 > E (55) vfs_fat_sdmmc: sdmmc_card_init failed (0x107).
22:54:00.840 > I (55) src/sdcard.cpp: No SD-card found (263)
22:54:00.852 > I (59) src/configmanager.cpp: Loading device configuration from NVRAM...
22:54:00.852 > I (67) src/configmanager.cpp: Runtime configuration v3.6.1 loaded
22:54:00.862 > I (72) src/main.cpp: Starting paxcounter_46157031 v3.6.1 (runmode=0 / restarts=18)
22:54:00.873 > I (78) src/main.cpp: code build date: 1710970318
22:54:00.873 > I (83) src/main.cpp: This is ESP32 chip with 2 CPU cores, WiFi/BT/BLE, silicon revision 1, 4MB embedded Flash
22:54:00.884 > I (92) src/main.cpp: Internal Total heap 223804, internal Free Heap 195576
22:54:00.884 > I (99) src/main.cpp: ChipRevision 1, Cpu Freq 240, SDK Version v4.4.5
22:54:00.895 > I (105) src/main.cpp: Flash Size 4194304, Flash Speed 40000000
22:54:00.907 > I (111) src/main.cpp: Wifi/BT software coexist version 1.2.0
22:54:00.907 > I (116) src/main.cpp: IBM LMIC version 1.6.1468577746
22:54:00.907 > I (121) src/main.cpp: Arduino LMIC version 4.1.1.0
22:54:00.918 > I (126) src/lorawan.cpp: DevEUI: HIDDEN
22:54:00.918 > I (130) /home/nacho/.platformio/packages/framework-arduinoespressif32@3.20011.230801/cores/esp32/esp32-hal-i2c.c: Initialising I2C Master: sda=21 scl=22 freq=100000
22:54:00.940 > I (145) src/i2c.cpp: Starting I2C bus scan...
22:54:00.940 > W (148) /home/nacho/.platformio/packages/framework-arduinoespressif32@3.20011.230801/libraries/Wire/src/Wire.cpp: Bus already started in Master Mode.
22:54:01.066 > I (281) src/i2c.cpp: Device found at 0x3C, type = Unknown
22:54:01.066 > I (281) src/i2c.cpp: 1 I2C device(s) found
22:54:01.066 > W (282) /home/nacho/.platformio/packages/framework-arduinoespressif32@3.20011.230801/libraries/Wire/src/Wire.cpp: Bus already started in Master Mode.
22:54:11.216 > I (10431) src/main.cpp: Starting LED Controller...
22:54:11.216 > I (10431) src/power.cpp: ADC characterization based on reference voltage stored in eFuse
22:54:11.227 > I (10435) src/main.cpp: Starting libpax...
22:54:11.227 > I (10436) src/main.cpp: WIFISCAN: on
22:54:11.227 > I (10439) src/main.cpp: BLESCAN: off
22:54:11.333 > I (10548) src/main.cpp: Starting rcommand interpreter...
22:54:11.333 > I (10548) src/rcommand.cpp: Rcommand send queue created, size 55 Bytes
22:54:11.340 > I (10549) src/lorawan.cpp: LORA send queue created, size 530 Bytes
22:55:11.228 > 
22:55:11.229 > assert failed: xTaskGenericNotify tasks.c:5545 (xTaskToNotify)
22:55:11.229 > 
22:55:11.229 > 
22:55:11.229 > Backtrace: 0x40083fcd:0x3ffbd630 0x4009774d:0x3ffbd650 0x4009e1bd:0x3ffbd670 0x4009a76e:0x3ffbd7a0 0x400d81e3:0x3ffbd7c0 0x400d88be:0x3ffbd7e0 0x4009b0a7:0x3ffbd800

May be due to a diferentes pin conections. Futher investigations have to be done ...
Thanks for contributing this amazing soft.

cyberman54, 2024-03-21:
Could you please ESP32 debugging tool to match the backtrace with lines of code?

cyberman54, 2024-03-21:
Probably your purchased board has not a SX1276x LoRa chip, but the newer SX1262. If this is the case, it is not compatible with paxcounter software, because the LMIC LoRaWAN library does not support SX1262.

If you don't need LoRA, you can disable LoRA code in paxcounter by excluding the define HAS_LORA. Then it should work.

nachoplus, 2024-03-21:
It is a SX1276 chip, the TCXO 868Mhz variant. This one: https://www.lilygo.cc/products/lora3?variant=43365016567989

In [https://github.com/Xinyuan-LilyGO/LilyGo-LoRa-Series/blob/master/examples/RadioLibExamples/SX1276/SX1276_Transmit_Interrupt/utilities.h ](https://github.com/Xinyuan-LilyGO/LilyGo-LoRa-Series/blob/master/examples/RadioLibExamples/SX1276/SX1276_Transmit_Interrupt/utilities.h#L155) it showed the diferent pinout
 
Regarding to "Could you please ESP32 debugging tool to match the backtrace with lines of code?" I have to learnt how to do it. I'll update the information when I do it.

Disabling HAS_LORA works but I need Lora.

thanks for the support

nachoplus, 2024-03-21:
**Regarding the match of backtrace with the code here I pasted the result of the following cmd:**
~/.platformio/packages/toolchain-xtensa32/bin/xtensa-esp32-elf-addr2line   -pfiaC   -e /home/nacho/work/PlatformIO/ESP32-Paxcounter/.pio/build/usb/firmware_ttgov21newTCXO_v3.6.1.elf 0x40083fd5:0x3ffbd640 0x40097755:0x3ffbd660 0x4009e1c5:0x3ffbd680 0x4009a776:0x3ffbd7b0 0x400d8d2b:0x3ffbd7d0 0x400d94ea:0x3ffbd7f0 0x4009b0af:0x3ffbd810

**result:**
0x40083fd5: panic_abort at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_system/panic.c:423
0x40097755: esp_system_abort at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_system/esp_system.c:128
0x4009e1c5: __assert_func at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/newlib/assert.c:91
0x4009a776: xTaskGenericNotify at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/freertos/esp_additions/private_include/freertos_tasks_c_additions.h:43
0x400d8d2b: setSendIRQ() at /home/nacho/work/PlatformIO/ESP32-Paxcounter/src/senddata.cpp:10
0x400d94ea: report(void*) at /home/nacho/work/PlatformIO/ESP32-Paxcounter/.pio/libdeps/usb/libpax/lib/libpax/libpax_api.cpp:86
0x4009b0af: prvProcessExpiredTimer at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/freertos/timers.c:523
 (inlined by) prvProcessTimerOrBlockTask at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/freertos/timers.c:634
 (inlined by) prvTimerTask at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/freertos/timers.c:597

As far as I understand the fail occurs in ESP32-Paxcounter/src/senddata.cpp:10

void setSendIRQ(void) { xTaskNotify(irqHandlerTask, SENDCYCLE_IRQ, eSetBits); }

cyberman54, 2024-03-22:
It seems TCXO version needs GPIO33 for the oscillator. And LoRa DIO1 pin, which is needed for operation with LMIC LoRaWAN library, is not connected. That means the board cannot work with LMIC.

cyberman54, 2024-04-14:
No fix availabe, due to hardware limitation.
See https://github.com/Xinyuan-LilyGO/LilyGo-LoRa-Series/issues/148