cyberman54/ESP32-Paxcounter #1007: No connection to lorawan

Edd77, 2024-04-07:
I have a Lilygo T3_V1.6.1 2010104 board and wanted to install the latest npaxcounter 3.6.1 with Visal Studio 1.88 and platcormio core v6.1.14. I did everything as in the instructions

This error occurs during the build

.pio/libdeps/usb/MCCI LoRaWAN LMIC library/src/hal/getpinmap_thisboard.cpp: In function 'const Arduino_LMIC::HalPinmap_t* Arduino_LMIC::GetPinmap_ThisBoard()':
.pio/libdeps/usb/MCCI LoRaWAN LMIC library/src/hal/getpinmap_thisboard.cpp:71:72: note: #pragma message: Board not supported -- use an explicit pinmap
         #pragma message("Board not supported -- use an explicit `pinmap")
                                                                        ^

APPEUI and APP KEY are missing when I transfer it to the lilygo

--- Terminal on COM3 | 115200 8-N-1
--- Available filters and text transformations: colorize, debug, default, direct, esp32_exception_decoder, hexlify, log2file, nocontrol, printable, send_on_enter, time
--- More details at https://bit.ly/pio-monitor-filters
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H
17:36:44.685 > ets J]␋␂�␂201␓��Ғ�:5�C�HH���0� (P�UI=�_RE�␔U�I����0x13�SP�e␅MQ_FL�*�%=�T)����˥�ͥ�: 1������2,�*�]A:0x�YC��k_d���0,qߖ��0x00��E��:0x�,csz���i0x0bhdߖ��0x00�␗}��v:0��j
17:36:44.685 > [␋��DIO�␚���k dZ��j
17:36:44.685 > loa����fff���bb���1184C!�+��0�␓���00����i13�&�H�녑�0x4�80��bb�n:3�&j�Y␋�� 0x���e4��I (392) src/sdcard.cpp: looking for SD-card...
17:36:44.685 > E (422) sdmmc_common: sdmmc_init_ocr: send_op_cond (1) returned 0x107
17:36:44.685 > E (422) vfs_fat_sdmmc: sdmmc_card_init failed (0x107).
17:36:44.685 > I (422) src/sdcard.cpp: No SD-card found (263)
17:36:44.685 > I (426) src/configmanager.cpp: Loading device configuration from NVRAM...
17:36:44.685 > I (434) src/configmanager.cpp: Runtime configuration v3.6.1 loaded
17:36:44.685 > D (439) src/reset.cpp: Timezone set to CET-1CEST,M3.4.0/2,M10.4.0/3
17:36:44.685 > I (445) src/main.cpp: Starting paxcounter_dd1fd171 v3.6.1 (runmode=0 / restarts=0)
17:36:44.685 > I (452) src/main.cpp: code build date: 1712507633
17:36:44.685 > I (456) src/main.cpp: This is ESP32 chip with 2 CPU cores, WiFi/BT/BLE, silicon revision 1, 4MB embedded Flash
17:36:44.685 > I (466) src/main.cpp: Internal Total heap 223660, internal Free Heap 195900
17:36:44.685 > I (473) src/main.cpp: ChipRevision 1, Cpu Freq 240, SDK Version v4.4.5
17:36:44.685 > I (479) src/main.cpp: Flash Size 4194304, Flash Speed 40000000
17:36:44.685 > I (485) src/main.cpp: Wifi/BT software coexist version 1.2.0
17:36:44.685 > I (490) src/main.cpp: IBM LMIC version 1.6.1468577746
17:36:44.685 > I (495) src/main.cpp: Arduino LMIC version 4.1.1.0
17:36:44.685 > I (500) src/lorawan.cpp: DevEUI: 308398DCBC38FEFE
17:36:44.685 > I (504) C:/Users/user123/.platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-i2c.c: Initialising I2C Master: sda=21 scl=22 freq=100000
17:36:44.685 > I (518) src/i2c.cpp: Starting I2C bus scan...
17:36:44.685 > W (521) C:/Users/user123/.platformio/packages/framework-arduinoespressif32/libraries/Wire/src/Wire.cpp: Bus already started in Master Mode.
17:36:44.685 > I (653) src/i2c.cpp: Device found at 0x3C, type = Unknown
17:36:44.685 > I (653) src/i2c.cpp: 1 I2C device(s) found
17:36:44.685 > W (654) C:/Users/user123/.platformio/packages/framework-arduinoespressif32/libraries/Wire/src/Wire.cpp: Bus already started in Master Mode.
17:36:49.699 > I (10801) src/main.cpp: Starting LED Controller...
17:36:49.705 > I (10801) src/power.cpp: ADC characterization based on reference voltage stored in eFuse
17:36:49.716 > I (10806) src/main.cpp: Starting libpax...
17:36:49.716 > I (10806) src/main.cpp: WIFISCAN: on
17:36:49.722 > I (10809) src/main.cpp: BLESCAN: on
17:36:49.727 > I (10813) .pio/libdeps/usb/libpax/lib/libpax/blescan.cpp: Initializing bluetooth scanner ...
17:36:50.429 > I (11530) .pio/libdeps/usb/libpax/lib/libpax/blescan.cpp: BLE Scanning started
17:36:50.630 > I (11731) .pio/libdeps/usb/libpax/lib/libpax/blescan.cpp: Bluetooth scanner started
17:36:50.638 > I (11740) src/main.cpp: Starting rcommand interpreter...
17:36:50.645 > I (11740) src/rcommand.cpp: Rcommand send queue created, size 55 Bytes
17:36:50.656 > I (11740) src/lorawan.cpp: LORA send queue created, size 530 Bytes
17:36:50.663 > I (11764) src/lorawan.cpp: Starting LMIC...
17:36:50.668 > D (11765) src/lorawan.cpp: JOINING
17:36:50.674 > I (11765) src/main.cpp: Starting Interrupt Handler...
17:36:50.679 > I (11766) src/main.cpp: Starting Timers...
17:36:50.706 > I (11801) src/timekeeper.cpp: Starting time pulse...
17:36:50.706 > I (11802) src/timekeeper.cpp: Timepulse: internal (ESP32 hardware timer)
17:36:51.449 > D (12550) src/lorawan.cpp: TXSTART
17:36:51.707 > I (12802) src/main.cpp: Features: DISP LED BATT OTA LORA PLAIN TIME
17:36:51.730 > I (12831) src/timesync.cpp: [12.460] Timeserver sync request started, seqNo#102
17:36:56.502 > D (17603) src/lorawan.cpp: RXSTART
17:36:57.501 > D (18602) src/lorawan.cpp: RXSTART
17:36:57.743 > D (18844) src/lorawan.cpp: JOIN_WAIT       
17:37:21.735 > D (42831) src/cyclic.cpp: Heap: Free:96544, Min:88264, Size:221364, Alloc:86004, StackHWM:2120
17:37:21.741 > D (42831) src/cyclic.cpp: IRQhandler 2120 bytes left | Taskstate = 0
17:37:21.747 > D (42835) src/cyclic.cpp: Rcommand interpreter 2528 bytes left | Taskstate = 2
17:37:21.752 > D (42842) src/cyclic.cpp: LMiCtask 2120 bytes left | Taskstate = 1
17:37:21.758 > D (42847) src/cyclic.cpp: Lorasendtask 2512 bytes left | Taskstate = 2
17:37:21.769 > D (42854) src/cyclic.cpp: LEDloop 424 bytes left | Taskstate = 1
17:37:49.734 > D (70830) src/senddata.cpp: Sending count results: pax=6 / wifi=1 / ble=5
17:37:49.740 > D (70833) src/senddata.cpp: sending Payload for Port 1
17:37:51.735 > D (72829) src/cyclic.cpp: Heap: Free:96544, Min:88264, Size:221364, Alloc:86004, StackHWM:2120
17:37:51.740 > D (72830) src/cyclic.cpp: IRQhandler 2120 bytes left | Taskstate = 0
17:37:51.746 > D (72833) src/cyclic.cpp: Rcommand interpreter 2528 bytes left | Taskstate = 2
17:37:51.751 > D (72840) src/cyclic.cpp: LMiCtask 2120 bytes left | Taskstate = 1
17:37:51.757 > D (72846) src/cyclic.cpp: Lorasendtask 2512 bytes left | Taskstate = 2
17:37:51.768 > D (72852) src/cyclic.cpp: LEDloop 424 bytes left | Taskstate = 1
17:37:58.751 > D (79851) src/lorawan.cpp: TXSTART        1
17:38:03.801 > D (84902) src/lorawan.cpp: RXSTART        1
17:38:04.806 > D (85903) src/lorawan.cpp: RXSTART        1
17:38:05.045 > D (86145) src/lorawan.cpp: JOIN_WAIT      1
17:38:21.730 > D (102830) src/cyclic.cpp: Heap: Free:96544, Min:88264, Size:221364, Alloc:86004, StackHWM:2120
17:38:21.741 > D (102831) src/cyclic.cpp: IRQhandler 2120 bytes left | Taskstate = 0
17:38:21.747 > D (102834) src/cyclic.cpp: Rcommand interpreter 2528 bytes left | Taskstate = 2
17:38:21.753 > D (102841) src/cyclic.cpp: LMiCtask 2120 bytes left | Taskstate = 1
17:38:21.758 > D (102847) src/cyclic.cpp: Lorasendtask 2512 bytes left | Taskstate = 2
17:38:21.769 > D (102854) src/cyclic.cpp: LEDloop 424 bytes left | Taskstate = 1
17:38:49.731 > D (130831) src/senddata.cpp: Sending count results: pax=6 / wifi=2 / ble=4
17:38:49.742 > D (130834) src/senddata.cpp: sending Payload for Port 1
17:38:51.735 > D (132829) src/cyclic.cpp: Heap: Free:96544, Min:88264, Size:221364, Alloc:86004, StackHWM:2120
17:38:51.741 > D (132830) src/cyclic.cpp: IRQhandler 2120 bytes left | Taskstate = 0
17:38:51.746 > D (132834) src/cyclic.cpp: Rcommand interpreter 2528 bytes left | Taskstate = 2
17:38:51.752 > D (132841) src/cyclic.cpp: LMiCtask 2120 bytes left | Taskstate = 1
17:38:51.758 > D (132847) src/cyclic.cpp: Lorasendtask 2512 bytes left | Taskstate = 2
17:38:51.769 > D (132853) src/cyclic.cpp: LEDloop 424 bytes left | Taskstate = 1
17:39:09.784 > D (150884) src/lorawan.cpp: TXSTART        2
17:39:14.836 > D (155936) src/lorawan.cpp: RXSTART        2
17:39:15.839 > D (156935) src/lorawan.cpp: RXSTART        2
17:39:16.081 > D (157177) src/lorawan.cpp: JOIN_WAIT      2
17:39:21.737 > D (162830) src/cyclic.cpp: Heap: Free:96544, Min:88264, Size:221364, Alloc:86004, StackHWM:2120
17:39:21.742 > D (162831) src/cyclic.cpp: IRQhandler 2120 bytes left | Taskstate = 0
17:39:21.747 > D (162834) src/cyclic.cpp: Rcommand interpreter 2528 bytes left | Taskstate = 2

where is my mistake?

cyberman54, 2024-04-07:
You can ignore the pinmap warning.
You must edit loraconf.h and put your APPEUI and APPKEY in it. Then reflash.

Edd77, 2024-04-07:
Yes i have ...

[image]

cyberman54, 2024-04-07:
looks like you have a problem with your toolchain.

Edd77, 2024-04-07:
Sorry, I'm not that familiar with visual studio yet. Could you help me pleas?

Edd77, 2024-04-07:
[image]

cyberman54, 2024-04-07:
you have no keys compiled, thus there is not network join. Again, check your toolchain. "u1_t" unknown makes this issue.

Edd77, 2024-04-07:
the only toolchain is platformio ?
[image]

cyberman54, 2024-04-07:
I would suggest you try a fresh installation of your toolchain from scratch.
You may use the setup from CI/CD workflow, see [here.](https://github.com/cyberman54/ESP32-Paxcounter/actions/workflows/build.yml)

I'm closing this issue, because CI/CD is passing, thus your issue is not a project issue.

closed 2024-04-07 by cyberman54