cyberman54/ESP32-Paxcounter #1008: Support heltec S 3 v3.1 version 3.6.2

mhuelsmann, 2024-06-05:
Hello,

i am unable to compile version 3.6.2 for heltec esp32-s3 v3.1 hardware. I try all 3 possible options in platfomio.ini 

halfile = ttgotdongles3.h
;halfile = ttgotdongledisplays3.h
;halfile = ttgotdisplays3.h

does this version only work for ttgo?
regards michael

mhuelsmann, 2024-06-05:
Is there an solution für sx1262 Lora chip?

cyberman54, 2024-06-05:
Paxcounter software should work on ESP32 S3, but there is no halfile for heltec esp32-s3 board yet. If you figure out the right pinput, it should run. You're welcome to raise a PR for this.

But there is no support for SX1262 in LMIC library, which paxcounter is based on. You may find patches on github on your own. Without paxcounter is not running on SX1262.

oseiler2, 2024-06-10:
FWIW, I've been using this branch with a newer TBeam S3 (SX1262) and can confirm that it works well. It includes a board definition for the Heltec
https://github.com/mcci-catena/arduino-lmic/pull/949

cyberman54, 2024-06-10:
Reopened this. Anyone willing to create a PR for Paxcounter SX1262?

TristanWebber, 2024-06-25:
@cyberman54 it looks like it would be possible to update for compatibility with SX126x by:

- Pointing to my fork of arduino-lmic in the platformio.ini file (I have made a PR in the MCCI master but it has not been merged)
- Updating the `src/lorawan.cpp` file to deal with the SX126x features that differ from SX127x
- Adding pinmaps to `shared/hal`

However, pointing to my fork rather than the MCCI master arduino-lmic library might not be ideal. What are your thoughts?

cyberman54, 2024-06-25:
Yes, that's the way to go.
I ordered an SX1262 based T-Beam for testing. If it works, i will ask Terry Moore if he is willing to add your fork as a branch to his MCCI repository. If not, i will check license and if okay i can fork it.

cyberman54, 2024-06-30:
@TristanWebber it works with your lmic branch! Thank you very much for this enhancement!
Tested with a TTGO T-Beam Supreme S3 on TTN network EU868.
See this commit: https://github.com/cyberman54/ESP32-Paxcounter/commit/b0c193839c32844c6965a124128e5eb861adea5c

TristanWebber, 2024-06-30:
Great news! I'm glad it is working for you and that it is useful to others. Let me know if there are any dramas.

At this point I know for sure it has been used for a couple of AU915 and a couple of EU868 examples so it's still relatively early days with knowing for sure how it performs for all bandplans. Similarly it has been tested with a small number of boards but over time I'm sure I'll hear feedback if / when there are bugs.

cyberman54, 2025-01-26:
HeltecV3 and Heltec Wifi Lora Stick Lite V3 are now supported.
/cc @mhuelsmann

closed 2025-01-26 by cyberman54

kskenyon, 2025-03-03:
My Heltec Lora ESP32 V3 will not transmit or display the battery voltage.  As far as I can tell it is enabled.  The display shows No Batt and the output from port 8 shows 0 voltage.  Must be missing something.  Set to use GPIO0 ADC1.  Don't see where the board contains a dedicated power controller.

Kevin

stiopicz, 2025-03-06:
> HeltecV3 and Heltec Wifi Lora Stick Lite V3 are now supported. /cc [@mhuelsmann](https://github.com/mhuelsmann)

Hi, what [board] shall be used for HeltecV3?

cyberman54, 2025-03-06:
> > HeltecV3 and Heltec Wifi Lora Stick Lite V3 are now supported. /cc [@mhuelsmann](https://github.com/mhuelsmann)
> 
> Hi, what [board] shall be used for HeltecV3?

use platformio_orig_s3.ini and select 
halfile = heltecv3.h
or
halfile = heltecsticklitev3.h

stiopicz, 2025-03-07:
> > > HeltecV3 and Heltec Wifi Lora Stick Lite V3 are now supported. /cc [@mhuelsmann](https://github.com/mhuelsmann)
> > 
> > 
> > Hi, what [board] shall be used for HeltecV3?
> 
> use platformio_orig_s3.ini and select halfile = heltecv3.h or halfile = heltecsticklitev3.h

Thanks, now:
      Missing file ...\ESP32-Paxcounter-3.6.3\shared\hal\heltecv3.h, please create it!
Where I can find it?

cyberman54, 2025-03-07:
The file is in the repo in that path. If you get this error, something with your toolchain is not working properly.

cyberman54, 2025-03-11:
Heltec V3 has no power management chip, but a battery probe pin with a voltage divider (390k/100k) on GPIO1 of ESP32, this is ADC Channel 0.

https://heltec.org/project/wifi-lora-32-v3/

This is how the hal file is configured. So it should work. Please check the value for the voltage divider, maybe it's wrong. Also please check that you turn on battery messages. Remember that settings are stored in NVRAM and persist. Try a clean of NVRAM flash if unsure.

Please report any findings here, for further debugging. Thanks.

stiopicz, 2025-03-11:
unfortunately can't even compile the setup - getting different errors, looks like my platformio skills are not enough

kskenyon, 2025-03-13:
// clang-format off
// upload_speed 921600
// board heltec_wifi_lora_32_V3

#ifndef _HELTECV3_H
#define _HELTECV3_H

#include <stdint.h>

// Hardware related definitions for Heltec V3 LoRa-32 S3 Board
// see https://resource.heltec.cn/download/WiFi_LoRa_32_V3

// LORA settings
#define HAS_LORA 1 // use on board LORA
#define CFG_sx1262_radio 1 // HPD16A
// Pins for LORA chip SPI interface, reset line and interrupt lines
#define LORA_IRQ DIO0
#define LORA_IO1 LMIC_UNUSED_PIN
#define LORA_IO2 LMIC_UNUSED_PIN
#define LORA_SCK SCK
#define LORA_MISO MISO
#define LORA_MOSI MOSI
#define LORA_RST RST_LoRa
#define LORA_BUSY BUSY_LoRa
#define LORA_CS SS

#define HAS_DISPLAY 1
#define HAS_LED LED_BUILTIN           
#define HAS_BUTTON 0

#define BAT_MEASURE_ADC ADC1_CHANNEL_0 // battery probe pin is GPIO1
#define BAT_VOLTAGE_DIVIDER 4 // voltage divider 100k/390k on board

// switches battery power and Vext, switch logic 0 = on / 1 = off
#define EXT_POWER_SW Vext
#define EXT_POWER_ON 0

// Pins for I2C interface of OLED Display
#define MY_DISPLAY_SDA SDA_OLED
#define MY_DISPLAY_SCL SCL_OLED
#define MY_DISPLAY_RST RST_OLED

#endif

This doesn’t work.  Shows NO BATT on the screen.

Kevin

cyberman54, 2025-03-15:
@kskenyon battery probing in heltec v2 now fixed by https://github.com/cyberman54/ESP32-Paxcounter/commit/3683164be3ad941490c175fdc12917cf48204bde

kskenyon, 2025-03-15:
> @kskenyon battery probing in heltec v3 now fixed by 3683164
> @kskenyon <https://github.com/kskenyon> battery probing in heltec v3 now fixed by 3683164 <https://github.com/cyberman54/ESP32-Paxcounter/commit/3683164be3ad941490c175fdc12917cf48204bde>

Almost...

./shared/hal/heltecv3.h:48:2: error: #endif without #if

cyberman54, 2025-03-15:
@kskenyon done.