cyberman54/ESP32-Paxcounter #1022: How to increase the sendcycle to measure several hours?

commeco, 2024-09-13:
Hello everyone,
I have tried to implement a factor of more than 2 to increase the measurement time. I need several hours of measurement time. 
Therefore it's possible to count more devices during the day and avoid multiple counts. (Paxcounter is powered externally)

I changed the multiplier in rcommand.cpp function ‘set_sendcycle’ and libpax_helpers.cpp function ‘init_libpax’ to 300. 

Nvram cleared etc.

But it does not work. With a ‘sendcycle’ of 72 and a multiplier of 300, the data are transferred every 125s instead of 6 hours (21600s).

Any ideas? I can't see it.. :-/

cyberman54, 2024-09-14:
Generally, it should work this way.

Maybe a value cast/conversion issue at some point? 
sendcycle value = uint8
libpax reporttimervalue = uint16
Xtimer value (inside libpax) = unit32

What happenes if you try *10, and *100?

commeco, 2024-09-14:
Yes, it goes in that direction.

I added this line in libpax_api.cpp in function "libpax_counter_init". 
```
ESP_LOGD("libpax", "xTimerCreate: xTimerPeriodInSencods => %ld ", pdMS_TO_TICKS(init_pax_report_interval_sec * 1000));
```
```
init_pax_report_interval_sec => 21600

The result of the macro ist: 125163
(It looks very similiar to my 125 seconds. :))

#define pdMS_TO_TICKS( xTimeInMs )    ( ( TickType_t ) ( ( ( TickType_t ) ( xTimeInMs ) * ( TickType_t ) configTICK_RATE_HZ ) / ( TickType_t ) 1000U ) )

configTICK_RATE_HZ is 1000 (defined in sdkconfig.h)

My expectation was 21600000. TickType_t is uint32. Plenty of space for this value.
```
Still no idea what it could be. :)

I'll do more deep dive..

commeco, 2024-09-14:
Okay, it's a known issue.
https://github.com/espressif/esp-idf/issues/4292

I'll use my own macro in FreeRTOSConfig.h. Due to the fact that the configTICK_RATE_HZ == 1000, i remove the divider and multiplier part because they're in my case unnecessary.

cyberman54, 2024-09-17:
Thanks for spotting this well hidden bug. Could you develop a workaround for this, and raise a PR?

commeco, 2024-09-17:
> Thanks for spotting this well hidden bug. Could you develop a workaround for this, and raise a PR?

https://github.com/dbinfrago/libpax/pull/28

cyberman54, 2024-09-17:
Thanks. Closing this issue here now.

closed 2024-09-17 by cyberman54