cyberman54/ESP32-Paxcounter #1023: Connect to the IoT cloud platform through the MQTT protocol

QehRGM, 2024-11-11:
There is no lorawan related gateway or device in my area, can I connect to the IoT cloud platform through the MQTT protocol after connecting to the public or my own wifi?

QehRGM, 2024-11-11:
I'm from China, and I want the project to be able to connect to wifi and then connect to a similar Alibaba Cloud IoT cloud platform through MQTT protocol

cyberman54, 2024-11-11:
Since the Wifi is used (and therefore busy with) sniffing, it's not possible via Wifi. But if you have a device with ethernet interface, you can use the eth to connect via TCP/IP and MQTT to cloud server. This is implemented, look in the code / documentation.

QehRGM, 2024-11-11:
I'm a novice, I tried to modify the code, successfully connected to my own wifi, and then connected to the cloud server through the MQTT protocol, but I don't know how to send the information such as the number of sniffed devices to the server through the MQTT protocol, the project can transmit relevant data through the MQTT protocol, which part should I modify

QehRGM, 2024-11-11:
After I connected to my own wifi, I was able to sniff the device as usual

cyberman54, 2024-11-20:
This is impossible, since the wifi sniffer is rotating the wifi channel continously. This would break your wifi server connection immediately after the channels rotation has started.

You may disable the channel wrapping using the WIFI_CHANNEL_MAP parameter in paxcounter.conf. But if do yo so, you won't get valid sniffing results since you're excluding ~90% of wifi radio traffic.

closed 2024-11-20 by cyberman54