cyberman54/ESP32-Paxcounter #1028: LilyGo T-Beam 1.2 Meshtastic not working with hal/ttgobeam12.h

juergen-gtlb, 2025-03-29:
Board: LilyGo T-Beam 1.2 [https://lilygo.cc/en-pl/products/t-beam](https://lilygo.cc/en-pl/products/t-beam)
Version: 868MHz soldered OLED
Model name printed on board: T-BEAM-AXP2101-V1.2
GPS: NEO-6M

With paxcounter from master and hal/ttgobeam12.h selected, the image crashes quickly after start.
Using other hal/ttgobeam*.h does not provide full functionality.

Reason: Board has a SX1276 radio, not a SX1262 radio.

Solution: in hal/ttgobeam12.h, define CFG_sx1276_radio instead of CFG_sx1262_radio (or create separate .h for this board)

cyberman54, 2025-03-29:
So it seems there are different T-Beam 1.2 versions: with LoRa chip 1262 and 1276. You already found the solution: select the appropriate chip type in hal file.

cyberman54, 2025-03-29:
push on master please. Maybe it would be better to create two different hal files, and give them appropriate files names?

juergen-gtlb, 2025-03-29:
Good idea to have separate hal files, I did that but cannot push - permission not granted.

I basically created a copy of ttgobeam12.h, named ttgobeam12-sx1276.h with a bit of documentation added and
#define CFG_sx1276_radio 1

Naming the HAL files seems a bit difficult, as naming just by boards does not cover different sub-versions of that board type,
and adding sub-version specifics (like radios) to the file name will lead to many different files.

cyberman54, 2025-03-30:
I [added hint ](https://github.com/cyberman54/ESP32-Paxcounter/commit/83b1b6c78b6eee25f29b6dd8ff27dc370f07de5f)in ttgobeam12.h file.
Thanks for reporting this. If you figure out a better solution, please open PR, thx.

closed 2025-03-30 by cyberman54